espressif/esp-adf #1341: Handling of single element pipeline

jaggi73, 2025-01-03:
## Environment

- Audio development kit:     none
-Module or chip used:  ESP32-WROVER-E
- IDF version v5.3.2
- [ADF version v2.7-47-g93a56fcd
- Build system:         idf.py
- Running log:      
`I (8022747) AUDIO_THREAD: The http task allocate stack on external memory
I (8022757) AUDIO_ELEMENT: [http-0x3f823334] Element task created
I (8022767) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:3863196 Bytes, Inter:84539 Bytes, Dram:83539 Bytes, Dram largest free:47104Bytes

I (8022777) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (8022777) HTTP_STREAM: _http_open
I (8022797) HTTP_STREAM: http_client_open start 0
I (8022777) AUDIO_PIPELINE: Pipeline started
I (8025037) esp-x509-crt-bundle: Certificate validated
I (8027387) HTTP_STREAM: http_client_open success
I (8028417) HTTP_STREAM: http_client_fetch_headers
I (8028417) HTTP_STREAM: Load URI: total_bytes=0

assert failed: xQueueSemaphoreTake queue.c:1709 (( pxQueue ))

Backtrace: 0x40081ffe:0x3f8333e0 0x4009674d:0x3f833400 0x4009d9cd:0x3f833420 0x40097082:0x3f833540 0x400e9070:0x3f833580 0x400e6def:0x3f8335b0 0x400ed6f9:0x3f8335e0 0x400e7382:0x3f833600 0x400e7516:0x3f833630 0x400973c1:0x3f833660
`

## Problem Description

I have a use case for a single element (http stream) pipeline where I use that to discover what the codec of the stream is.  Once that is known I then add to the pipeline based on what the codec is.  The problem is that after running through the pipeline multiple times it can crash on startup.  I believe this is caused the variable rb in function _pipeline_rb_linked in audio_pipeline.c.  I believe it is used after being freed from previous pipeline run.  
What fixes it for me is to check for both first and last being true and exit function before doing anything else.  Like this:

`static esp_err_t _pipeline_rb_linked(audio_pipeline_handle_t pipeline, audio_element_handle_t el, bool first, bool last)
{
    static ringbuf_handle_t rb;
    ringbuf_item_t *rb_item;

    if (first && last)
        return ESP_OK;`

### Expected Behavior
Single element pipeline runs

### Actual Behavior
Can crash after multiple runs

### Steps to Reproduce

1.  Run through a normal pipeline with multiple elements and at some point stop
2. Then start another pipeline with a single http stream element to discover codec
3.  There is a potential for crash

jason-mao, 2025-01-17:
@jaggi73 Thanks for your report; let me check and reply.

huisemaoy, 2025-02-26:
Although I was unable to reproduce your error, I have a general understanding of the cause of the problem. In the _pipeline_rb_linked function, the static ringbuf_handle_t rb is used. When you create the second pipeline and attempt to link it, at this point, rb is not empty (specifically, it should be the output_ringbuf of the second - to - last element in your first pipeline). Due to the particularity of a single - element pipeline, both the first and last variables are set to 1, so you executed the statement if (last) { audio_element_set_input_ringbuf(el, rb); }, which caused the second pipeline to use the ringbuf of the first pipeline, ultimately leading to the error.
Your modification to the code can indeed fix the error. I have made the following changes, which may be more in line with the logic of the code for your reference:
```c
static esp_err_t _pipeline_rb_linked(audio_pipeline_handle_t pipeline, audio_element_handle_t el, bool first, bool last)
{
    static ringbuf_handle_t rb;
    ringbuf_item_t *rb_item;
    if(first){         //Add this logical check.
        rb = 0;
    }
    if (last) {
        audio_element_set_input_ringbuf(el, rb);
    } 
    ...
}
```

closed 2025-02-28 by espressif-bot