espressif/esp-adf #1342: The Recording doesn't stop after specified seconds using example "pipeline_recording_to_sdcard"

osugiw, 2025-01-04:
## Environment

- Audio development kit:     Custom Board using ESP32C6, SD Card Module, and INMP441 Microphone
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): None
- [Required] Module or chip used:  ESP32C6
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v5.3.2-282-gcfea4f7c98
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): v2.7-47-g93a56fcd
- Build system:        idf.py
- [Required] Running log:     Ok

- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): xtensa-esp-elf-gcc.exe (crosstool-NG esp-13.2.0_20240530) 13.2.0
- Operating system:     Windows
- (Windows only) Environment type: Command Prompt
- Using an IDE?: Yes, VS Code ESPRESSIF Extension
- Power supply:      USB

## Problem Description

I tried to run "pipeline_recording_to_sdcard" with modified code that eliminates unnecessary lines for my application. The code can be compiled successfully but the recording process has never stopped when the time duration is exceeded also never save into the SD card.

### Expected Behavior
Stop the recording process and save the audio into the SD Card.

### Actual Behavior
Recording process exceeded the duration (10s) and never save it to the SD Card.

### Steps to Reproduce

1. Have no clue.

[image]

### Code to Reproduce This Issue

```cpp

[(https://gist.github.com/osugiw/e6c3737668f148fefa0a5bc2051adda2.js)]

```

## Debug Logs

```
[log.txt](https://github.com/user-attachments/files/18305697/log.txt)
```

## Other Items If Possible

- sdkconfig file
[sdkconfig.txt](https://github.com/user-attachments/files/18305717/sdkconfig.txt)

- Custom Board configs files
[esp32_c6_custom.zip](https://github.com/user-attachments/files/18305703/esp32_c6_custom.zip)

jason-mao, 2025-01-17:
I have seen an error `E (1002) AUDIO_PIPELINE: There is 1 link_tag invalid: i2s`; please check why this would happen. BTW, I can't get your codes.

osugiw, 2025-01-17:
> I have seen an error `E (1002) AUDIO_PIPELINE: There is 1 link_tag invalid: i2s`; please check why this would happen. BTW, I can't get your codes.

Oh my mistake, I forgot to attach it. Here is the code, I tried to use INMP441 microphone (which already has output of I2S Signal):

[main.zip](https://github.com/user-attachments/files/18453653/main.zip)