espressif/esp-adf #1343: MCLK disappeared after the `esp_codec_dev_open` is called when using APLL

L-KAYA, 2025-01-07:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version ESP32-LyraT v4.3
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): qa-test-full-master-20240913-2220-g3168cd3e95
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): [esp_codec_dev v1.3.3](https://components.espressif.com/components/espressif/esp_codec_dev/versions/1.3.3)
- Build system:         idf.py
- [Required] Running log: 
``` plain text
I (318) main_task: Calling app_main()
I (318) i2s_dsf_test: Running
D (318) i2s_common: tx channel is registered on I2S0 successfully
D (318) i2s_common: rx channel is registered on I2S0 successfully
D (328) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
D (338) i2s_common: APLL expected frequency is 24576000 Hz, real frequency is 24575996 Hz
D (338) i2s_std: Clock division info: [sclk] 24575996 Hz [mdiv] 1 [mclk] 12288000 Hz [bdiv] 8 [bclk] 1536000 Hz
D (348) i2s_common: MCLK is pinned to GPIO0 on I2S0
D (358) i2s_std: The rx channel on I2S0 has been initialized to STD mode successfully
D (368) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 960
W (378) i2s_common: APLL is occupied already, it is working at 24575996 Hz while the expected frequency is 24576000 Hz
W (388) i2s_common: Trying to work at 24575996 Hz...
D (388) i2s_common: APLL expected frequency is 24576000 Hz, real frequency is 24575996 Hz
D (398) i2s_std: Clock division info: [sclk] 24575996 Hz [mdiv] 1 [mclk] 12288000 Hz [bdiv] 8 [bclk] 1536000 Hz
D (408) i2s_common: MCLK is pinned to GPIO0 on I2S0
D (408) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
D (418) i2s_common: i2s rx channel enabled
D (428) i2s_common: i2s tx channel enabled
D (448) i2s_common: i2s tx channel disabled
D (448) i2s_common: i2s rx channel disabled
I (448) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
D (448) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 13 [mclk] 12288000 Hz [bdiv] 8 [bclk] 1536000 Hz
I (458) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:48000 mask:3
I (458) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
D (468) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 13 [mclk] 12288000 Hz [bdiv] 8 [bclk] 1536000 Hz
I (478) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:48000 mask:3
D (488) i2s_common: i2s tx channel enabled
D (488) i2s_common: i2s rx channel enabled
I (498) Adev_Codec: Open codec device OK
Read 19200 bytes
219 103 219 103
--------------------------
Read 19200 bytes
219 103 219 103
--------------------------
Read 19200 bytes
219 103 219 103
--------------------------
Read 19200 bytes
219 103 219 103
```

(Keep reading same data because no MCLK for the codec es8388)

- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
   xtensa-esp-elf-gcc (crosstool-NG esp-14.2.0_20241119) 14.2.0
- Operating system: macOS
- Using an IDE?:  No
- Power supply:  USB

## Problem Description

The MCLK still exist before `esp_codec_dev_open` but disappeared after it is called. For more detailed, it disappeared after i2s is re-configured in [these lines](https://github.com/espressif/esp-adf/blob/8a3b56a9b65af796164ebffc4e4bc45f144760b3/components/esp_codec_dev/esp_codec_dev.c#L177-L183)

### Code to Reproduce This Issue

[i2s_lyrat.zip](https://github.com/user-attachments/files/18330259/i2s_lyrat.zip)

L-KAYA, 2025-01-07:
Haven't went further to check if it caused by IDF driver eventually, post here to remind if anyone else got the same issue.

ThunderDai, 2025-12-16:
hi, esp_codec_dev use the defaults clk source at this: https://github.com/espressif/esp-adf/blob/fcce23e3d5ecf7ba75ce9f4e3761489dc9e0299e/components/esp_codec_dev/platform/audio_codec_data_i2s.c#L255

but you can config your clk source at this(>= 1.5.4): https://github.com/espressif/esp-adf/blob/fcce23e3d5ecf7ba75ce9f4e3761489dc9e0299e/components/esp_codec_dev/include/esp_codec_dev_defaults.h#L70