espressif/esp-adf #1344: 关于我想在wwe例子里面添加G711A或者opus解码，路线如下 iis（16000） ->raw,raw的数据用于唤醒，然后唤醒里面添加vad filter（16000-8000）g711a

PoohWoah, 2025-01-11:
`static audio_pipeline_handle_t app_wakeup_stream_init()
{
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    audio_pipeline_handle_t pipeline = audio_pipeline_init(&pipeline_cfg);
    if (NULL == pipeline)
    {
        return;
    }
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(CODEC_ADC_I2S_PORT, 16000, CODEC_BITS_PER_SAMPLE, AUDIO_STREAM_READER);
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);

    rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg.src_rate = 48000;
    rsp_cfg.dest_rate = 16000;
    rsp_cfg.task_core = 1;
    filter_read = rsp_filter_init(&rsp_cfg);

    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_READER;
    raw_read = raw_stream_init(&raw_cfg);

    audio_pipeline_register(pipeline, i2s_stream_reader, "i2s");
    audio_pipeline_register(pipeline, raw_read, "raw");

    if (filter_read)
    {
        audio_pipeline_register(pipeline, filter_read, "filter_read");
        const char *link_tag[3] = {"i2s", "filter_read", "raw"};
        audio_pipeline_link(pipeline, &link_tag[0], 3);
    }
    else
    {
        const char *link_tag[2] = {"i2s", "raw"};
        audio_pipeline_link(pipeline, &link_tag[0], 2);
    }

    // audio_pipeline_run(pipeline);

    ESP_LOGI(TAG, "Recorder has been created");

    recorder_sr_cfg_t recorder_sr_cfg = DEFAULT_RECORDER_SR_CFG();
    recorder_sr_cfg.afe_cfg.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
    recorder_sr_cfg.afe_cfg.wakenet_init = WAKENET_ENABLE;
    recorder_sr_cfg.afe_cfg.aec_init = RECORD_HARDWARE_AEC;
    recorder_sr_cfg.afe_cfg.agc_mode = AFE_MN_PEAK_NO_AGC;
    recorder_sr_cfg.afe_cfg.vad_init = VAD_ENBLE;
    recorder_sr_cfg.afe_cfg.vad_mode = VAD_MODE_4;
    recorder_sr_cfg.multinet_init = MULTINET_ENABLE;
    recorder_sr_cfg.mn_language = ESP_MN_CHINESE;

    audio_rec_cfg_t cfg = AUDIO_RECORDER_DEFAULT_CFG();
    cfg.read = (recorder_data_read_t)&input_cb_for_afe;
    cfg.sr_handle = recorder_sr_create(&recorder_sr_cfg, &cfg.sr_iface);

    cfg.vad_start = 50;
    cfg.event_cb = rec_engine_cb;
    cfg.vad_off = 600;
    recorder = audio_recorder_create(&cfg);

    return pipeline;
}`

这是我唤醒的代码，但是我看到例子里面有在唤醒里面进行数据处理的代码，当我打开这些代码后（例子是acc，我改成了G711），编译会跑飞。
`#if RECORDER_ENC_ENABLE
    recorder_encoder_cfg_t recorder_encoder_cfg = {0};
#if RECORDER_ENC_ENABLE == ENC_2_AMRNB
    rsp_filter_cfg_t filter_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    filter_cfg.src_ch = 1;
    filter_cfg.src_rate = 16000;
    filter_cfg.dest_ch = 1;
    filter_cfg.dest_rate = 8000;
    filter_cfg.stack_in_ext = true;
    filter_cfg.max_indata_bytes = 1024;

    amrnb_encoder_cfg_t amrnb_cfg = DEFAULT_AMRNB_ENCODER_CONFIG();
    amrnb_cfg.contain_amrnb_header = true;
    amrnb_cfg.stack_in_ext = true;

    recorder_encoder_cfg.resample = rsp_filter_init(&filter_cfg);
    recorder_encoder_cfg.encoder = amrnb_encoder_init(&amrnb_cfg);
#elif RECORDER_ENC_ENABLE == ENC_2_AMRWB
    amrwb_encoder_cfg_t amrwb_cfg = DEFAULT_AMRWB_ENCODER_CONFIG();
    amrwb_cfg.contain_amrwb_header = true;
    amrwb_cfg.stack_in_ext = true;
    amrwb_cfg.out_rb_size = 4 * 1024;

    recorder_encoder_cfg.encoder = amrwb_encoder_init(&amrwb_cfg);
#endif`

是不能再这里加入编码的内容吗？

PoohWoah, 2025-01-11:
意思就是我用了两个降低频率的初始化就会跑飞_filter_init

PoohWoah, 2025-01-11:
因为我服务器需要G711A管道编码，所以我想在唤醒后进行VAD检测然后把检测到的数据进行编码后输出给服务器，我试过不用管道直接raw元数据利用G711A编解码库进行编码，可行，但是效率很低。

请问你们有什么办法让我在唤醒后可以正常vad识别，然后可以利用第二条录音管道？
第一条唤醒：iis、filter、raw
第二条：iis filter alg g711a raw
关闭了唤醒管道后就无法进行vad检测了 ，貌似是无解的

shootao, 2025-02-12:
可以参考一下 相关的 [case](https://github.com/espressif/esp-adf/blob/f9660e31f08d45477637c8a060a4a0ad751b0254/examples/ai_agent/volc_rtc/components/audio_processor/audio_processor.c#L389)

closed 2025-04-18 by jason-mao