espressif/esp-adf #1348: audio_pipeline_stop无法立即停止http_stream_read_mp3的pipeline

casitelwithdean, 2025-01-14:
IDF 5.4 RC-1
ADF  Master
我使用的pipeline是 http_read-->mp3-->i2s，因为经常需要开启和关闭这个pipeline，但是关闭的时候有时候会等上1~2秒（有时甚至更久），才能完全关闭pipeline,即audio_pipeline_wait_for_stop不再阻塞。
我个人分析：这1~2秒的时间都花费在了http的读包的过程中，https://github.com/espressif/esp-adf/blob/93a56fcddcabe3be856df547fbc2db566ee2f740/components/audio_stream/http_stream.c#L805   这个函数里read_cb.cb一直在读包，默认一包是4096b，如果他没有读完这4096b的话，那么这个pipeline就不会关闭。     然后尤其是在刚和MP3建立连接的过程中，那么读取第一包4096b的时间将会比较长，有时甚至3~4秒（可能和网络有一些原因）。
所以有没有立马将含有http_stream_read的pipeline立马停止的方法呢？或许说强制性的将读取4096b的这个过程给打断，从而完全关闭pipeline呢？

jason-mao, 2025-01-17:
可以在外面先调用esp_http_client_close然后再stop pipeline

gongyb, 2025-07-02:
idf: v5.2.3; adf: v2.6
audio_element_process_deinit(http_stream_reader); 或 esp_http_client_close(http->client);
audio_pipeline_stop(pipeline);
播放http流，在audio_pipeline_stop 前执行 deinit 或 client_close， 高概率触发 core panic. 
遇到最长的 audio_pipeline_wait_for_stop(pipeline) 阻塞了有25秒。
还有没有其他办法解决这个阻塞的问题？