espressif/esp-adf #1352: VoIP例程语音通话时的回音问题

chmod777-h, 2025-01-20:
开发板用的ESP32-S3-Korvo-2 跑Voip的例程， 对端用的microSIP进行通话测试。

## Problem Description
1.问题现象是microSIP那边讲话时回音太大了，感觉跟开不开启AEC没啥效果。

  .algo_mask = ALGORITHM_STREAM_DEFAULT_MASK,
2.有尝试input_type改成2，参考信号的延迟，结果完全听不到声音。
#define ALGORITHM_STREAM_CFG_DEFAULT() {                                                          \
    .input_type = ALGORITHM_STREAM_INPUT_TYPE2, 
#if 0
    av_stream->ringbuf_ref = rb_create(AUDIO_MAX_SIZE, 1);
    audio_element_set_multi_input_ringbuf(av_stream->element_algo, av_stream->ringbuf_ref, 0);

    /* When the playback signal far ahead of the recording signal,
        the playback signal needs to be delayed */
    algo_stream_set_delay(av_stream->element_algo, av_stream->ringbuf_ref, 5);
#endif

各位老板测试过程中有遇到这个回音问题吗，都是怎么搞定的

TempoTian, 2025-01-20:
如果是MicroSIP那边的回音大，你调esp32这边应该没啥效果才是。调esp32这边可以在MicroSIP那边播放端听到效果。
ESP32-S3-Korvo-2用的是硬件回采不需要调algo_stream_set_delay，一般如果效果不好主要调录音的增益：避免失真或者信号太小。

bakecool, 2025-02-24:
@chmod777-h 你尝试algorithm示例，测试自己是否能够收到第三个麦克风的数据，如果没有参考信号（这个第三个麦克风的数据），那么就能确定你的AEC无效。