espressif/esp-adf #1354: GDMA TX alloc fail

kungfupizza, 2025-01-20:
So I have multiple audio pipeline setup along with wifi, SD, and keys peripherals on Korvo2 board.

Pipelines:
http -> i2s_out

fatfs -> mp3_decoder -> i2s_out

i2s_in -> https

I also have wifi, SD card, led_indicator, and keys peripherals enabled along with event listeners.

The init fails and gives out error as below after 3rd pipeline is attempted to initialised.

```
E (508) gdma: gdma_install_rx_interrupt(776): alloc interrupt failed
E (508) gdma: gdma_register_rx_event_callbacks(436): install interrupt service failed
```

What can I do to solve this?

TempoTian, 2025-01-21:
You can use API [esp_intr_dump()](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.4/esp32s3/api-reference/system/intr_alloc.html#_CPPv413esp_intr_dumpP4FILE) to dump the the interrupt occupied in each core. For you are using many peripheral, each peripheral needs allocate some interrupt, and allocate happened on the init function executed core. For most peripheral init happened on core0, you can move some peripheral init to core1 if core0 is full, like create a task pin to core and call sdcard init instead.

jason-mao, 2025-02-18:
How is your issue progressing? I would appreciate any updates on it.