espressif/esp-adf #1355: esp32-s3-box-3运行ai_agent/volc_rtc 唤醒词没反应

tangyiyong, 2025-01-21:
看见推送了这个volc_rtc就编译烧录测试了一下
环境如下：

esp-adf: master
esp-idf: 5.3.1
esp32-s3-box-3
idf.py set-target esp32s3
idf.py menuconfig
idf.py build flash monitor
申请了rtc的appid，生成了临时token，输出如下：

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x400450f2
--- 0x400450f2: ets_sha_process in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe5c
load:0x403cb700,len:0x303c
entry 0x403c893c
I (26) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Jan 21 2025 23:55:46
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling QIO for flash chip GD
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot:  3 model            Unknown data     01 82 00310000 0040e000
I (100) boot:  4 spiffs_data      Unknown data     01 82 0071e000 00010000
I (107) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c180020 size=52054h (335956) map
I (171) esp_image: segment 1: paddr=0006207c vaddr=3fc9f900 size=07e2ch ( 32300) load
I (177) esp_image: segment 2: paddr=00069eb0 vaddr=40378000 size=06168h ( 24936) load
I (182) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=179420h (1545248) map
I (420) esp_image: segment 4: paddr=001e9448 vaddr=4037e168 size=116cch ( 71372) load
I (444) boot: Loaded app from partition at offset 0x10000
I (444) boot: Disabling RNG early entropy source...
I (456) octal_psram: vendor id    : 0x0d (AP)
I (456) octal_psram: dev id       : 0x03 (generation 4)
I (456) octal_psram: density      : 0x05 (128 Mbit)
I (461) octal_psram: good-die     : 0x01 (Pass)
I (466) octal_psram: Latency      : 0x01 (Fixed)
I (472) octal_psram: VCC          : 0x00 (1.8V)
I (477) octal_psram: SRF          : 0x01 (Fast Refresh)
I (483) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (489) octal_psram: BurstLen     : 0x01 (32 Byte)
I (494) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (500) octal_psram: DriveStrength: 0x00 (1/1)
I (506) MSPI Timing: PSRAM timing tuning index: 6
I (511) esp_psram: Found 16MB PSRAM device
I (515) esp_psram: Speed: 80MHz
I (519) cpu_start: Multicore app
I (1087) esp_psram: SPI SRAM memory test OK
I (1096) cpu_start: Pro cpu start user code
I (1096) cpu_start: cpu freq: 240000000 Hz
I (1096) app_init: Application information:
I (1099) app_init: Project name:     volc_rtc
I (1104) app_init: App version:      1
I (1108) app_init: Compile time:     Jan 21 2025 23:55:33
[... 116 lines cut ...]
I (4076) volc_rtc: join room success

I (4076) RAW_OPUS_ENC: Raw Opus encoder init
I (4086) MODEL_LOADER: The storage free size is 14464 KB
I (4086) MODEL_LOADER: The partition size is 4152 KB
I (4096) MODEL_LOADER: Successfully load srmodels
I (4096) RECORDER_SR: The first wakenet model: wn9_nihaoxiaozhi_tts

I (4106) AFE_SR: afe interface for speech recognition

I (4116) AFE_SR: AFE version: SR_V220727

I (4116) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (4126) AFE_SR: aec_init: 1, se_init: 0, vad_init: 0

I (4136) AFE_SR: wakenet_init: 0

I (4286) AFE_SR: wake num: 2, mode: 0, (Sep  4 2024 11:49:31)

I (4286) AUDIO_RECORDER: RECORDER_CMD_TRIGGER_START
I (4286) main_task: Returned from app_main()

唤醒词没有响应，可能是什么原因？
谢谢！

qixinyo, 2025-01-23:
大哥，你有碰到日志报
E (3022) i2c.master: I2C transaction unexpected nack detected
E (3032) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (3042) i2c.master: i2c_master_transmit(1116): I2C transaction failed
这个i2c错误的问题吗

tangyiyong, 2025-01-23:
没有碰到啊，板子完好的话，不应该有I2C的问题吧
刚注意日志里时间好像不对，1970-01-01.。。

qixinyo, 2025-01-23:
你是啥板子了，板子上就带有对应音频模块的那种吗

qixinyo, 2025-01-23:
老哥，可以加你交流一下吗，我现在是在做sparkbot,一个桌面机器人，然后音频解码是es8311，单片机是esp32s3n16r8,现在实现的是文心一言的和小智的，但是无法实时对话，所以想把这套代码和进去

PoohWoah, 2025-02-08:
> 老哥，可以加你交流一下吗，我现在是在做sparkbot,一个桌面机器人，然后音频解码是es8311，单片机是esp32s3n16r8,现在实现的是文心一言的和小智的，但是无法实时对话，所以想把这套代码和进去

这套有点贵 你还得出RTC费用，进房间就开始算RTC费用

HaydnLiao, 2025-02-08:
> 大哥，你有碰到日志报 E (3022) i2c.master: I2C transaction unexpected nack detected E (3032) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed E (3042) i2c.master: i2c_master_transmit(1116): I2C transaction failed 这个i2c错误的问题吗

我也是，原来能正常用，后面就报这个错误。

ALToast, 2025-02-10:
> 大哥，你有备份日志报告 E (3022) i2c.master: I2C transaction意外nack检测到 E (3032) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed E (3042) i2c.master: i2c_master_transmit(1116): I2C transaction failed 这个i2c错误的问题吗

Hi @qixinyo, 
i2c hw nack 问题请参考 #I2C fails with hw NACK on ES8388 while I2S stream is running (AUD-5942) #1334

tianrongqin, 2025-02-10:
请问这个问题解决了吗？我也出现该问题了

JlaYun, 2025-02-18:
怎么样？我也是这样？有解决方案吗？

shootao, 2025-02-18:
@tangyiyong 
现在默认模式是 连续交互模式， 使用 唤醒的话， 可以menuconfig 配置成 **LANGUAGE_WAKEUP_MODE** 模式

JlaYun, 2025-02-19:
> [@tangyiyong](https://github.com/tangyiyong) 现在默认模式是连续交互模式，使用唤醒的话，可以menuconfig配置成**LANGUAGE_WAKEUP_MODE**模式

唤醒和还是不能对话，查看VolcEngineRTCLite对应的api发现已经发送到智能体所在rtc房间了

shootao, 2025-02-24:
@JlaYun 
目前 豆包有开临时 账号， 可以试一下
curl --location 'http://115.190.92.216:8899/startvoicechat' \
--header 'Content-Type: application/json' \
--header 'Authorization: af78e3067582ac8c04ba20174410bd1' \
--data '{"bot_id":"ep-20241104163951-sdmjc", "audio_codec":"G711A"}

closed 2025-02-25 by tangyiyong