espressif/esp-adf #1360: Memory leak in algorithm_stream when pausing and resuming pipeline

kylewang4929, 2025-02-10:
## Environment

- Audio development kit: ESP32-S3-BOX3
- Module or chip used: ESP32-S3
- IDF version: v5.3.1
- ADF version: v2.7-51-g60054d26
- Build system: idf.py
- Compiler version: (crosstool-NG esp-13.2.0_20240530) 13.2.0
- Operating system: macOS
- Using an IDE: cursor
- Power supply: usb

## Problem Description

There is a memory leak in the algorithm_stream component when calling `audio_pipeline_pause` and `audio_pipeline_resume` on a pipeline that contains an algorithm_stream.

The issue occurs because the AFE data is not properly cleaned up during pause/resume operations. When resuming, new AFE data is allocated without freeing the previous instance.

### Expected Behavior

When pausing and resuming a pipeline with algorithm_stream, the memory usage should remain stable.

### Actual Behavior

Memory usage gradually increases with each pause/resume cycle due to AFE data not being properly freed.

### Steps to Reproduce

1. Create an audio pipeline with an algorithm_stream
2. Call `audio_pipeline_pause`
3. Call `audio_pipeline_resume`
4. Repeat steps 2-3 multiple times
5. Observe increasing memory usage

## Additional Information

A minimal reproduction demo has been created to help isolate and verify this issue. The demo can be found at:
https://github.com/gizwits/algorithm_stream_demo

### Fix

The memory leak can be fixed by properly cleaning up AFE data in the `_algo_open` function before creating new instances. Here's the proposed fix:

```
c:components/audio_stream/algorithm_stream.c
static esp_err_t algo_open(audio_element_handle_t self)
{
…
// Add cleanup of existing AFE data before creating new instance
if (algo->afe_data) {
algo->afe_handle->destroy(algo->afe_data);
algo->afe_data = NULL;
}
...
}

```

I tried to clear memory during close, but it affected downstream nodes.

shootao, 2025-02-12:
Hi @kylewang4929 
Thank you for your feedback. We will fix it as soon as possible
The following method may be better
```c
diff --git a/components/audio_stream/algorithm_stream.c b/components/audio_stream/algorithm_stream.c
index 8caef6fc..08e44e9d 100644
--- a/components/audio_stream/algorithm_stream.c
+++ b/components/audio_stream/algorithm_stream.c
@@ -95,17 +95,16 @@ static esp_err_t _algo_close(audio_element_handle_t self)
     while (xEventGroupWaitBits(algo->state, FETCH_STOPPED_BIT, false, true, 10 / portTICK_PERIOD_MS) != FETCH_STOPPED_BIT) {
         algo->afe_handle->feed(algo->afe_data, algo->aec_buff);
     }
+    if (algo->afe_data) {
+        algo->afe_handle->destroy(algo->afe_data);
+        algo->afe_data = NULL;
+    }
     return ESP_OK;
 }
 
 static esp_err_t _algo_destroy(audio_element_handle_t self)
 {
     algo_stream_t *algo = (algo_stream_t *)audio_element_getdata(self);
-    if (algo->afe_data) {
-        algo->afe_handle->destroy(algo->afe_data);
-        algo->afe_data = NULL;
-    }
-
     if (algo->aec_buff) {
         audio_free(algo->aec_buff);
        
```

shootao, 2025-02-14:
Hi @kylewang4929 
[ESP ADF](https://github.com/espressif/esp-adf/commit/29de65cd50ebd8f46b75e55aaada0233c6a44f0d) has fixed this issue

closed 2025-02-14 by jason-mao