espressif/esp-adf #1362: audio noise in play_http_mp3_example when changing mp3 to wav

crazydiamondzgy, 2025-02-10:
## Environment

- Audio development kit:      [custom]
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v5.2.3
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.7-51-g60054d26
- Build system:         [Make|CMake|idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // xtensa-esp-elf-gcc.exe (crosstool-NG esp-13.2.0_20230928) 13.2.0
- Operating system:     [Windows]
- (Windows only) Environment type: [vscode]
- Using an IDE?: [vscode]

## Problem Description

i made a custom board for ESP-ADF, using TAS5805A,  everythings runs normally with play_http_mp3_example without modify
i confirmed that hardware and driver is right.

### Expected Behavior
no noise
### Actual Behavior
noisy
### Steps to Reproduce

1) change the mp3 file to test human voice (see attachments)
```    
  //audio_element_set_uri(http_stream_reader, "https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3");
  audio_element_set_uri(http_stream_reader, "http://192.168.1.100/foo.mp3");
```
the voice is normal
2) convert the foo.mp3 to fo

[mp3.zip](https://github.com/user-attachments/files/18728998/mp3.zip)

o.wav by goldwave.exe, the wav file was verified in computer (see attachments)
```
wav_decoder_cfg_t wav_cfg = DEFAULT_WAV_DECODER_CONFIG();
wav_decoder = wav_decoder_init(&wav_cfg);
....
audio_element_set_uri(http_stream_reader, "http://192.168.1.100/foo.wav");
```
the voice is noisy. most playback is right, but some noise can hear.

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

houhaiyan, 2025-02-13:
@crazydiamondzgy Please provide your test code

crazydiamondzgy, 2025-02-16:
play_http_mp3_example.c 
``` C
/* Play an MP3 file from HTTP

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "sdkconfig.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "http_stream.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
#include "wav_decoder.h"
#include "esp_peripherals.h"
#include "periph_wifi.h"
#include "board.h"

#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
#include "esp_netif.h"
#else
#include "tcpip_adapter.h"
#endif

static const char *TAG = "HTTP_MP3_EXAMPLE";

void app_main(void)
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
    ESP_ERROR_CHECK(esp_netif_init());
#else
    tcpip_adapter_init();
#endif

    audio_pipeline_handle_t pipeline;
    audio_element_handle_t http_stream_reader, i2s_stream_writer, mp3_decoder;

    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    ESP_LOGI(TAG, "[ 1 ] Start audio codec chip");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);

    ESP_LOGI(TAG, "[2.0] Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline);

    ESP_LOGI(TAG, "[2.1] Create http stream to read data");
[... 87 lines cut ...]
    audio_pipeline_terminate(pipeline);

    /* Terminate the pipeline before removing the listener */
    audio_pipeline_unregister(pipeline, http_stream_reader);
    audio_pipeline_unregister(pipeline, i2s_stream_writer);
    audio_pipeline_unregister(pipeline, mp3_decoder);

    audio_pipeline_remove_listener(pipeline);

    /* Stop all peripherals before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release all resources */
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(http_stream_reader);
    audio_element_deinit(i2s_stream_writer);
    audio_element_deinit(mp3_decoder);
    esp_periph_set_destroy(set);
}

```