espressif/esp-adf #1363: 为什么esp32s3-korvo-2 v3使用ai_agent/volc_rtc示例，在开启智能体的时候为什么出现以下问题？

tianrongqin, 2025-02-10:
在esp32s3-korvo-2 v3使用ai_agent/volc_rtc在火山引擎个人版去配置对应的数据以及开启智能体，日志打印如下：
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x19fc
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe8c
load:0x403cb700,len:0x3164
entry 0x403c8940
I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (27) boot: compile time Feb 10 2025 08:47:44
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : QIO
I (54) boot.esp32s3: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (82) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00300000
I (97) boot:  3 model            Unknown data     01 82 00310000 0040e000
I (104) boot:  4 spiffs_data      Unknown data     01 82 0071e000 00010000
I (112) boot: End of partition table
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c180020 size=4b9ech (309740) map
I (171) esp_image: segment 1: paddr=0005ba14 vaddr=3fca0000 size=04604h ( 17924) load
I (174) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=179c30h (1547312) map
I (406) esp_image: segment 3: paddr=001d9c58 vaddr=3fca4604 size=038cch ( 14540) load
I (409) esp_image: segment 4: paddr=001dd52c vaddr=40378000 size=17f10h ( 98064) load
I (441) boot: Loaded app from partition at offset 0x10000
I (441) boot: Disabling RNG early entropy source...
I (453) octal_psram: vendor id    : 0x0d (AP)
I (453) octal_psram: dev id       : 0x02 (generation 3)
I (453) octal_psram: density      : 0x03 (64 Mbit)
I (458) octal_psram: good-die     : 0x01 (Pass)
I (464) octal_psram: Latency      : 0x01 (Fixed)
I (469) octal_psram: VCC          : 0x01 (3V)
I (474) octal_psram: SRF          : 0x01 (Fast Refresh)
I (480) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (486) octal_psram: BurstLen     : 0x01 (32 Byte)
I (491) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (497) octal_psram: DriveStrength: 0x00 (1/1)
I (503) MSPI Timing: PSRAM timing tuning index: 4
I (508) esp_psram: Found 8MB PSRAM device
I (512) esp_psram: Speed: 80MHz
I (516) cpu_start: Multicore app
I (802) esp_psram: SPI SRAM memory test OK
I (811) cpu_start: Pro cpu start user code
I (811) cpu_start: cpu freq: 240000000 Hz
I (811) app_init: Application information:
I (814) app_init: Project name:     volc_rtc
I (819) app_init: App version:      1
I (823) app_init: Compile time:     Feb 10 2025 08:46:43
I (829) app_init: ELF file SHA256:  e0476f476...
I (834) app_init: ESP-IDF:          v5.3.2-dirty
I (840) efuse_init: Min chip rev:     v0.0
I (844) efuse_init: Max chip rev:     v0.99
I (849) efuse_init: Chip rev:         v0.2
I (854) heap_init: Initializing. RAM available for dynamic allocation:
I (861) heap_init: At 3FCAF458 len 0003A2B8 (232 KiB): RAM
I (867) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (874) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (880) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (888) spi_flash: detected chip: gd
I (892) spi_flash: flash io: qio
[... 110 lines cut ...]
I (12318) volc_rtc: join channel success test_room123 elapsed 268 ms now 268 ms

I (12318) volc_rtc: join room success

I (12318) RAW_OPUS_ENC: Raw Opus encoder init
I (12328) MODEL_LOADER: The storage free size is 22720 KB
I (12328) MODEL_LOADER: The partition size is 4152 KB
I (12338) MODEL_LOADER: Successfully load srmodels
I (12348) RECORDER_SR: The first wakenet model: wn9_hilexin

I (12348) AFE_SR: afe interface for speech recognition

I (12358) AFE_SR: AFE version: SR_V220727

I (12358) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (12368) AFE_SR: aec_init: 1, se_init: 0, vad_init: 0

I (12378) AFE_SR: wakenet_init: 0

I (12528) AFE_SR: wake num: 2, mode: 0, (Sep  4 2024 11:49:31)

I (12528) AUDIO_RECORDER: RECORDER_CMD_TRIGGER_START
I (12528) main_task: Returned from app_main()
请问怎么解决该问题？是开启智能体存在什么问题吗？具体可以怎么解决？

Zhentao-Lin, 2025-02-11:
My problem is similar to yours, but I print a lot of callback events that shouldn't be there.

[image]

shootao, 2025-02-11:
Hi @Zhentao-Lin @tianrongqin  
关于豆包， 它有两个阶段
第一个阶段是 设备端 进入房间， 就是 “join room success
第二个阶段是 智能体进入房间， 当两个同时 进入房间后 会有一个  WelcomeMessage 的下发， 设备端就会听到 智能体下发的 问候语
只有先听到 问候语后，才能 正常对话，先确定一下在什么阶段

Zhentao-Lin, 2025-02-11:
处于第二阶段，我使用的是V5.3.2版本，不太行。使用5.3.1版本则可以，请问是修改了哪些代码导致了图片中日志的问题呢

moonlight729, 2025-02-11:
byte_rtc_request 这个让智能体加入的代码都没有调用啊啊啊啊，代码有问题吧

tianrongqin, 2025-02-12:
> Hi [@Zhentao-Lin](https://github.com/Zhentao-Lin) [@tianrongqin](https://github.com/tianrongqin) 关于豆包， 它有两个阶段 第一个阶段是 设备端 进入房间， 就是 “join room success 第二个阶段是 智能体进入房间， 当两个同时 进入房间后 会有一个 WelcomeMessage 的下发， 设备端就会听到 智能体下发的 问候语 只有先听到 问候语后，才能 正常对话，先确定一下在什么阶段

现在已经可以听到问候语，但是问问题的时候，esp32s3-korvo-2 v3板没反应或者反应很慢，需要问几次才做出应答，并且输出的声音断断续续的，抖得很厉害，请问这种怎么解决？

tianrongqin, 2025-02-12:
> byte_rtc_request 这个让智能体加入的代码都没有调用啊啊啊啊，代码有问题吧

使用火山引擎开启调试去开启智能体加入房间的

tianrongqin, 2025-02-12:
@shootao hello
现在已经可以听到欢迎词了，但是问问题的时候，esp32s3-korvo-2 v3板没反应或者反应很慢，需要问几次才做出应答，并且输出的声音断断续续的，抖得很厉害，请问这种是什么原因导致的，可以怎么解决呢？

moonlight729, 2025-02-12:
> > byte_rtc_request 这个让智能体加入的代码都没有调用啊啊啊啊，代码有问题吧
> 
> 使用火山引擎开启调试去开启智能体加入房间的

感谢，能具体指出哪一行代码吗？我之前字节的跑起来了，这个一起看下

moonlight729, 2025-02-12:
> [@shootao](https://github.com/shootao) hello 现在已经可以听到欢迎词了，但是问问题的时候，esp32s3-korvo-2 v3板没反应或者反应很慢，需要问几次才做出应答，并且输出的声音断断续续的，抖得很厉害，请问这种是什么原因导致的，可以怎么解决呢？

说话声音大点试下

TempoTian, 2025-02-14:
@yejinghang 要在网页上startchat配置好， README有写：
然后在 [api-explorer](https://api.volcengine.com/api-explorer/?action=AssumeRole&groupName=%E8%A7%92%E8%89%B2%E6%89%AE%E6%BC%94&serviceCode=sts&version=2018-01-01) 中启动智能体，之后设备就可以连接到火山引擎进行语音交互了。

moonlight729, 2025-02-17:
@Zhentao-Lin 请问这个问题有解决吗？我现在也这样，有欢迎语，但是说话没反应

TempoTian, 2025-02-18:
@moonlight729 我这边测试是可以对话的，log是一样的，不过没有设定WAKEUP_MODE
```
1970-01-01 00:04:45.007 [E]  Counter.c:90 AudioRecevied fps 53
1970-01-01 00:04:45.036 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:04:45.288 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:04:45.401 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:04:45.748 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:04:46.009 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:04:46.297 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:04:46.722 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7

Done
tempo@tempo-OptiPlex-3080:~/adf-commit/esp-adf-internal/examples/ai_agent/volc_rtc$ cat sdkconfig |grep CONFIG_LANGUAGE_WAKEUP_MODE
# CONFIG_LANGUAGE_WAKEUP_MODE is not set

```

tianrongqin, 2025-02-18:
> > [@shootao](https://github.com/shootao) hello 现在已经可以听到欢迎词了，但是问问题的时候，esp32s3-korvo-2 v3板没反应或者反应很慢，需要问几次才做出应答，并且输出的声音断断续续的，抖得很厉害，请问这种是什么原因导致的，可以怎么解决呢？
> 
> 说话声音大点试下

还是一样，反应迟钝或者没啥反应

tianrongqin, 2025-02-18:
> [@moonlight729](https://github.com/moonlight729) 我这边测试是可以对话的，log是一样的，不过没有设定WAKEUP_MODE
>
> ```
> 1970-01-01 00:04:45.007 [E]  Counter.c:90 AudioRecevied fps 53
> 1970-01-01 00:04:45.036 [E] [...]

我这边是有设置唤醒词的，然后就是声音卡顿，说话没反应或者反应慢

moonlight729, 2025-02-18:
@tianrongqin 我的QQ是312394539。加我一下或者说下你的QQ，我加一下，谢谢

MichaelDu9226, 2025-02-18:
> [@tianrongqin](https://github.com/tianrongqin) 我的QQ是312394539。加我一下或者说下你的QQ，我加一下，谢谢

我这边提示加入房间成功，但是没有听到欢迎词。可能是智能体的语音识别和合成等没有配置成功。
请问你是如何配置的，有参考教程吗？另外你的火山账号是个人还是公司的，目前我看我的账号下都没有大模型语音合成这个选项。

tianrongqin, 2025-02-18:
> > [@tianrongqin](https://github.com/tianrongqin) 我的QQ是312394539。加我一下或者说下你的QQ，我加一下，谢谢
> 
> 我这边提示加入房间成功，但是没有听到欢迎词。可能是智能体的语音识别和合成等没有配置成功。 请问你是如何配置的，有参考教程吗？另外你的火山账号是个人还是公司的，目前我看我的账号下都没有大模型语音合成这个选项。

个人账号，具体如下：

tianrongqin, 2025-02-18:
> > [@tianrongqin](https://github.com/tianrongqin) 我的QQ是312394539。加我一下或者说下你的QQ，我加一下，谢谢
> 
> 我这边提示加入房间成功，但是没有听到欢迎词。可能是智能体的语音识别和合成等没有配置成功。 请问你是如何配置的，有参考教程吗？另外你的火山账号是个人还是公司的，目前我看我的账号下都没有大模型语音合成这个选项。

个人的，感觉不是很稳定

shootao, 2025-02-18:
服务开通可以参考 [火山文档](https://www.volcengine.com/docs/6348/1315561)
如果确定 账号， 可以先用 [web 端](https://www.volcengine.com/docs/6348/77374)先测试一下

moonlight729, 2025-02-18:
@MichaelDu9226 需要先了解一下服务开通。我的账号是个人的，跟个人还是公司没区别。

MichaelDu9226, 2025-02-18:
> [@MichaelDu9226](https://github.com/MichaelDu9226) 需要先了解一下服务开通。我的账号是个人的，跟个人还是公司没区别。

@shootao @moonlight729 照着文档申请了一圈，然后我现在可以看到有远程用户加入到房间了，但是听不到没有欢迎词。Hi乐鑫后，听到叮的一声，跟他对话也没反应。
看log有个mute audio和video，这是静音的意思？
请问这有可能是什么原因？谢谢
`I (1023457) volc_rtc: remote user offline  test_room:voiceChat_test_id_1739878003
I (1026177) volc_rtc: remote user joined  test_room:voiceChat_test_id_1739878167
1970-01-01 00:17:05.620 [E]  rx_net_delay_manager.c:1130 
I (1026427) volc_rtc: remote user mute audio  test_room:voiceChat_test_id_1739878167 0
I (1026427) volc_rtc: remote user mute video  test_room:voiceChat_test_id_1739878167 1
1970-01-01 00:17:05.637 [E]  EngineImplX.c:350 callback pEngineImplX->eventHandler.on_user_mute_video used too many times 5
I (1026477) volc_rtc: remote user mute audio  test_room:voiceChat_test_id_1739878167 0
I (1026477) volc_rtc: remote user mute video  test_room:voiceChat_test_id_1739878167 1`

shootao, 2025-02-18:
@MichaelDu9226 这个可能就是 设备进房了， 但是智能体没有进房
建议先用 web 端的先测一下

HaoWenT, 2025-02-20:
> [@shootao](https://github.com/shootao) [@moonlight729](https://github.com/moonlight729) 照着文档申请了一圈，然后我现在可以看到有远程用户加入到房间了，但是听不到没有欢迎词。Hi乐鑫后，听到叮的一声，跟他对话也没反应。 看log有个mute audio和video，这是静音的意思？ [...]

请问这个问题解决了吗？我的打印log跟你一样

NingZiXi, 2025-03-07:
> [@TempoTian](https://github.com/TempoTian) 请问，你的问题解决了吗？说话没有回应

请问您现在解决了吗，我也是这个问题

TempoTian, 2025-03-07:
@NingZiXi 你在网页上配置智能体了吗？要先在网页上启动智能体，后面才能对话。要不只能听到一句问候语。

NingZiXi, 2025-03-07:
我是有回复，但是断断续续不稳定

宁子希
***@***.***

TempoTian, 2025-03-10:
@NingZiXi 可以打下CPU利用率看看，可以用默认的sdkconfig排除sdkconfig配置导致的问题.  如果还有问题你在这里找shootao 帮你再看下。
```bash
# 这个命令可以用默认的sdkconfig
idf.py set-target esp32s3
```

NingZiXi, 2025-03-10:
万分感谢，我这就去试一下

宁子希
***@***.***

closed 2025-04-18 by jason-mao