espressif/esp-adf #1366: esp-adf缺少h文件：fatal error: raw_opus_encoder.h: No such file or directory

HaoWenT, 2025-02-11:
#if defined (CONFIG_AUDIO_SUPPORT_OPUS_DECODER)
#include "raw_opus_encoder.h"
#include "raw_opus_decoder.h"
#elif defined (CONFIG_AUDIO_SUPPORT_AAC_DECODER)
#include "aac_encoder.h"
#include "aac_decoder.h"
#elif defined (CONFIG_AUDIO_SUPPORT_G711A_DECODER)
#include "g711_encoder.h"
#include "g711_decoder.h"
#endif
请问我使用最新esp-adf的ai_agent/volc_rtc这个example，配置为ESP32-S3-Krovo-2,会自动在sdkconfig的# Example Configuration中使能CONFIG_AUDIO_SUPPORT_OPUS_DECODER这个宏，但是这个最新的esp-adf包中没有找到raw_opus_encoder.h这个库，然后我查找了一下发现aac_decoder.h \g711_decoder.h也是没有的，这几个文件我需要在哪里下载？

moonlight729, 2025-02-12:
我昨天刚跑了这个例程，没有报这个问题啊

shootao, 2025-02-12:
@HaoWenT 更新一下 [adf libs](https://github.com/espressif/esp-adf-libs/blob/master/esp_codec/include/codec/raw_opus_decoder.h)

jason-mao, 2025-02-18:
推荐使用`git submodule update --init --recursive`获取匹配的 `esp-adf-libs`

closed 2025-02-18 by jason-mao