espressif/esp-adf #1367: 使用adf的ota和idf的ota进行升级时，播放spiffs音乐会卡顿，不连贯

casitelwithdean, 2025-02-12:
adf版本：master
idf版本：5.2.1
spiffs播放音乐pipeline spiffs->mp3->i2s
在ota更新时，通过按键使其播放spiffs音频
ADF版本ota下：在ota前10秒左右，播放音频出现十分十分严重卡顿，10秒之后，再播放就流畅了
IDF版本ota下：ota的全部过程中播放音频都会有卡顿，但没有ADF前十秒的严重。
我尝试将spiffs_reader和mp3_decoder这两个ele的优先级调到了15，依旧没有改善。

crazydiamondzgy, 2025-02-12:
这个肯定啊, 一个写flash,一个读flash, 避免这种操作就好了

casitelwithdean, 2025-02-12:
> 这个肯定啊, 一个写flash,一个读flash, 避免这种操作就好了

好的 谢谢

casitelwithdean, 2025-02-12:
> 这个肯定啊, 一个写flash,一个读flash, 避免这种操作就好了

你好 我又尝试了另一种方式，一种不读取flash音频的方式：
第一步：我在cmakelist里通过EMBED_FILES "otaing.mp3"嵌入到flash里
第二步：我创建了一个char *ota_buffer，并heap_malloc_caps到spiram里，然后通过for循环将otaing.mp3里的数据赋值给ota_buffer，这样音频就存到了spiram里
第三步：建立一个mp3->i2s的 pipeline，并通过audio_element_set_read_cb(mp3_decoder, otaing_music_read_cb, NULL);读取spiram里的ota_buffer来播放

这么做还是会卡顿，请问可以帮忙看一下吗？谢谢！

crazydiamondzgy, 2025-02-13:
[OS 函数层目前支持访问锁和延迟的方法。

锁（见 :ref:spi_bus_lock）用于解决同一 SPI 总线上的设备访问和 SPI flash 芯片访问之间的冲突。例如：

经 SPI1 总线访问 flash 芯片时，应当禁用 cache（平时用于获取代码和 PSRAM 数据）。
经其他总线访问 flash 芯片时，应当禁用 flash 上 SPI 主驱动器注册的 ISR 以避免冲突。
SPI 主驱动器上某些没有 CS 线或者 CS 线受软件（如 SDSPI）控制的设备需要在一段时间内独占总线。
延时则用于某些长时操作，需要主机处于等待状态或执行轮询。

顶层 API 将芯片驱动和 OS 函数封装成一个完整的组件，并提供参数检查。

使用 OS 函数还可以在一定程度上避免在擦除大块 flash 区域时出现看门狗超时的情况。在这段时间内，CPU 将被 flash 擦除任务占用，从而阻止其他任务的执行，包括为看门狗定时器 (WDT) 供电的空闲任务。若已选中配置选项 :ref:CONFIG_ESP_TASK_WDT_PANIC，并且 flash 操作时间长于看门狗的超时时间，系统将重新启动。

不过，由于不同的 flash 芯片擦除时间不同，flash 驱动几乎无法兼容，很难完全规避超时的风险，这一点需要格外注意。请遵照以下指南：

建议启用 :ref:CONFIG_SPI_FLASH_YIELD_DURING_ERASE 选项，允许调度器在擦除 flash 时进行重新调度。此外，还可以使用下列参数。
在 menuconfig 中增加 :ref:CONFIG_SPI_FLASH_ERASE_YIELD_TICKS 或减少 :ref:CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS 的时间。
在 menuconfig 中增加 :ref:CONFIG_ESP_TASK_WDT_TIMEOUT_S 的时间，以设置更长的看门狗超时周期。然而，看门狗超时周期拉长后，可能无法再检测到以前可检测到的超时。
请注意，在进行长时间的 SPI flash 操作时，启用 :ref:CONFIG_ESP_TASK_WDT_PANIC 选项将会在超时时触发紧急处理程序。不过，启用该选项也可以帮助处理应用程序中的意外异常，请根据实际情况决定是否需要启用这个选项。
在开发过程中，请根据项目对擦除 flash 的具体要求和时间限制，谨慎进行 flash 操作。在配置 flash 擦除超时周期时，请在实际产品要求的基础上留出合理的冗余时间，从而提高产品的可靠性](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.2.3/esp32s3/api-reference/peripherals/spi_flash/index.html#os)

一般来说flash操作都是独占的, 除非万不得已, 没必要去在ota期间还做其它事情.

casitelwithdean, 2025-02-13:
> [OS 函数层目前支持访问锁和延迟的方法。
> 
> 锁（见 :ref:spi_bus_lock）用于解决同一 SPI 总线上的设备访问和 SPI flash 芯片访问之间的冲突。例如：
> 
> 经 SPI1 总线访问 flash 芯片时，应当禁用 cache（平时用于获取代码和 PSRAM 数据）。 经其他总线访问 flash 芯片时，应当禁用 flash 上 SPI 主驱动器注册的 ISR 以避免冲突。 SPI 主驱动器上某些没有 CS 线或者 CS 线受软件（如 SDSPI）控制的设备需要在一段时间内独占总线。 延时则用于某些长时操作，需要主机处于等待状态或执行轮询。
> 
> 顶层 API 将芯片驱动和 OS 函数封装成一个完整的组件，并提供参数检查。
> 
> 使用 OS 函数还可以在一定程度上避免在擦除大块 flash 区域时出现看门狗超时的情况。在这段时间内，CPU 将被 flash 擦除任务占用，从而阻止其他任务的执行，包括为看门狗定时器 (WDT) 供电的空闲任务。若已选中配置选项 :ref:CONFIG_ESP_TASK_WDT_PANIC，并且 flash 操作时间长于看门狗的超时时间，系统将重新启动。
> 
> 不过，由于不同的 flash 芯片擦除时间不同，flash 驱动几乎无法兼容，很难完全规避超时的风险，这一点需要格外注意。请遵照以下指南：
> 
> 建议启用 :ref:CONFIG_SPI_FLASH_YIELD_DURING_ERASE 选项，允许调度器在擦除 flash 时进行重新调度。此外，还可以使用下列参数。 在 menuconfig 中增加 :ref:CONFIG_SPI_FLASH_ERASE_YIELD_TICKS 或减少 :ref:CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS 的时间。 在 menuconfig 中增加 :ref:CONFIG_ESP_TASK_WDT_TIMEOUT_S 的时间，以设置更长的看门狗超时周期。然而，看门狗超时周期拉长后，可能无法再检测到以前可检测到的超时。 请注意，在进行长时间的 SPI flash 操作时，启用 :ref:CONFIG_ESP_TASK_WDT_PANIC 选项将会在超时时触发紧急处理程序。不过，启用该选项也可以帮助处理应用程序中的意外异常，请根据实际情况决定是否需要启用这个选项。 在开发过程中，请根据项目对擦除 flash 的具体要求和时间限制，谨慎进行 flash 操作。在配置 flash 擦除超时周期时，请在实际产品要求的基础上留出合理的冗余时间，从而提高产品的可靠性](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.2.3/esp32s3/api-reference/peripherals/spi_flash/index.html#os)
> 
> 1. 一般来说flash操作都是独占的, 除非万不得已, 没必要去在ota期间还做其它事情.
> 2. 如果两个task都是去读,问题不大, 这个时候是容许调度的,但是ota期间,有大量擦除操作,默认处理器是不调度的, 所以相当于要卡顿若干毫秒.
> 3. 按照文档说法, CONFIG_SPI_FLASH_YIELD_DURING_ERASE 就是专门解决你的问题的, 个人不推荐使用, 影响了ota, 问题会更多

谢谢答复！！！

closed 2025-02-13 by casitelwithdean