espressif/esp-adf #1369: A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.1 - v0.99] (this chip is revision v1.0). Use --force to flash anyway.

WhiteDoveBuct, 2025-02-12:
A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.1 - v0.99] (this chip is revision v1.0). Use --force to flash anyway.
idf: v5.3.1 
chip type: esp32p4
example project: esp-adf/examples/get-started/play_mp3_control
stage: flash

TempoTian, 2025-02-13:
Use following menuconfig to set chip version:

[image]

If not fixed try to use IDF v5.4 or master version but there maybe need fix some build issues for ADF.

WhiteDoveBuct, 2025-02-13:
您好，我会尽快回复您的！谢谢！

WhiteDoveBuct, 2025-02-13:
yes, how to fix some build issues for ADF v2.7 with IDF v5.4? compile error : 
/home/lzm/esp-adf/components/esp_peripherals/periph_lcd.c:102:46: error: passing argument 1 of 'periph_lcd->new_panel_io' makes pointer from integer without a cast [-Wint-conversion]
  102 |     ESP_ERROR_CHECK(periph_lcd->new_panel_io((esp_lcd_spi_bus_handle_t)periph_lcd->io_bus,
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              int

WhiteDoveBuct, 2025-02-13:
[image]
set like this, the same mistake

TempoTian, 2025-02-13:
you can comment it directly for LCD not used in playmp3 demo, for IDFv5.4 support, we will fix all the related build error all together afterwards.

jason-mao, 2025-02-14:
@WhiteDoveBuct ADF 2.7 does not support IDF v5.4. However, I have successfully compiled the ADF master with IDF v5.4. And the P4 revision v1.0 is support on v5.4.

closed 2025-02-14 by jason-mao