espressif/esp-adf #1370: esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream> 这个例子编译不过，要如何解决。esp32p4

catrowlee, 2025-02-12:
、 D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote\./Kconfig.soc_wifi_caps.in:205
info: INFO: Symbol SLAVE_SOC_WIFI_PHY_NEEDS_USB_WORKAROUND defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote\./Kconfig.soc_wifi_caps.in:181
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote\./Kconfig.soc_wifi_caps.in:109
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote\./Kconfig.soc_wifi_caps.in:153
info: INFO: Choice ESP_WIFI_TX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote/Kconfig:45
    D:/esp/esp-adf-2.7/esp-adf-2.7/Espressif/frameworks/esp-idf-v5.3.2/components/esp_wifi/Kconfig:62
info: INFO: Choice ESP_WIFI_MGMT_RX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp/esp-adf-2.7/esp-adf-2.7/Espressif/frameworks/esp-idf-v5.3.2/components/esp_wifi/Kconfig:133
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote/Kconfig:116
info: INFO: Choice ESP_WIFI_TASK_CORE_ID defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote/Kconfig:212
    D:/esp/esp-adf-2.7/esp-adf-2.7/Espressif/frameworks/esp-idf-v5.3.2/components/esp_wifi/Kconfig:229
[1341/1343] Linking CXX executable living_stream_m3u_aac.elf
FAILED: living_stream_m3u_aac.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && D:\esp\esp-adf-2.7\esp-adf-2.7\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -nostartfiles -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32P4=0 -Wl,--Map=D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/build/living_stream_m3u_aac.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32p4.peripherals.ld -T esp32p4.rom.ld -T esp32p4.rom.api.ld -T esp32p4.rom.rvfp.ld -T esp32p4.rom.wdt.ld -T esp32p4.rom.systimer.ld -T esp32p4.rom.version.ld -T esp32p4.rom.newlib.ld -T memory.ld -T sections.ld -Wl,--enable-non-contiguous-regions @CMakeFiles\living_stream_m3u_aac.elf.rsp -o living_stream_m3u_aac.elf && cd ."
D:/esp/esp-adf-2.7/esp-adf-2.7/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_set_ps':
D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:102:(.text.esp_wifi_set_ps+0x4): undefined reference to `esp_wifi_remote_set_ps'
D:/esp/esp-adf-2.7/esp-adf-2.7/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_get_config':
D:/esp/esp-adf-2.7/esp-adf-2.7/examples/player/pipeline_living_stream/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:202:(.text.esp_wifi_get_config+0x4): undefined reference to `esp_wifi_remote_get_config'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream\build\log\idf_py_stderr_output_25364 and D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream\build\log\idf_py_stdout_output_25364
PS D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream>
PS D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream>
PS D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream>
PS D:\esp\esp-adf-2.7\esp-adf-2.7\examples\player\pipeline_living_stream>

catrowlee, 2025-02-12:
ADF 2.7
IDF 5.3.2

TempoTian, 2025-02-13:
升级为 IDF release v5.4试一试

catrowlee, 2025-02-13:
> ADF 2.7 IDF 5.3.2

    D:/esp-adf-master/esp-idf/Espressif/frameworks/esp-idf-v5.4/components/esp_wifi/Kconfig:61
info: INFO: Choice ESP_WIFI_MGMT_RX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp-adf-master/esp-idf/Espressif/frameworks/esp-idf-v5.4/components/esp_wifi/Kconfig:132
    D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/Kconfig:116
info: INFO: Choice ESP_WIFI_TASK_CORE_ID defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/Kconfig:212
    D:/esp-adf-master/esp-idf/Espressif/frameworks/esp-idf-v5.4/components/esp_wifi/Kconfig:228
[1384/1386] Linking CXX executable play_http_mp3.elf
FAILED: play_http_mp3.elf
C:\Windows\system32\cmd.exe /C "cd . && D:\esp-adf-master\esp-idf\Espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -nostartfiles -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32P4=0 -Wl,--Map=D:/esp-adf-master/examples/player/pipeline_http_mp3/build/play_http_mp3.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32p4.peripherals.ld -T esp32p4.rom.ld -T esp32p4.rom.api.ld -T esp32p4.rom.rvfp.ld -T esp32p4.rom.wdt.ld -T esp32p4.rom.systimer.ld -T esp32p4.rom.version.ld -T esp32p4.rom.newlib.ld -T memory.ld -T sections.ld -Wl,--enable-non-contiguous-regions @CMakeFiles\play_http_mp3.elf.rsp -o play_http_mp3.elf && cd ."
D:/esp-adf-master/esp-idf/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_set_ps':
D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:102:(.text.esp_wifi_set_ps+0x4): undefined reference to `esp_wifi_remote_set_ps'
D:/esp-adf-master/esp-idf/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_get_config':
D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:202:(.text.esp_wifi_get_config+0x4): undefined reference to `esp_wifi_remote_get_config'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp-adf-master\examples\player\pipeline_http_mp3\build\log\idf_py_stderr_output_26516 and D:\esp-adf-master\examples\player\pipeline_http_mp3\build\log\idf_py_stdout_output_26516
PS D:\esp-adf-master\examples\player\pipeline_http_mp3>
还是编译不过

TempoTian, 2025-02-13:
应该是你切换idf了但是没有clean 导致的
```bash
rm -rf dependencies.lock; rm -rf build; rm -rf managed_components
```

catrowlee, 2025-02-13:
> 应该是你切换idf了但是没有clean 导致的
> 
> rm -rf dependencies.lock; rm -rf build; rm -rf managed_components

clean了还是不行啊。
idf.py set-target esp32p4 
=======
NOTICE: Processing 9 dependencies:
NOTICE: [1/9] espressif/eppp_link (0.2.0)
NOTICE: [2/9] espressif/esp-dsp (1.4.12)
NOTICE: [3/9] espressif/esp_hosted (0.0.27)
NOTICE: [4/9] espressif/esp_serial_slave_link (1.1.0)
NOTICE: [5/9] espressif/esp_wifi_remote (0.3.0)
NOTICE: [6/9] espressif/jsmn (1.1.0)
NOTICE: [7/9] espressif/nghttp (1.62.1)
NOTICE: [8/9] espressif/zlib (1.3.1)
NOTICE: [9/9] idf (5.4.0)
=======
idf.py build 
========================================================
info: INFO: Choice ESP_WIFI_TASK_CORE_ID defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/Kconfig:212
    D:/esp-adf-master/esp-idf/Espressif/frameworks/esp-idf-v5.4/components/esp_wifi/Kconfig:228
[1384/1386] Linking CXX executable play_http_mp3.elf
FAILED: play_http_mp3.elf
C:\Windows\system32\cmd.exe /C "cd . && D:\esp-adf-master\esp-idf\Espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -nostartfiles -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32P4=0 -Wl,--Map=D:/esp-adf-master/examples/player/pipeline_http_mp3/build/play_http_mp3.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32p4.peripherals.ld -T esp32p4.rom.ld -T esp32p4.rom.api.ld -T esp32p4.rom.rvfp.ld -T esp32p4.rom.wdt.ld -T esp32p4.rom.systimer.ld -T esp32p4.rom.version.ld -T esp32p4.rom.newlib.ld -T memory.ld -T sections.ld -Wl,--enable-non-contiguous-regions @CMakeFiles\play_http_mp3.elf.rsp -o play_http_mp3.elf && cd ."
D:/esp-adf-master/esp-idf/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_set_ps':
D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:102:(.text.esp_wifi_set_ps+0x4): undefined reference to `esp_wifi_remote_set_ps'
D:/esp-adf-master/esp-idf/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a(esp_wifi_with_remote.c.obj): in function `esp_wifi_get_config':
D:/esp-adf-master/examples/player/pipeline_http_mp3/managed_components/espressif__esp_wifi_remote/esp_wifi_with_remote.c:202:(.text.esp_wifi_get_config+0x4): undefined reference to `esp_wifi_remote_get_config'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp-adf-master\examples\player\pipeline_http_mp3\build\log\idf_py_stderr_output_5996 and D:\esp-adf-master\examples\player\pipeline_http_mp3\build\log\idf_py_stdout_output_5996
PS D:\esp-adf-master\examples\player\pipeline_http_mp3>
=============================================================

jason-mao, 2025-02-14:
@catrowlee 对这下面两个文件进行修改：
1. esp-adf-internal/examples/player/pipeline_living_stream/sdkconfig.defaults.esp32p4
```c
#### Comment below two lines if present:
# CONFIG_ESP_HOST_WIFI_ENABLED=y
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE=y

#### Add Wi-Fi Remote config for better performance:
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=32
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64

CONFIG_LWIP_TCP_SACK_OUT=y

```
2. esp-adf-internal/examples/player/pipeline_living_stream/main/idf_component.yml
```c
dependencies:
  espressif/esp_hosted:
    version: "~1.1"
    rules:
      - if: "target in [esp32p4]"
  espressif/esp_wifi_remote:
    version: "~0.6"
    rules:
      - if: "target in [esp32p4]"

```

3. `rm -rf managed_components dependencies.lock build`; `rm sdkconfig`;`idf.py set-target esp32p4`

catrowlee, 2025-02-14:
dependencies:
  espressif/esp_hosted:
    version: "~1.1"
    rules:
      - if: "target in [esp32p4]"
  espressif/esp_wifi_remote:
    version: "~0.4"
    rules:
      - if: "target in [esp32p4]"
    感谢。昨天将esp_wifi_remote的version换成0.4也可以解决。

closed 2025-02-14 by catrowlee