espressif/esp-adf #1372: AEC启用失败的问题

welkinchan, 2025-02-13:
我有一个项目，使用的是korvo2的开发板，
当将sr_cfg设置成单个麦克风的时候，可以正常工作。

由于开发板上有2个麦克风以及1个喇叭，
所以我想把AEC打开，测试下AEC的效果，但是测试下来，录音就彻底失效了。
原本可以进行语音唤醒，结果AEC打开的配置后，完全无法唤醒。

所以：
1. 想咨询下，打开AEC的话，除了下面的这些设置参数以外，是否还需要对其他的参数进行设置？
    recorder_sr_cfg.afe_cfg.aec_init = true;
    recorder_sr_cfg.afe_cfg.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
    recorder_sr_cfg.afe_cfg.agc_mode = AFE_MN_PEAK_NO_AGC;
    recorder_sr_cfg.multinet_init = true;
    recorder_sr_cfg.fetch_task_core = 0;
    recorder_sr_cfg.feed_task_core = 1;
    recorder_sr_cfg.afe_cfg.pcm_config.mic_num = 2;
    recorder_sr_cfg.afe_cfg.pcm_config.ref_num = 1;
    recorder_sr_cfg.afe_cfg.pcm_config.total_ch_num = 3;
    recorder_sr_cfg.input_order[0] = DAT_CH_0;
    recorder_sr_cfg.input_order[1] = DAT_CH_1;

    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(I2S_NUM_0, 16000, I2S_DATA_BIT_WIDTH_16BIT, AUDIO_STREAM_READER);
    i2s_cfg.std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_BOTH;
    i2s_cfg.std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_STEREO;
    i2s_stream_set_clk(_volc->i2s_stream_reader, 16000, I2S_DATA_BIT_WIDTH_16BIT, 2); 

2. 请问乐鑫是否有官方的打开了AEC功能的项目可供参考的？

welkinchan, 2025-02-13:
> [@welkinchan](https://github.com/welkinchan) 可以参考 dueros 的 [example](https://github.com/espressif/esp-adf/blob/f9660e31f08d45477637c8a060a4a0ad751b0254/examples/dueros/main/duer_audio_wrapper.c#L258)

我看了下这个项目里边的代码，里边的AEC配置是关了的啊。

#ifndef RECORD_HARDWARE_AEC
#warning "The hardware AEC is disabled!"
#define RECORD_HARDWARE_AEC (false)
#endif  /* RECORD_HARDWARE_AEC */

而且，我测试过，如果要打开AEC的话，程序会提示至少要2个MIC，
这个例程里的MIC_NUM配置的参数也是设置为1个。

还是我理解错了？

TempoTian, 2025-02-14:
目前的 AEC如果采集3 channel实际要用到TDM模式采集4声道的数据，现在ADF的例子都是用32bits 2channel来模拟采集16bits 4 channel的，你配置i2s就不能配置为16bits 2channel了。你看例子的代码也是处理的4声道的数据
```
#if defined CONFIG_ESP32_S3_KORVO2_V3_BOARD || defined CONFIG_ESP32_S3_BOX_3_BOARD
    rsp_cfg.mode = RESAMPLE_UNCROSS_MODE;
    rsp_cfg.src_ch = 4;
    rsp_cfg.dest_ch = 4;
    rsp_cfg.max_indata_bytes = 1024;
#endif
```

welkinchan, 2025-02-14:
> 目前的 AEC如果采集3 channel实际要用到TDM模式采集4声道的数据，现在ADF的例子都是用32bits 2channel来模拟采集16bits 4 channel的，你配置i2s就不能配置为16bits 2channel了。你看例子的代码也是处理的4声道的数据
>
> ```
> #if defined CONFIG_ESP32_S3_KORVO2_V3_BOARD || defined [...]

抱歉，我是业余搞这个项目的，你的解释我没理解哈。
意思要把i2s_stream_set_clk(_volc->i2s_stream_reader, 16000, I2S_DATA_BIT_WIDTH_16BIT, 2);
这里改成4吗？

dueros项目里边的aec方法看过去并不是配置afe_cfg.aec_init的方法来实现的。
我把这个项目扔给deepseek，deepseek也没看懂它是怎么实现AEC的。

================
好的，我需要分析用户提供的代码，看看AEC（Acoustic Echo Cancellation）是如何实现的。用户提到他们是个初学者，所以解释时要尽量简单明了。
首先，我查看提供的代码文件，特别是与音频处理相关的部分。在duer_audio_wrapper.c中，有一个函数duer_audio_start_recorder，它负责初始化录音器。这里可能涉及到AEC的设置。
在代码的第247-250行，有一个recorder_sr_cfg_t结构体的配置，其中afe_cfg.aec_init被设置为false。这让我有点困惑，因为AEC通常需要启用。不过，可能在其他地方有相关配置，或者使用了其他模块来处理回声消除。
接着，在rec_engine_cb回调函数（位于dueros_app.c）中，当检测到唤醒事件时，会启动录音并播放提示音。这里可能需要AEC来避免麦克风捕捉到扬声器的声音。
另外，在input_cb_for_afe函数中，通过raw_stream_read读取原始音频数据，这可能与前端处理（包括AEC）有关。但代码中没有直接显示AEC算法的实现，可能依赖于底层的ESP-ADF库或外部组件。
考虑到ESP-ADF通常集成了一些音频处理算法，AEC可能是在recorder_sr模块中默认启用的，或者通过配置afe_cfg的其他参数来激活。虽然当前配置中aec_init为false，但可能在硬件或库的底层有默认处理。
总结来说，项目可能通过ESP-ADF的recorder_sr模块和AFE（Audio Front-End）配置来实现AEC，尽管在代码中直接看到的设置有限，主要依赖库的内部处理。需要查阅ESP-ADF的文档确认AEC的具体实现方式。

sumic, 2025-02-17:
解决了吗？

welkinchan, 2025-02-18:
> 解决了吗？

还没呢。没资料，不知道怎么搞。

shootao, 2025-02-18:
@welkinchan  参考 dueros example 就可以了
1. 默认 dueros 使用的双麦 和 一路c参考的 [Links](https://github.com/espressif/esp-sr/blob/363e2991c3210ed05af979d011961d3aee7e7286/include/esp32s3/esp_afe_config.h#L145)
2. korov2 HARDWARE AEC 是使能的 [Link](https://github.com/espressif/esp-adf/blob/29de65cd50ebd8f46b75e55aaada0233c6a44f0d/components/audio_board/esp32_s3_korvo2_v3/board_def.h#L116)

JosephTang, 2025-02-18:
@welkinchan 

## 数据读取

- 如果是KORVO2的板子，有两路麦克风和一路回采信号，所以需要三个channel的数据，从[原理图](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-KORVO-2_V3_0_20210918.pdf)中可以看到，实际上这三个channel是接到了7210的三个输入上
- 目前的 [wwe](https://github.com/espressif/esp-adf/tree/master/examples/speech_recognition/wwe) 和 [dueros](https://github.com/espressif/esp-adf/tree/master/examples/dueros) 中都是使用的 32bit ，2 ch 的方式将这三个channel的数据读上来的，也就是7210将4个通道的16bit的数据，组合成了 2个通道，每个通道32bit。
- 所以i2s的配置应该是2ch，32bit。
- 结合7210的文档，将 i2s 读上来的 2ch，32bit 认为是 4ch，16 bit的数据之后，实际上 ch0 是回采信号，ch1 和 ch3 是麦克风数据，而ch2是无效数据。对应的有个默认的[input order](https://github.com/espressif/esp-adf/blob/29de65cd50ebd8f46b75e55aaada0233c6a44f0d/components/audio_recorder/include/recorder_sr.h#L84)

## AEC 的实现

- 上文提及的example中都是根据板子是否有回采信号来决定是否开启AEC
- AEC的实现是封装在esp-sr提供的 afe (audio front end) 组件中。只需要数据按照其规定排序后送入afe就可以实现aec的功能，这也正是目前recorder_sr 配合 input_order 这个配置项要做的事情。
- afe 支持的是 16k 采样率，16bit 的pcm数据。

## 用户代码修正

所以根据以上信息，需要注意以下几点

-  i2s的配置，16k 采样率，32bit数据，2ch
- recorder_sr_cfg.input_order 参考默认值

lamar-studio, 2025-02-19:
> [@welkinchan](https://github.com/welkinchan)
>
> ## 数据读取
> * [...]

您好，想问下您这边提到的afe组件是否开源呢？ 没有找到aec的相关实现代码

closed 2025-07-17 by jason-mao