espressif/esp-adf #1373: av_audio_enc_start  Ringbuffer of AFE is full的问题,  VOIP例程中出现的

JeremyXXJ, 2025-02-13:
我在VOIP例程上进行修改，我有两块ESP32_S3_KORVO2_V3_BOARD开发板
我将SIP服务期去除掉并使用 UDP socket进行音频数据的传输
我的路径是这样的
发送端  av_audio_enc_start  然后av_audio_enc_read(&udp_send_audio_frame, av_stream);  udp_send_audio_frame是我设置的缓存区 
再接着通过sendto 发送出去 
接收端 av_audio_dec_start 收到以后 int received = recvfrom(sock, udp_receive_audio_frame.data, 160, 0, (struct sockaddr *)&source_addr, &addr_len); ，av_audio_dec_write(&udp_receive_audio_frame, av_stream);再解码播放。
这是只接受和发送没有问题。

但是当我全双工的时候 一边录制一边播放的时候出现了问题，尤其是我发现在我单端接发仅仅在接收端调用了av_audio_enc_start 函数的时候出现了一个问题  
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
日志中频发打印这个消息
同时声音出现了一定程度的延迟，我该怎么解决和排查这个问题呢？

JeremyXXJ, 2025-02-13:
值得一提是当我修改 UDP数据收发线程的 延时时间从30到100也会出现这个日志 同时出现声音出现了延迟

ALToast, 2025-02-13:
Hi @JeremyXXJ ，该打印提示您 afe 的输出 buffer 已满，可以提高 afe->fetch() 的执行频率尽快取走数据。

shootao, 2025-02-13:
@JeremyXXJ 打印出现[task 信息](https://github.com/espressif/esp-adf/blob/f9660e31f08d45477637c8a060a4a0ad751b0254/components/audio_sal/include/audio_sys.h#L82), 看一下 task 的状态

shootao, 2025-02-13:
你这个怀疑是 aec 出来之后的数据 没有处理导致的

JeremyXXJ, 2025-02-13:
> Hi [@JeremyXXJ](https://github.com/JeremyXXJ) ，该打印提示您 afe 的输出 buffer 已满，可以提高 [afe->fetch()](https://github.com/espressif/esp-adf/blob/master/components/audio_recorder/recorder_sr.c#L269) 的执行频率尽快取走数据。

好的感谢您 我会排查一下您说的问题

JeremyXXJ, 2025-02-13:
> 你这个怀疑是 aec 出来之后的数据 没有处理导致的

static void _audio_enc(void* pv)
{
    av_stream_handle_t av_stream = (av_stream_handle_t) pv;
    xEventGroupClearBits(av_stream->aenc_state, ENCODER_STOPPED_BIT);

    char *frame_buf = NULL;
    int len = 0, timeout = 25 / portTICK_PERIOD_MS;
    int16_t *g711_buffer_16 = NULL;

    if (av_stream->config.acodec_type == AV_ACODEC_AAC_LC) {
        frame_buf = audio_malloc_align(16, AUDIO_MAX_SIZE);
    } else {
        frame_buf = audio_calloc(1, AUDIO_MAX_SIZE);
        g711_buffer_16 = (int16_t *)(frame_buf);
    }
    AUDIO_NULL_CHECK(TAG, frame_buf, return);

    while (av_stream->aenc_run) {
        int read_len = raw_stream_read(av_stream->audio_enc_read, frame_buf, av_stream->config.hal.audio_framesize);
        if (read_len == AEL_IO_TIMEOUT) {
            continue;
        } else if (read_len < 0) {
            break;
        }

        av_stream_frame_t enc;
        enc.len = read_len;
        enc.data = audio_calloc(1, AUDIO_MAX_SIZE);
        AUDIO_NULL_CHECK(TAG, enc.data, break);
        av_stream->audio_pos += enc.len;
        enc.pts = (av_stream->audio_pos * 1000) / (av_stream->config.hal.audio_samplerate * 1 * 16 / 8);
        switch ((int)av_stream->config.acodec_type) {
            case AV_ACODEC_PCM:
                memcpy(enc.data, frame_buf, enc.len);
                break;
            case AV_ACODEC_AAC_LC:
                esp_aac_enc_process(av_stream->aac_enc, (uint8_t *)frame_buf, enc.len, enc.data, &len);
                enc.len = len;
                timeout = 0;
                break;
            case AV_ACODEC_G711A:
                for (int i = 0; i < enc.len; i++) {
                    enc.data[i] = esp_g711a_encode(g711_buffer_16[i]);
                }
                enc.len = enc.len/2;
                break;
            case AV_ACODEC_G711U:
                for (int i = 0; i < enc.len; i++) {
                    enc.data[i] = esp_g711u_encode(g711_buffer_16[i]);
                }
                enc.len = enc.len/2;
                break;
        }
        if (xQueueSend(av_stream->aenc_queue, &enc, timeout) != pdTRUE) {
            ESP_LOGD(TAG, "send aenc buf queue timeout !");
            free(enc.data);
        }
    }
    ESP_LOGI(TAG, "_audio_enc task stoped");

    if (av_stream->config.acodec_type == AV_ACODEC_AAC_LC) {
        jpeg_free_align(frame_buf);
    } else {
        free(frame_buf);
    }

    xEventGroupSetBits(av_stream->aenc_state, ENCODER_STOPPED_BIT);
    vTaskDelete(NULL);
} 您好这会是库函数的问题吗  因为我没有对库函数做修改 我仅仅是调用了这个函数  我在发送端调用该函数的时候没有出现问题 只是修改循环延时的时候 出现了 这个问题已经困扰了我很久了     aec 出来之后的数据是在哪里进行处理呢  很抱歉这个时间打扰到您

JeremyXXJ, 2025-02-13:
> 你这个怀疑是 aec 出来之后的数据 没有处理导致的

从声音来看 他确实和VOIP例程里的声音不一样 有一些回音

shootao, 2025-02-13:
把你的 整个 c 文件贴出来看看

JeremyXXJ, 2025-02-13:
> 把你的 整个 c 文件贴出来看看

/* VoIP Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_mem.h"
#include "esp_peripherals.h"
#include "input_key_service.h"
#include "wifi_service.h"
#include "smart_config.h"
#include "sip_service.h"
#include "audio_sys.h"
#include "algorithm_stream.h"
#include "audio_idf_version.h"

#include <inttypes.h> 
#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include <lwip/netdb.h>
#include "av_stream.h"
#include "esp_rtc.h"

#define BROADCAST_IP "255.255.255.255"
#define BROADCAST_PORT 12345
// #define COMFIG_EXAMPLE_IPV4_ADDR "255.255.255.255"
// #define HOST_ID_ADDR COMFIG_EXAMPLE_IPV4_ADDR
#define PORT 3333
static int sock = -1;
// struct sockaddr_in dest_addr;
struct sockaddr_storage source_addr;
socklen_t socklen = sizeof(source_addr);
#define MAX_UDP_PACKET_SIZE 1024

struct sockaddr_storage audio_source_addr; // Large enough for both IPv4 or IPv6
socklen_t audio_socklen = sizeof(audio_source_addr);
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
#include "esp_netif.h"
#else
#include "tcpip_adapter.h"
#endif

static const char *TAG = "VoIP_EXAMPLE";

#define WIFI_SSID   CONFIG_WIFI_SSID
#define WIFI_PWD    CONFIG_WIFI_PASSWORD
#define SIP_URI     CONFIG_SIP_URI

static esp_rtc_handle_t esp_sip;
static bool is_smart_config;
static av_stream_handle_t av_stream;

static esp_err_t input_key_service_cb(periph_service_handle_t handle, periph_service_event_t *evt, void *ctx)
{
    int player_volume;
    periph_service_handle_t wifi_serv = (periph_service_handle_t) ctx;
    if (evt->type == INPUT_KEY_SERVICE_ACTION_CLICK_RELEASE) {
        ESP_LOGD(TAG, "[ * ] input key id is %d", (int)evt->data);
        switch ((int)evt->data) {
[... 251 lines cut ...]
    ESP_LOGI(TAG, "[ 2 ] Initialize av stream");
    av_stream_config_t av_stream_config = {
        .algo_mask = ALGORITHM_STREAM_DEFAULT_MASK,
        .acodec_samplerate = AUDIO_CODEC_SAMPLE_RATE,
        .acodec_type = AV_ACODEC_G711A,
        .vcodec_type = AV_VCODEC_NULL,
        .hal = {
            .audio_samplerate = AUDIO_HAL_SAMPLE_RATE,
            .audio_framesize = PCM_FRAME_SIZE,
        },
    };
    av_stream = av_stream_init(&av_stream_config);
    av_audio_enc_start(av_stream);
    av_audio_dec_start(av_stream);

    xTaskCreate(udp_server_task,"udp_audio_server",4096 * 2,(void *)av_stream,5,NULL  );
    AUDIO_NULL_CHECK(TAG, av_stream, return);

    while (1) {
        audio_sys_get_real_time_stats();
        AUDIO_MEM_SHOW(TAG);
        vTaskDelay(15000 / portTICK_PERIOD_MS);
    }
} 主要修改的就是将sip初始化注释掉换成了  xTaskCreate(udp_server_task,"udp_audio_server",4096 * 2,(void *)av_stream,5,NULL  );
差别再main函数和udp_server_task函数中

JeremyXXJ, 2025-02-13:
> 把你的 整个 c 文件贴出来看看

[voip.txt](https://github.com/user-attachments/files/18784772/voip.txt)或有一些语法错误 此时板子不在身边

shootao, 2025-02-13:
av_stream.c 里面的代码有改过嘛？

JeremyXXJ, 2025-02-13:
> av_stream.c 里面的代码有改过嘛？

应该没有 我明天再确认下

JeremyXXJ, 2025-02-14:
> av_stream.c 里面的代码有改过嘛？

[voip.txt](https://github.com/user-attachments/files/18792185/voip.txt)
[av_stream.txt](https://github.com/user-attachments/files/18792186/av_stream.txt) 没有修改过

TempoTian, 2025-02-14:
看代码应该是自己实现的 VOIP，encode 完的数据可以保存下来听听是否有问题。没问题整个encoder通路都就是OK的。
encode 完的数据会通过网络发送，确保发送接收task的优先级较高比如编码抢占而不被占用。另外用socket尽量不要用delay的方式可以用 selec t取，monitor可用状态。

JeremyXXJ, 2025-02-14:
> 看代码应该是自己实现的 VOIP，encode 完的数据可以保存下来听听是否有问题。没问题整个encoder通路都就是OK的。 encode 完的数据会通过网络发送，确保发送接收task的优先级较高比如编码抢占而不被占用。另外用socket尽量不要用delay的方式可以用 selec t取，monitor可用状态。

谢谢您的指导  我想知道我现在暂时不用select 出现我所提到的问题吗     encode在半双工时看起来是没有问题的  “确保发送接收task的优先级较高比如编码抢占而不被占用”这句话我不太理解 是要网络发送任务的优先级高于编解码中 21的优先级吗

JeremyXXJ, 2025-02-14:
> 是的，编解码配置为21太高了，网络任务优先级要高于编解码任务的优先级。保证网络低CPU占用率的任务能及时调度到

我已经修改了，但问题没有解决 ，此时我又丰富了一下 出现情况的描述，希望您能给予指导  

情况1：但是当我全双工的时候 一边录制一边播放的时候出现了问题，AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!! 报错并且此时抓包没有数据传输 ，打印日志发现两端均在打印AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
 

情况2： 在半双工的基础上，当我再接收端加上网络发送和编码的代码时，发现可以正常发送数据，此时我又加了一个板子作为接收端，第三个板子出现了声音的延迟。

情况3： 在半双工的基础上，当我再发送端加上网络接收和解码的代码时，报错
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
AFE_VC: Ringbuffer of AFE is full, Please use fetch() to retrieve data from the ringbuffer soon to avoid data loss or overwriting!!!
此时无法正常工作了。

TempoTian, 2025-02-14:
看样子你编码发送的逻辑是通的。你可以加一下log对比下。正常和NG情况下从AFE读了多少比数据，编码了多少次。log看主要是读慢了。你可以把一些环节注释掉看到底是哪里引起的。比如注释掉发送，注释掉解码等。

JeremyXXJ, 2025-02-15:
> 看样子你编码发送的逻辑是通的。你可以加一下log对比下。正常和NG情况下从AFE读了多少比数据，编码了多少次。log看主要是读慢了。你可以把一些环节注释掉看到底是哪里引起的。比如注释掉发送，注释掉解码等。

感谢！

closed 2025-02-17 by JeremyXXJ