espressif/esp-adf #1375: esp32s3-korvo-2 v3使用ai_agent/volc_rtc示例，有欢迎提示音，说唤醒词也有反应。但是和大模型对话，没有任何反应

moonlight729, 2025-02-17:
`I (80322) volc_rtc: rec_engine_cb - AUDIO_REC_VAD_START
I (80322) audio processor: player pipe stop running
1970-01-01 00:01:20.003 [E]  Counter.c:90 AudioRecevied fps 50
I (81142) volc_rtc: rec_engine_cb - AUDIO_REC_WAKEUP_START
W (81142) audio processor: player pipe is idle state
I (81142) audio processor: audio_tone_play: spiffs://spiffs/dingding.wav
I (81152) SPIFFS_STREAM: File size is 12024 byte, pos:0
I (81162) CODEC_ELEMENT_HELPER: The element is 0x3c1d138c. The reserve data 2 is 0x0.
I (81172) WAV_DECODER: a new song playing
I (81172) audio processor: [ * ] Receive music info from wav decoder, sample_rates=16000, bits=16, ch=1
W (81212) SPIFFS_STREAM: No more data, ret:0
I (81212) DEC_WAV: Closed
I (81332) audio processor: [ * ] Stop event received
I (81332) audio processor: [ * ] Stop event received
1970-01-01 00:01:21.008 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:22.021 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:23.005 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:01:24.011 [E]  Counter.c:90 AudioRecevied fps 50
I (85442) volc_rtc: rec_engine_cb - AUDIO_REC_VAD_START
W (85442) audio processor: player pipe is idle state
1970-01-01 00:01:25.009 [E]  Counter.c:90 AudioRecevied fps 55
1970-01-01 00:01:26.006 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:01:27.009 [E]  Counter.c:90 AudioRecevied fps 50
I (88012) volc_rtc: rec_engine_cb - AUDIO_REC_VAD_END
I (88012) audio processor: player pipe start running
1970-01-01 00:01:27.564 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:27.599 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:27.619 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:27.878 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:28.016 [E]  Counter.c:90 AudioRecevied fps 47
1970-01-01 00:01:28.259 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:28.359 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:28.379 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:28.459 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:28.519 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:28.889 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:28.914 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:29.006 [E]  Counter.c:90 AudioRecevied fps 48
1970-01-01 00:01:29.279 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:29.478 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:29.665 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:29.839 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:30.018 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:30.819 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:30.939 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:31.010 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:31.034 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:31.579 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:31.733 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:31.777 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:32.007 [E]  Counter.c:90 AudioRecevied fps 52
1970-01-01 00:01:32.396 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:32.436 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:32.939 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:33.009 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:01:33.099 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:33.996 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:34.011 [E]  Counter.c:90 AudioRecevied fps 53
1970-01-01 00:01:34.096 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:34.406 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:34.618 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:35.009 [E]  Counter.c:90 AudioRecevied fps 52
1970-01-01 00:01:35.186 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:35.698 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:36.005 [E]  Counter.c:90 AudioRecevied fps 43
1970-01-01 00:01:36.407 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:36.719 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:36.779 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:36.967 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:37.007 [E]  Counter.c:90 AudioRecevied fps 45
[... 7 lines cut ...]
1970-01-01 00:01:38.439 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:38.479 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:39.018 [E]  Counter.c:90 AudioRecevied fps 52
1970-01-01 00:01:39.239 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:39.299 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:39.459 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:40.002 [E]  Counter.c:90 AudioRecevied fps 53
1970-01-01 00:01:40.799 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:40.899 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:40.919 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:41.011 [E]  Counter.c:90 AudioRecevied fps 47
1970-01-01 00:01:41.017 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:41.559 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:41.599 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:41.768 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:42.018 [E]  Counter.c:90 AudioRecevied fps 55
1970-01-01 00:01:42.359 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:42.777 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:42.926 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:43.010 [E]  Counter.c:90 AudioRecevied fps 48
1970-01-01 00:01:43.139 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:43.628 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:44.006 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:01:44.545 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:44.646 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5`

moonlight729, 2025-02-17:
咨询了字节，说“看是否调整硬件采集增益调整大一些”

moonlight729, 2025-02-18:
这个问题没人解答吗？

moonlight729, 2025-02-18:
抱歉，代码有问题，发送音频的房间信息没有修改过来

closed 2025-02-18 by moonlight729

HaoWenT, 2025-02-24:
> 抱歉，代码有问题，发送音频的房间信息没有修改过来

您好，请问一下需要修改什么参数？我现在不用唤醒词可以正常对答，加上唤醒词后对话没反应

EmpertyZ, 2025-03-17:
> > 抱歉，代码有问题，发送音频的房间信息没有修改过来
> 
> 您好，请问一下需要修改什么参数？我现在不用唤醒词可以正常对答，加上唤醒词后对话没反应

这个问题， 朋友你解决了没，我也是这个情况