espressif/esp-adf #1377: ESP32-S3-KORVO-2运行ai_agent/volc_rtc加入房间后没声音

HaoWenT, 2025-02-19:
(27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Feb 17 2025 21:11:32
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot:  3 model            Unknown data     01 82 00310000 0040e000
I (100) boot:  4 spiffs_data      Unknown data     01 82 0071e000 00010000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c180020 size=5243ch (336956) map
I (171) esp_image: segment 1: paddr=00062464 vaddr=3fc9f900 size=07edch ( 32476) load
I (177) esp_image: segment 2: paddr=0006a348 vaddr=40378000 size=05cd0h ( 23760) load
I (182) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=179d74h (1547636) map
I (420) esp_image: segment 4: paddr=001e9d9c vaddr=4037dcd0 size=11bb4h ( 72628) load
I (445) boot: Loaded app from partition at offset 0x10000
I (445) boot: Disabling RNG early entropy source...
I (457) octal_psram: vendor id    : 0x0d (AP)
I (457) octal_psram: dev id       : 0x02 (generation 3)
I (457) octal_psram: density      : 0x03 (64 Mbit)
I (462) octal_psram: good-die     : 0x01 (Pass)
I (467) octal_psram: Latency      : 0x01 (Fixed)
I (473) octal_psram: VCC          : 0x01 (3V)
I (478) octal_psram: SRF          : 0x01 (Fast Refresh)
I (483) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (489) octal_psram: BurstLen     : 0x01 (32 Byte)
I (495) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (501) octal_psram: DriveStrength: 0x00 (1/1)
I (507) MSPI Timing: PSRAM timing tuning index: 5
I (511) esp_psram: Found 8MB PSRAM device
I (516) esp_psram: Speed: 80MHz
I (520) cpu_start: Multicore app
I (809) esp_psram: SPI SRAM memory test OK
I (818) cpu_start: Pro cpu start user code
I (818) cpu_start: cpu freq: 240000000 Hz
I (818) app_init: Application information:
I (821) app_init: Project name:     volc_rtc
I (826) app_init: App version:      1
I (830) app_init: Compile time:     Feb 18 2025 19:39:37
I (836) app_init: ELF file SHA256:  5d477f077...
I (841) app_init: ESP-IDF:          v5.3.1-dirty
I (847) efuse_init: Min chip rev:     v0.0
I (851) efuse_init: Max chip rev:     v0.99 
I (856) efuse_init: Chip rev:         v0.2
I (861) heap_init: Initializing. RAM available for dynamic allocation:
I (868) heap_init: At 3FCAED48 len 0003A9C8 (234 KiB): RAM
I (874) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (881) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (887) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (895) spi_flash: detected chip: gd
I (898) spi_flash: flash io: qio
W (903) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (911) sleep: Configure to isolate all GPIO pins in sleep state
I (918) sleep: Enable automatic switching of GPIO sleep configuration
I (925) main_task: Started on CPU0
I (937) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (938) main_task: Calling app_main()
I (946) main: Initialize board peripherals
I (950) PERIPH_SPIFFS: Partition size: total: 52961, used: 12299
I (954) AUDIO_THREAD: The esp_periph task allocate stack on internal memory
W (961) i2c_bus_v2: I2C master handle is NULL, will create new one
I (968) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (977) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
[... 136 lines cut ...]
I (34604) wifi:<ba-add>idx:1 (ifx:0, bc:5f:f6:8f:30:8c), tid:6, ssn:5, winSize:64
1970-01-01 00:00:34.004 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:35.007 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:36.003 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:37.004 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:38.009 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:39.004 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:40.006 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:41.001 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:42.033 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:00:43.002 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:00:44.008 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:45.003 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:46.002 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:47.015 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:48.011 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:49.008 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:50.010 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:00:51.005 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:52.022 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:00:53.001 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:00:54.022 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:00:55.004 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:00:56.001 [E]  Counter.c:90 AudioRecevied fps 50
请问一下：用GetRoomOnlineUsers这个API看到设备和智能体都进入了同一个房间，但是没有听到欢迎词，提问也没有回复，是什么问题？需要怎么排查？调试web端交互式AIGC场景 AIGC Demo这个是调试成功的，用网页端连接是可以问答对话的，只是ESP设备进入房间后听不到声音

shootao, 2025-02-24:
@HaoWenT 
参考一下 [Issue1378](https://github.com/espressif/esp-adf/issues/1378), 先试用 临时账号试一下

shunyuchu, 2025-03-07:
@shootao 我也是这样的问题，你这个解决了嘛？

HaoWenT, 2025-03-07:
> [@HaoWenT](https://github.com/HaoWenT) 我也是这样的问题，你这个解决了嘛？

配置智能体的问题，重新检查一下配置

wcu1117, 2025-03-13:
> > [@HaoWenT](https://github.com/HaoWenT) 我也是这样的问题，你这个解决了嘛？
> 
> 配置智能体的问题，重新检查一下配置

是哪个地方的配置？查了一遍看不出什么问题

shootao, 2025-07-17:
@wcu1117 可以直接使用 esp-adf 中默认的账号试一下