espressif/esp-adf #1378: S3-KORVO V3开发板运行ai_agent/volc_rtc例程，AI智能体经常没反应，问几遍才反应，回复有时也会断断续续

FlexLua, 2025-02-19:
感觉volc_rtc这套接入扣子ai语音对话的库运行不太稳定，AI智能体经常没反应，问几遍才反应，回复有时也会断断续续，我截取了一些看起来可能有助于扣子研发人员诊断问题的部分日志如下：

I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Feb 18 2025 22:26:40
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00300000
I (92) boot:  3 model            Unknown data     01 82 00310000 0040e000
I (100) boot:  4 spiffs_data      Unknown data     01 82 0071e000 00010000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c170020 size=60360h (394080) map
I (180) esp_image: segment 1: paddr=00070388 vaddr=3fc9f900 size=07edch ( 32476) load
I (186) esp_image: segment 2: paddr=0007826c vaddr=40378000 size=07dach ( 32172) load
I (193) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=164458h (1459288) map
I (416) esp_image: segment 4: paddr=001e4480 vaddr=4037fdac size=0fa88h ( 64136) load

I (6530) volc_rtc: start join room

1970-01-01 00:00:05.874 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
I (6690) CODEC_ELEMENT_HELPER: The element is 0x3c23f6a4. The reserve data 2 is 0x0.
1970-01-01 00:00:06.261 [E]  Cache.c:309 operation returned status code: 0x00000009
1970-01-01 00:00:06.266 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:00:06.268 [E]  LiteHttp.c:641 ID 1208224117  E_LOGIC : NO need keepAlive
1970-01-01 00:00:06.276 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:00:06.410 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:00:06.644 [E]  rx_net_delay_manager.c:1130 
I (7460) volc_rtc: join channel success CozeRTC_H264_G711A_7473119034368868xxx elapsed 192 ms now 192 ms

I (7460) volc_rtc: join room success

I (7460) volc_rtc: remote user joined  CozeRTC_H264_G711A_747311903436886xxx:7471196559376597xxx
I (7470) audio processor: Create g711a decoder
1970-01-01 00:00:06.666 [E]  EngineImplX.c:103 callback pEngineImplX->eventHandler.on_user_joined used too many times 13
I (7480) MODEL_LOADER: The storage free size is 22656 KB
I (7500) MODEL_LOADER: The partition size is 4152 KB
I (7510) MODEL_LOADER: Successfully load srmodels
I (7510) RECORDER_SR: The first wakenet model: wn9_hilexin

I (7520) AFE_SR: afe interface for speech recognition

I (7520) AFE_SR: AFE version: SR_V220727

I (7530) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (7540) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1

I (7540) AFE_SR: wakenet_init: 1

MC Quantized wakenet9: wakenet9_v1h24_嗨，乐鑫_3_0.608_0.615, tigger:v3, mode:2, p:0, (Sep  4 2024 11:49:31)
I (7810) AFE_SR: wake num: 3, mode: 0, (Sep  4 2024 11:49:31)

I (7810) main_task: Returned from app_main()
1970-01-01 00:00:07.111 [E]  rx_net_audio_jitterbuffer.c:1266 
1970-01-01 00:00:07.112 [E]  rx_net_audio_jitterbuffer.c:1190 
1970-01-01 00:00:07.113 [E]  rx_net_delay_manager.c:1130 
1970-01-01 00:00:07.117 [E]  rx_net_delay_manager.c:1130 
1970-01-01 00:00:07.122 [E]  Counter.c:90 AudioRecevied fps 0
1970-01-01 00:00:07.208 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:07.238 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
I (8230) wifi:<ba-add>idx:1 (ifx:0, 24:69:8e:f0:a9:4e), tid:6, ssn:0, winSize:64
1970-01-01 00:00:07.718 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
[... 123 lines cut ...]
0x42138d9d: __sfvwrite_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x4214a073: __sprint_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x42149a51: _vfprintf_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1774 (discriminator 1)
0x4213946d: vprintf at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x420214ca: defaultLogPrint at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/utils/Logger.c:72
0x42012dc1: __engineImplXOnAudioFrame at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/logic/Engine/EngineImplX.c:242 (discriminator 10)
0x420157b5: __mediaServiceOnReceiveAudioFrame at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/logic/Media/MediaService.c:142 (discriminator 2)
0x4201a457: networkLiteRecvAudioStreamOnFrameReady at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/rtc_kernel/PeerConnection/Rtp.c:51
0x4202c7ee: rx_net_audio_jitter_buffer_extract_packet at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/audio_jitter_buffer/rx_net_audio_jitterbuffer.c:905
0x4202c85a: rx_net_audio_jitter_buffer_get_out_data at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/audio_jitter_buffer/rx_net_audio_jitterbuffer.c:931
0x4202ce5e: rx_net_audio_jitter_buffer_may_extract_packet at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/audio_jitter_buffer/rx_net_audio_jitterbuffer.c:1244
0x4202ce92: rx_net_audio_jitter_buffer_may_extract_packet_in_100ms at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/audio_jitter_buffer/rx_net_audio_jitterbuffer.c:1257
0x42024dc9: rx_net_connection_audio_recv_streams_repeat_task at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/rx_net_connection.c:319
0x42024e0d: rx_net_connection_repeat_task_callback at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/network_lite/modules/rx_net_connection.c:333
0x4201edf1: rx_net_timer_callback_func at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/rtc_kernel/NetworkLiteWrapper/NetworkLiteTaskQueueAdapter.c:112 (discriminator 1)
0x4214e7a2: __onLiteTimerComelete at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/frameworks/LiteTimer.c:30
0x42023042: __liteEvLoopProcessTimer at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/frameworks/LiteEvLoopImpl.c:253
0x420238c3: liteEvLoopRun at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/frameworks/LiteEvLoopImpl.c:383
0x42034b68: __threadRoutine at C:\Espressif\frameworks\esp-idf-v5.3.1\esp-booka\build/C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/logic/EngineCtx/ThreadPool.c:29

_data used too many times 10
W (305720) volc_rtc: audio frame queue full
1970-01-01 00:05:04.909 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
W (305750) volc_rtc: audio frame queue full

shootao, 2025-02-22:
Hi @FlexLua 
1. 如果里的设备远的话 可以将 麦的增益改大一点， 在[这里](https://github.com/espressif/esp-adf/blob/a2ab6cefd8f9618a0470fd3f549c913fdf7ca1b8/examples/ai_agent/volc_rtc/components/audio_processor/audio_stream_dual_microphones.c#L80)加上   es7210_adc_set_gain(ES7210_INPUT_MIC1 | ES7210_INPUT_MIC2, GAIN_37_5DB);
2. 如果断续先看一下网络情况， 也可以看先用 火山临时的账号试一下

curl --location 'http://115.190.92.216:8899/startvoicechat' \
--header 'Content-Type: application/json' \
--header 'Authorization: af78e3067582ac8c04ba20174410bd1' \
--data '{"bot_id":"ep-20241104163951-sdmjc", "audio_codec":"G711A"}'

FlexLua, 2025-02-23:
> Hi [@FlexLua](https://github.com/FlexLua)
>
> 1. 如果里的设备远的话 可以将 麦的增益改大一点， [...]

谢谢，我试一下

NingZiXi, 2025-03-10:
> Hi [@FlexLua](https://github.com/FlexLua)
>
> 1. 如果里的设备远的话 可以将 麦的增益改大一点， [...]

按照你这个方法试过了，还是断断续续的我使用的硬件是box3

whosmyqueen, 2025-03-25:
@NingZiXi 我也是断断续续的， 怀疑是采样率问题， 你解决了吗