espressif/esp-adf #1380: esp32s3-box 3 运行volc_rtc无法沟通，疑似本地数据未上云

JlaYun, 2025-02-20:
[image]

JlaYun, 2025-02-21:
00:16:19.017 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:20.059 [E]  Counter.c:90 AudioRecevied fps 43
1970-01-01 00:16:21.001 [E]  Counter.c:90 AudioRecevied fps 57
1970-01-01 00:16:22.081 [E]  Counter.c:90 AudioRecevied fps 46
1970-01-01 00:16:23.005 [E]  Counter.c:90 AudioRecevied fps 54
1970-01-01 00:16:24.003 [E]  Counter.c:90 AudioRecevied fps 48
1970-01-01 00:16:25.006 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:26.009 [E]  Counter.c:90 AudioRecevied fps 52
1970-01-01 00:16:27.012 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:28.005 [E]  Counter.c:90 AudioRecevied fps 36
1970-01-01 00:16:29.181 [E]  Counter.c:90 AudioRecevied fps 57
1970-01-01 00:16:30.007 [E]  Counter.c:90 AudioRecevied fps 45
1970-01-01 00:16:31.003 [E]  Counter.c:90 AudioRecevied fps 52
1970-01-01 00:16:32.011 [E]  Counter.c:90 AudioRecevied fps 53
1970-01-01 00:16:33.005 [E]  Counter.c:90 AudioRecevied fps 46
1970-01-01 00:16:34.009 [E]  Counter.c:90 AudioRecevied fps 48
1970-01-01 00:16:35.013 [E]  Counter.c:90 AudioRecevied fps 55
1970-01-01 00:16:36.003 [E]  Counter.c:90 AudioRecevied fps 44
1970-01-01 00:16:37.000 [E]  Counter.c:90 AudioRecevied fps 55
1970-01-01 00:16:38.004 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:39.009 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:40.011 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:16:41.015 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:16:42.010 [E]  Counter.c:90 AudioRecevied fps 51
唤醒后的日志是这个，示例demo没动，唤醒有声音

shootao, 2025-02-22:
@JlaYun 
可以看一下 数据上行时的数据 [link](https://github.com/espressif/esp-adf/blob/a2ab6cefd8f9618a0470fd3f549c913fdf7ca1b8/examples/ai_agent/volc_rtc/main/volc_rtc.c#L345), 来确定一下数据没有正确发出去
也可以使用 火山的临时账号试一下
curl --location 'http://115.190.92.216:8899/startvoicechat' \
--header 'Content-Type: application/json' \
--header 'Authorization: af78e3067582ac8c04ba20174410bd1' \
--data '{"bot_id":"ep-20241104163951-sdmjc", "audio_codec":"G711A"}'

JlaYun, 2025-02-24:
@shootao 使用byte_rtc_send_audio_data返回值是0，按照api的描述是已经上发成功，但是豆包的同学却说没有音频上传，怎么查看设备端的音频采集和处理呢？设备应该没有问题，可以正常跑小智ai

aikoder, 2025-02-25:
这个例子使用box-3，有显示的么？我怎么跑历程没亮起来

JlaYun, 2025-02-25:
@aikoder 你是说屏幕吗？

JlaYun, 2025-02-25:
屏幕应该要开视频，会有一个豆包的虚拟形象

aikoder, 2025-02-25:
> [@aikoder](https://github.com/aikoder) 你的可以正常交互吗？

还没有，刚买回来这个box-3,正在跑这个例程，请问这个视频是如何开的呢？

NingZiXi, 2025-03-07:
我是可以运行但是一卡一卡的

shootao, 2025-04-18:
可以看一下log。
最新的 volc_rtc example 有在弱网下有优化，可以先试一下