espressif/esp-adf #1382: esp-adf examples advanced_examples algorithm示例无法得到ES7210第三个麦克风的数据

bakecool, 2025-02-22:
## Environment

- Audio development kit:     ESP32-S3-Korvo-2
- Audio kit version (ESP32-S3-Korvo-2): v3
- [Required] Module or chip used:  ESP32-S3-WROOM-1
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v5.3.1
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.7-51-g60054d26
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    esp-13.2.0_20240530
- Operating system:     Windows10
- (Windows only) Environment type: PowerShell
- Using an IDE?: Yes (vscode)
- Power supply:         USB

## Problem Description

打开algorithm调试模式后，获得的/sdcard/aec_in.wav双声道只有麦克风的声音，没有参考信号的声音。

### Expected Behavior
双声道数据是为了提供给algorithm算法进行AEC，只有具备第三个麦克风的参考信号，才能狗消除。

### Actual Behavior
缺失了第三个麦克风的参考信号

### Steps to Reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

shootao, 2025-02-24:
@bakecool 
algorithm example 默认配置的是32bit， 单通道， 你改一下 双通道 [I2S_CHANNEL_FMT_RIGHT_LEFT](https://github.com/espressif/esp-adf/blob/a2ab6cefd8f9618a0470fd3f549c913fdf7ca1b8/examples/advanced_examples/algorithm/main/algorithm_examples.c#L41)

bakecool, 2025-02-24:
非常感谢。我去试试

bakecool, 2025-02-24:
你有没有成功的.wave文件。我听一下。你说的这个方案，我试了。依然没有获得第三个麦克风的数据。

shootao, 2025-02-24:
@bakecool 
你有使能 DEBUG_ALGO_INPUT 么？ 如果使能了话， 直接去掉前面 44 字节， 直接看 16k，16bit， ch4 的pcm 数据看看

bakecool, 2025-02-24:
好的我再试试

bakecool, 2025-02-24:
我一直是使能debug-algo才会获得双通道wav的。试了把i2s改成双通道，但是掩码只能获取前16位信息。而第三通道无法获取。所以双通道之后获得的wav文件是mic1和mic2的声音数据。第三个麦克风的数据还是没有。

bakecool, 2025-02-24:
怎么才能读到第一通道的第二个slot我看adf和idf都没有相关代码。你有成功的wav文件吗？给我点儿信心让我知道确实能成功？

shootao, 2025-02-24:
目前，由于 es8311 不支持 tdm， 所以 对于 es7210 我们是将 32bit， 双通道的配置去读取 16bit， 四通道的

bakecool, 2025-02-24:
或者，像我的附件一样，有没有成功之后的aec_in.wav文件展示，让我看一下第二个通道（也就是第三个麦克风的声音数据），让我有一些信心和希望？

bakecool, 2025-02-27:
疑似硬件问题，我先关闭。如有后续我再补充。

closed 2025-02-27 by bakecool