espressif/esp-adf #1388: espadf 暂停后不能继续播放

skySj, 2025-02-28:
日志输出如下，不知道为什么暂停后点继续播放失败

//暂停成功输出日志
I (20970) AUDIO_PIPELINE: pause [file]  0x3c2f4288, state 0
I (20970) AUDIO_ELEMENT: [file]evt internal cmd = 4
I (20980) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (20990) AUDIO_ELEMENT: [file] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (21050) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (21050) AUDIO_ELEMENT: [131072] [file] audio_element_on_cmd run, state:3
I (21050) AUDIO_ELEMENT: [file] AEL_MSG_CMD_PAUSE Begin
I (21060) AUDIO_ELEMENT: [file] AEL_MSG_CMD_PAUSE
E (21060) AUDIO_ELEMENT: [file] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (21070) AUDIO_PIPELINE: pause [flac]  0x3c40fbb4, state 0
I (21080) AUDIO_ELEMENT: [flac]evt internal cmd = 4
I (21090) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (21100) AUDIO_ELEMENT: [flac] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (21120) AUDIO_EVT: xQueueReceive msg_source[1048579], on_cmd[1108739660] run
I (21120) AUDIO_EVT: audio_event_iface_cmd[1048579] success
W (21150) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (21150) AUDIO_ELEMENT: [131072] [flac] audio_element_on_cmd run, state:3
I (21160) AUDIO_ELEMENT: [flac] AEL_MSG_CMD_PAUSE Begin
I (21160) FLAC_DECODER: Closed
I (21170) AUDIO_ELEMENT: [flac] AEL_MSG_CMD_PAUSE
E (21170) AUDIO_ELEMENT: [flac] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (21180) AUDIO_PIPELINE: pause [filter]  0x3c40fa58, state 0
I (21190) AUDIO_ELEMENT: [filter]evt internal cmd = 4
I (21190) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (21200) AUDIO_ELEMENT: [filter] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (21210) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (21220) AUDIO_ELEMENT: [131072] [filter] audio_element_on_cmd run, state:3
I (21220) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_PAUSE Begin
I (21230) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_PAUSE
E (21240) AUDIO_ELEMENT: [filter] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (21240) AUDIO_PIPELINE: pause [i2s]  0x3c2f4544, state 0
I (21250) AUDIO_ELEMENT: [i2s]evt internal cmd = 4
I (21260) AUDIO_EVT: audio_event_iface_cmd[131072] success
W (21260) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (21270) AUDIO_ELEMENT: [131072] [i2s] audio_element_on_cmd run, state:3
I (21280) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE Begin
I (21280) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
E (21290) AUDIO_ELEMENT: [i2s] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
E (21300) AUDIO_ELEMENT: [i2s] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (21310) player: fatfs end state 4
I (21310) player: decoder end state 4
I (21310) player: rsp end state 4
I (21320) player: i2s end state 4

//暂停失败输出日志
I (100540) AUDIO_PIPELINE: pause [file]  0x3c2f4288, state 0
I (100540) AUDIO_ELEMENT: [file]evt internal cmd = 4
I (100550) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (100560) AUDIO_ELEMENT: [file] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (100580) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (100580) AUDIO_ELEMENT: [131072] [file] audio_element_on_cmd run, state:3
I (100580) AUDIO_ELEMENT: [file] AEL_MSG_CMD_PAUSE Begin
I (100590) AUDIO_ELEMENT: [file] AEL_MSG_CMD_PAUSE
E (100590) AUDIO_ELEMENT: [file] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (100600) AUDIO_PIPELINE: pause [flac]  0x3c40fbb4, state 0
I (100610) AUDIO_ELEMENT: [flac]evt internal cmd = 4
I (100610) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (100620) AUDIO_ELEMENT: [flac] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (101120) AUDIO_EVT: xQueueReceive msg_source[1048579], on_cmd[1108739660] run
I (101120) AUDIO_EVT: audio_event_iface_cmd[1048579] success
W (102120) AUDIO_EVT: xQueueReceive msg_source[1048579], on_cmd[1108739660] run
I (102120) AUDIO_EVT: audio_event_iface_cmd[1048579] success
E (102630) AUDIO_ELEMENT: [flac] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (102630) AUDIO_PIPELINE: pause [filter]  0x3c40fa58, state 0
I (102630) AUDIO_ELEMENT: [filter]evt internal cmd = 4
I (102640) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (102640) AUDIO_ELEMENT: [filter] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (103120) AUDIO_EVT: xQueueReceive msg_source[1048579], on_cmd[1108739660] run
I (103120) AUDIO_EVT: audio_event_iface_cmd[1048579] success
W (104120) AUDIO_EVT: xQueueReceive msg_source[1048579], on_cmd[1108739660] run
I (104120) AUDIO_EVT: audio_event_iface_cmd[1048579] success
E (104650) AUDIO_ELEMENT: [filter] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (104650) AUDIO_PIPELINE: pause [i2s]  0x3c2f4544, state 0
I (104650) AUDIO_ELEMENT: [i2s]evt internal cmd = 4
I (104650) AUDIO_EVT: audio_event_iface_cmd[131072] success
E (104660) AUDIO_ELEMENT: [i2s] Element xEventGroupWaitBits begin AUDIO_ELEMENT_PAUSE
W (104670) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (104680) AUDIO_ELEMENT: [131072] [i2s] audio_element_on_cmd run, state:3
I (104680) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE Begin
I (104690) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
E (104700) AUDIO_ELEMENT: [i2s] Element xEventGroupWaitBits end AUDIO_ELEMENT_PAUSE
I (104700) player: fatfs end state 4
I (104710) player: decoder end state 3
I (104710) player: rsp end state 3
I (104720) player: i2s end state 4

//暂停失败后点击继续播放输出日志，不能继续播放
I (127800) AUDIO_ELEMENT: [file]evt internal cmd = 5
I (127800) AUDIO_EVT: audio_event_iface_cmd[131072] success
W (127800) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (127810) AUDIO_ELEMENT: [131072] [file] audio_element_on_cmd run, state:4
I (127820) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:4
I (127820) AUDIO_ELEMENT: [i2s]evt internal cmd = 5
W (127830) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (127840) AUDIO_ELEMENT: [131072] [i2s] audio_element_on_cmd run, state:4
I (127840) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:4
I (127870) AUDIO_EVT: audio_event_iface_cmd[131072] success
I (127900) FATFS_STREAM: File size: 26459338 byte, file position: 3317760
W (127920) AUDIO_EVT: xQueueReceive msg_source[131072], on_cmd[1107845616] run
I (127920) AUDIO_ELEMENT: [131072] [filter] audio_element_on_cmd run, state:3
I (127930) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_PAUSE Begin
I (127930) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_PAUSE

shootao, 2025-02-28:
贴一下 你的 pause / resume 的代码看一下

skySj, 2025-02-28:
> 贴一下 你的 pause / resume 的代码看一下

//这是我的pause和resume的代码

//这是暂停和继续的函数，我在LVGL的UI界面中调用它们
void player_pause(player_handle_t player)
{
    if(audio_element_get_state(player->i2s_stream_writer) == AEL_STATE_RUNNING)
    {
        audio_pipeline_pause(player->pipeline);
    }
}

void player_resume(player_handle_t player)
{
    if(audio_element_get_state(player->i2s_stream_writer) == AEL_STATE_PAUSED)
    {
        audio_pipeline_resume(player->pipeline);
    }
}

//link顺序是file->decoder->filter->i2s

//调试发现
//函数audio_event_iface_cmd可以发送所有element暂停消息到队列中
//但函数audio_event_iface_waiting_cmd_msg 只读取到了file和i2s的暂停消息
esp_err_t audio_event_iface_waiting_cmd_msg(audio_event_iface_handle_t evt)
{
    audio_event_iface_msg_t msg;
    if (evt->internal_queue && (xQueueReceive(evt->internal_queue, (void *)&msg, evt->wait_time) == pdTRUE)) {
        if (evt->on_cmd) {
            ESP_LOGW(TAG, "xQueueReceive msg_source[%d] on_cmd[%d]", msg.source_type, evt->on_cmd);
            return evt->on_cmd((void *)&msg, evt->context);
        }
    }
    return ESP_OK;
}

esp_err_t audio_event_iface_cmd(audio_event_iface_handle_t evt, audio_event_iface_msg_t *msg)
{
    if (evt->internal_queue && (xQueueSend(evt->internal_queue, (void *)msg, 0) != pdPASS)) {
        ESP_LOGW(TAG, "There are no space to dispatch queue");
        return ESP_FAIL;
    }
    ESP_LOGW(TAG, "xQueueSend msg_source[%d] on_cmd[%d]", msg->source_type, evt->on_cmd);
    return ESP_OK;
}

skySj, 2025-02-28:
队列没有处理暂停消息，导致audio_element_pause函数等待超时，应该是这样子的，因为暂停失败的时候界面会卡2秒