espressif/esp-adf #1390: ESP-ADF组件中的AFE版本问题导致编译错误

hwngn, 2025-03-04:
使用的vscode中的esp-idf插件，安装的esp-idf版本是v5.2.5，ADF版本是直接从gitee上clone下来的，编译ADF示例工程的时候错误，提示：
D:/Expressif/esp-adf/components/audio_recorder/recorder_sr.c:62:45: error: 'ESP_AFE_SR_HANDLE' undeclared here (not in a function)
   62 | static const esp_afe_sr_iface_t *esp_afe = &ESP_AFE_SR_HANDLE;
      |                                             ^~~~~~~~~~~~~~~~~
D:/Expressif/esp-adf/components/audio_recorder/recorder_sr.c: In function 'recorder_sr_afe_result_convert':
D:/Expressif/esp-adf/components/audio_recorder/recorder_sr.c:117:43: error: comparison between 'vad_state_t' and 'enum <anonymous>' [-Werror=enum-compare]
  117 |                 if (afe_result->vad_state == AFE_VAD_SILENCE) {
      |                                           ^~
D:/Expressif/esp-adf/components/audio_recorder/recorder_sr.c:119:50: error: comparison between 'vad_state_t' and 'enum <anonymous>' [-Werror=enum-compare]
  119 |                 } else if (afe_result->vad_state == AFE_VAD_SPEECH) {
      |                                                  ^~
cc1.exe: some warnings being treated as errors

查询乐鑫官方手册里面说AFE版本从1迁移到2的过程中已经把ESP_AFE_SR_HANDLE 移除掉了，“ESP_AFE_SR_HANDLE and ESP_AFE_VC_HANDLE have been removed. Use esp_afe_handle_from_config to create instances:”

但是在recorder_sr.c文件里面依然使用了这个参数，造成编译失败，是不是因为没有同步更新导致的问题？如何解决？

hwngn, 2025-03-05:
已经解决，是更新子模块没有完全完成造成的

closed 2025-03-05 by hwngn

jackiebjtu, 2025-03-10:
您好，想问一下该怎么解决？

Faddy-man, 2025-03-13:
遇到了同样的问题

pr52, 2025-04-02:
你好，我也遇到了同样的问题，ESP-IDF为5.2.5，ESP-ADF、ESP-SR、esp-adf-ilb均升级到了最新的master版本，但是还是报错，请问你是用的什么版本的，怎么解决该问题的？

orangeagain, 2025-04-19:
我也发现这个错误，但在2.7上有编译错误，在最新版本可以了