espressif/esp-adf #1392: esp32s3+IDF v5.4+ADF运行报 RestrictedPinnedToCore的错误 (IDFGH-14780)

hwngn, 2025-03-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-KORVO-2 v3

### Power Supply used.

USB

### What is the expected behavior?

能够正常启动audio processor的功能

### What is the actual behavior?

IDF无论是用V5.4，V5.3还是V5.3.2都会报这个错误：
E (3366) gpio: gpio_install_isr_service(502): GPIO isr service already installed
E (3376) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v5.4_freertos.patch" first

E (3396) AUDIO_THREAD: Error creating RestrictedPinnedToCore audio_player_state_task
E (3396) DISPATCHER: exe first list: 0x0

然后按照错误提示去应用补丁文件也不行，提示信息如下：
PS D:\Expressif\v5.4\esp-idf> git apply $env:ADF_PATH/idf_patches/idf_v5.4_freertos.patch
error: patch failed: components/freertos/esp_additions/include/freertos/idf_additions.h:107
error: components/freertos/esp_additions/include/freertos/idf_additions.h: patch does not apply
error: patch failed: components/freertos/linker_common.lf:24
error: components/freertos/linker_common.lf: patch does not apply

### Steps to reproduce.

用的样例工程中的VOLC_RTC，自己没有添加任何代码
...

### Debug Logs.

```plain

```

DrWangCN, 2025-03-05:
> 升级补丁之后解决，但是还有下述错误： E (3367) gpio: gpio_install_isr_service(502): GPIO isr service already installed E (3377) DISPATCHER: exe first list: 0x0

请问下是怎么升级补丁的呢

shootao, 2025-03-06:
"E (3367) gpio: gpio_install_isr_service(502): GPIO isr service already installed
E (3377) DISPATCHER: exe first list: 0x0"
这个异常暂时客户忽略。

shootao, 2025-03-21:
在 IDF 里面打上这个 patch

closed 2025-04-21 by jason-mao