espressif/esp-adf #1393: wwe  和 pipeline_tts_stream 的问题

WhiteDoveBuct, 2025-03-05:
# 环境
- idf： v5.4
- adf: master
- chip:  esp32-p4

# 问题 2 
- esp-adf/examples/speech_recognition/wwe 编译失败：

[image]

# 问题 2
- esp-adf/examples/player/pipeline_tts_stream 运行失败：

[image]

WhiteDoveBuct, 2025-03-05:
在 pipeline_tts_stream的cmakelists.txt中加入下面这行代码，可以跑成功，但是并没有听到设备播放声音
```
set(embed_files /home/lzm/esp-adf/components/esp-sr/esp-tts/esp_tts_chinese/esp_tts_voice_data_xiaoxin_small.dat)
partition_table_get_partition_info(size "--partition-name voice_data" "size")
partition_table_get_partition_info(offset "--partition-name voice_data" "offset")

if("${size}" AND "${offset}")
    esptool_py_flash_to_partition(flash "voice_data" "${embed_files}")
else()
    set(message "Failed to find model in partition table file"
                "Please add a line(Name=model, Size>recommended size in log) to the partition file.")
endif()
```

WhiteDoveBuct, 2025-03-07:
上面没有声音的原因是：没有外接扬声器，现在外接扬声器报错：

[image]

hekdh, 2025-04-07:
尝试一下最新版本的adf

WhiteDoveBuct, 2025-04-07:
您好，我会尽快回复您的！谢谢！