espressif/esp-adf #1397: I2S clocks not running permanently

MKSounds, 2025-03-09:
Hi,

I am working on an audio project where a custom board should integrate the esp32 or esp32-S3 device.
I did some testing with the examples from the ADF, which was basically working.

I notized that the I2S clocks are only running if audio is played. That is certainly nonsense. 
If the output clock does not run continuously, it is almost impossible to attach different devices on the I2S bus, as the synchronization is lost and leads to clock errors with connected DSPs, DACs or whatever. You would need to reinit the attached devices everytime a stream stops or is paused.
Such a clock must therefore always run continuously. At least there must be a config option to set the I2S clocks to run permanently.

## Environment
- IDF version: 5.3.2, ADF version 2.7
- Operating system: Windows 11
- IDE: VS Code

TempoTian, 2025-03-10:
@MKSounds So you are working with i2s_stream? For I2S stream there is one option  `uninstall_drv` https://github.com/espressif/esp-adf/blob/master/components/audio_stream/i2s_stream_idf5.c#L618 to not uninstall I2S driver after stop. During our test, when use `es8311` to do playback, codec do not need to reconfig if I2S have no signal. If your special codec need i2s signal always output you can turn off the config and have a try.

TempoTian, 2025-03-13:
In current situation, I think you can init a fake `i2s_stream` firstly, this will do I2S clock setup and output. Then you can use pipeline or esp_audio with other i2s_stream inside of the pipeline with `uninstall_drv = false`, this may solve your concern.