espressif/esp-adf #1400: It is hoped that ESP-ADF can add programs to support PCM5101 drivers

Sail-211010, 2025-03-10:
Hello:

Since PCM5101 is used on several pieces of hardware here, it is hoped that PCM5101 driver can be added:

It is hoped that ESP-ADF can add programs to support the following operations:

1, speaker and microphone use two I2S (speaker one way, microphone one way, the use of pins can be modified in the example, without modifying the underlying file)

2. Since PCM5101 has no hardware volume adjustment interface, it is hoped that the volume adjustment function is not directly empty, and the volume adjustment can be achieved by adjusting the software gain

3.PCM5101 does not require an I2C interface

At present, it is mainly used on DLNA, so the compatibility can be based on DLNA sample test

At present, ADF only sees ESxxxx type number decoding chip, hoping to add some models

Looking forward to receiving relevant reply!

shootao, 2025-03-12:
PCM5101  driver can refer [es7148](https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es7148/es7148.c)

Sail-211010, 2025-03-13:
好的  我尝试下

Sail-211010, 2025-03-13:
希望可以预留一个GPIO修改的接口 ，否则只能通过修改底层文件实现

shootao, 2025-03-24:
我们有 [customer_board  ](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control/components/my_board) 结构去修改的

Sail-211010, 2025-03-24:
选择自定义板会编译失败，在安装的ADF中并没有自定义板的相关文件，需要由客户自行添加？

shootao, 2025-03-24:
客户 需要根据自己硬件去添加相应的 codec， sdcard 等外设，每个客户的硬件不一样。 所以可以参考 我们现在已有的板子

wcu1117, 2025-03-31:
> 客户 需要根据自己硬件去添加相应的 codec， sdcard 等外设，每个客户的硬件不一样。 所以可以参考 我们现在已有的板子

@shootao 你好，请问一个问题，我自定义的板子，能够使用示例里面的代码录音，但是接入豆包AI_AGETN这个例子的时候一直卡住录音这部分，另一个房间的用户收不到声音，能播放远端传过来的声音，而且使用这个例子的录音声音比较小，但是能听清。
examples\recorder\pipeline_raw_http
`
    ESP_LOGI(TAG, "[ 1 ] Initialize Button Peripheral & Connect to wifi network");
    // Initialize peripherals management
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);

    periph_wifi_cfg_t wifi_cfg = {
        .wifi_config.sta.ssid = CONFIG_WIFI_SSID,
        .wifi_config.sta.password = CONFIG_WIFI_PASSWORD,
    };
    esp_periph_handle_t wifi_handle = periph_wifi_init(&wifi_cfg);

    // Start wifi & button peripheral
    esp_periph_start(set, wifi_handle);
    periph_wifi_wait_for_connected(wifi_handle, portMAX_DELAY);

    ESP_LOGI(TAG, "[ 2 ] Start codec chip");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_ENCODE, AUDIO_HAL_CTRL_START);
     //
     audio_hal_set_volume(board_handle->audio_hal, 100);

    ESP_LOGI(TAG, "[3.0] Create audio pipeline for recording");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline);

    ESP_LOGI(TAG, "[3.1] Create http stream to post data to server");

    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.type = AUDIO_STREAM_WRITER;
    http_cfg.event_handle = _http_stream_event_handle;
    http_stream_writer = http_stream_init(&http_cfg);
    audio_element_set_uri(http_stream_writer, CONFIG_SERVER_URI);

    ESP_LOGI(TAG, "[3.2] Create i2s stream to read audio data from codec chip");
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_TYLE_AND_CH(CODEC_ADC_I2S_PORT, 44100, 16, AUDIO_STREAM_READER, 2);
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_cfg.out_rb_size = 16 * 1024; // Increase buffer to avoid missing data in bad network conditions
    i2s_cfg.volume = 100;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);

    ESP_LOGI(TAG, "[3.3] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline, i2s_stream_reader, "i2s");
    audio_pipeline_register(pipeline, http_stream_writer, "http");

    ESP_LOGI(TAG, "[3.4] Link it together [codec_chip]-->i2s_stream->http_stream-->[http_server]");
    const char *link_tag[2] = {"i2s", "http"};
    audio_pipeline_link(pipeline, &link_tag[0], 2);

    audio_pipeline_run(pipeline); // 立即开始录制
    
    // 等待5秒
    ESP_LOGI(TAG, "开始录制，持续5秒...");
    vTaskDelay(5000 / portTICK_PERIOD_MS); // 关键修改点：延时5秒
    
    // 停止并释放资源
    ESP_LOGI(TAG, "停止录制并退出");
    audio_pipeline_stop(pipeline);
    audio_pipeline_wait_for_stop(pipeline);
    audio_pipeline_terminate(pipeline);
    audio_pipeline_unregister(pipeline, http_stream_writer);
    audio_pipeline_unregister(pipeline, i2s_stream_reader);
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(http_stream_writer);
    audio_element_deinit(i2s_stream_reader);
`

shootao, 2025-04-18:
你可以先用默认的 volc rtc 先跑一下看看