espressif/esp-adf #1403: ESP S3 BOX 尝试跑通 VOLC RTC 语音交互例程 控制台输出一切正常，然而没有任何音视频输出

kramist, 2025-03-11:
1. BOX硬件一切正常 我烧录了其他DEMO示例，没有任何问题
2. 智能体和硬件终端都可以正常加入房间，ASR服务甚至有额度消耗
3. 我使用的是智能体（BotId）
4. 使用的G711A解码，已经开白，编译配置已经配置
5. StartVoiceChat（2024-12-01）请求数据如下：(应该没有错漏的地方)```JSON
{
  "query": {},
  "body": {
    "AppId": "67c2b59·········721fb",
    "RoomId": "G711A_·······_001",
    "TaskId": "taskid001",
    "Config": {
      "ASRConfig": {
        "Provider": "volcano",
        "ProviderParams": {
          "Mode": "smallmodel",
          "AppId": "993···12",
          "AccessToken": "nPXEP3s···axmgXsIdDml",
          "ApiResourceId": "volc.bigasr.sauc.duration",
          "Cluster": "volcengine_streaming_common",
          "StreamMode": 0
        }
      },
      "TTSConfig": {
        "Provider": "volcano_bidirection",
        "ProviderParams": {
          "app": {
            "appid": "8948···020",
            "token": "vpIHHqCYUn····FJFwivGm5KI",
            "cluster": "volcano_tts"
          }
        }
      },
      "LLMConfig": {
        "Mode": "ArkV3",
        "BotId": "bot-202·····04-···"
      }
    },
    "AgentConfig": {
      "TargetUserId": [
        "haversun"
      ],
      "WelcomeMessage": "你好，我是小语，很高兴认识你！",
      "UserId": "xiaoyuyuyuyuyu"
    }
  }
}

```

监控输出如下：
```bash
I (25840) audio_stream_8311: Create g711a decoder
I (25840) audio processor: Register all elements to playback pipeline
I (25850) audio processor: ENBALE_AUDIO_STREAM_DUAL_MIC
I (25850) audio processor: Link playback element together raw-->audio_decoder-->rsp-->i2s_stream-->[codec_chip]
I (25860) CODEC_ELEMENT_HELPER: The element is 0x3c2108e0. The reserve data 2 is 0x0.
I (25870) audio processor: player pipe start running
I (25870) wifi:<ba-add>idx:0 (ifx:0, 68:77:da:36:bb:cc), tid:5, ssn:11, winSize:64
I (25880) volc_rtc: start join room

1970-01-01 00:41:10.509 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
I (25890) wifi:<ba-add>idx:1 (ifx:0, 68:77:da:36:bb:cc), tid:0, ssn:1443, winSize:64
1970-01-01 00:41:11.422 [E]  Cache.c:309 operation returned status code: 0x00000009
1970-01-01 00:41:11.427 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:41:11.429 [E]  LiteHttp.c:641 ID 2108409576  E_LOGIC : NO need keepAlive
1970-01-01 00:41:11.438 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:41:11.560 [E]  RoomImplX.c:167 operation returned status code: 0x52000057
1970-01-01 00:41:12.014 [E]  rx_net_delay_manager.c:1130 
I (27400) volc_rtc: join channel success G711A_roomid_001 elapsed 395 ms now 395 ms

I (27400) volc_rtc: join room success

I (27400) volc_rtc: remote user joined  G711A_roomid_001:xiaoyuyuyuyuyu
I (27400) MODEL_LOADER: The storage free size is 22848 KB
I (27420) MODEL_LOADER: The partition size is 4152 KB
1970-01-01 00:41:12.029 [E]  EngineImplX.c:103 callback pEngineImplX->eventHandler.on_user_joined used too many times 6
I (27420) MODEL_LOADER: Successfully load srmodels
I (27440) RECORDER_SR: The first wakenet model: wn9_hilexin

I (27440) AFE_SR: afe interface for speech recognition

I (27450) AFE_SR: AFE version: SR_V220727

I (27450) AFE_SR: Initial auido front-end, total channel: 2, mic num: 1, ref num: 1

I (27460) AFE_SR: aec_init: 1, se_init: 0, vad_init: 0

I (27470) AFE_SR: wakenet_init: 0

I (27620) AFE_SR: wake num: 1, mode: 0, (Sep  4 2024 11:49:31)

I (27620) AUDIO_RECORDER: RECORDER_CMD_TRIGGER_START
1970-01-01 00:41:12.282 [E]  rx_net_audio_jitterbuffer.c:1266 
I (27670) main_task: Returned from app_main()
1970-01-01 00:41:12.291 [E]  rx_net_audio_jitterbuffer.c:1190
1970-01-01 00:41:12.292 [E]  rx_net_delay_manager.c:1130
1970-01-01 00:41:12.292 [E]  rx_net_delay_manager.c:1130
1970-01-01 00:41:12.297 [E]  Counter.c:90 AudioRecevied fps 0
1970-01-01 00:41:13.019 [E]  Counter.c:90 AudioRecevied fps 35
1970-01-01 00:41:14.020 [E]  Counter.c:90 AudioRecevied fps 54
1970-01-01 00:41:15.010 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:16.038 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:41:17.018 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:41:18.003 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:19.003 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:20.014 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:21.018 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:22.000 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:23.019 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:24.008 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:25.031 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:41:26.018 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:41:27.018 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:41:28.002 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:41:29.019 [E]  Counter.c:90 AudioRecevied fps 51
```

shootao, 2025-03-11:
我看你的编码 格式 是 G711A的， 你的 appid 有加白嘛 ？

kramist, 2025-03-11:
> 我看你的编码 格式 是 G711A的， 你的 appid 有加白嘛 ？

有的

shootao, 2025-03-12:
第一次进入房间的欢迎提示语也没有的吗 ？

wcu1117, 2025-03-13:
我也遇到了同样的问题，你解决了吗

zhoujian-github1, 2025-03-26:
有解决么，我也碰到。。。。不出声了

guozi7788, 2025-04-18:
> 我看你的编码 格式 是 G711A的， 你的 appid 有加白嘛 ？

请教下这里所说的在 appid加白是指什么？如何操作（有示例最好）？
3Q

shootao, 2025-04-18:
@guozi7788 你现在用最新的 ADF试试， 里面有测试账号
如果需要 G711 的需要找火山的同事， 建议使用 OPUS 的