espressif/esp-adf #1404: There is a "pop" noise when adjusting the volume

GCTechnology, 2025-03-12:
## Environment

- Module or chip used:  `ESP32-S3-WROOM-1U`
- Codec used: `ES8388`
- IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): `v5.3.2`
- ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): `v2.7-55-g29de65cd`
- Build system: `idf.py`
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): `xtensa-esp-elf-gcc.exe (crosstool-NG esp-13.2.0_20240530) 13.2.0`
- Operating system: `Windows`
- Using an IDE?: `VScode ESP-IDF extension`
- Power supply: `USB / external 5V`

## Problem Description
当我调节音量的时候，每次调节都会伴随着“噼啪”的噪声，无论调节幅度或者频率，音量越大噪声越响。
已使用 `pipeline_play_sdcard_music `和 `play_mp3_control `测试过，都存在该问题。且尝试改变过调节幅度和频率，均存在该问题。已在三种硬件上尝试，都存在问题。
请问是否有能够解决或规避该问题的方案？

When I adjust the volume, every time I do so, there will be a "pop" noise accompanying it. No matter the adjustment range or frequency, the louder the volume, the louder the noise will be.
The problem has been tested with `pipeline_play_sdcard_music `and `play_mp3_control`, and it exists in both cases. Moreover, attempts have been made to adjust the amplitude and frequency, but the problem persists in all cases.
If there is any solution that can solve or circumvent this problem?

### Steps to Reproduce
1. Flash the program.
2. Adjust volume using `audio_hal_set_volume` API.
3. Here comes the problem.

## Hardware schematic diagram

[image]

[image]

## Debug Logs
No error logs. All the code works fine.

shootao, 2025-03-12:
如果规避的话，是不是可以 先 mute， 然后设置 音量， 然后在 unmute 再试一下

GCTechnology, 2025-03-13:
> 如果规避的话，是不是可以 先 mute， 然后设置 音量， 然后在 unmute 再试一下

尝试了一下mute的方案，即使用 `audio_hal_set_mute(board_handle->audio_hal, true);` 接口，但依然存在问题。

TempoTian, 2025-03-13:
确认一下面的寄存器看看是什么值，把斜率设置为最小看看
[image]

GCTechnology, 2025-03-14:
> 确认一下面的寄存器看看是什么值，把斜率设置为最小看看 [image]

ADC值为默认值 00
[image]

TempoTian, 2025-03-17:
设置为11100000 = 0xE0看看

TempoTian, 2025-03-19:
如果调节硬件codec无法满足要求可以加一个软件的volume 模块，你可以把硬件寄存器设置为最大音量然后前面加个alc模块来实现软件的音量调节。

GCTechnology, 2025-03-20:
> 如果调节硬件codec无法满足要求可以加一个软件的volume 模块，你可以把硬件寄存器设置为最大音量然后前面加个alc模块来实现软件的音量调节。

请问如何实现这样的调节音量模块，能具体展开说一下吗

TempoTian, 2025-03-20:
如果你用i2s_stream, i2s_stream是带alc功能的，设置函数 ·int i2s_alc_volume_set(audio_element_handle_t i2s_stream, int volume)` 注意这个volume设置的是db值。
https://github.com/espressif/esp-adf/blob/master/components/audio_stream/include/i2s_stream.h#L69 `use_alc`开启这个功能。如果自己实现可以参考里面的代码

GCTechnology, 2025-03-23:
我联系了ES8388 codec的芯片厂商，移植了他们提供的codec驱动但仍未解决问题。

sourxsunny, 2025-03-24:
我用es8311，也碰到了同样的问题