espressif/esp-adf #1408: 我在运行ai_agent/volc_rtc例程的时候，在和RTC的智能体通话的过程中，如何播放tips的提示词音频？

Freak-luo, 2025-03-13:
我想在运行RTC的时候，从SD卡再插播一条简短的.mp3系统提示音，例如网络状态不佳的这种语音播放。这种应该怎么做呢？不能够重新挂掉和RTC的链接再重新进房间吧？我写了一个暂停RTC的pipeline流，然后创建播放tips的管道，但是这时候没有声音播放出。如果我是单独播放tips的管道就有声音播出。

[image]

[image]

请各位大佬指点一番

wuliaodejijiji, 2025-03-13:
大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下

Freak-luo, 2025-03-13:
> 大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下

不卡 就是官方案例  一样的

wuliaodejijiji, 2025-03-13:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

wuliaodejijiji, 2025-03-13:
> > 大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下
> 
> 不卡 就是官方案例 一样的

能发一份给我不，我也是用官方示例的，现在卡了还久了，一直解决不了，

wuliaodejijiji, 2025-03-13:
> > 大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下
> 
> 不卡 就是官方案例 一样的
我IDF使用的5.4D，跪求大佬demo

Freak-luo, 2025-03-13:
> > > 大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下
> > 
> > 
> > 不卡 就是官方案例 一样的
> > 我IDF使用的5.4D，跪求大佬demo

你试试5.3吧 代码都是一样的

shootao, 2025-03-13:
[0001-example-add-cache-buffer-for-volc-rtc-example.patch](https://github.com/user-attachments/files/19225752/0001-example-add-cache-buffer-for-volc-rtc-example.patch)

@Freak-luo 用这个 patch 试一下， 里面有 测试账号
提示音在 唤醒模式有的， 可以参考一下

Freak-luo, 2025-03-13:
> [0001-example-add-cache-buffer-for-volc-rtc-example.patch](https://github.com/user-attachments/files/19225752/0001-example-add-cache-buffer-for-volc-rtc-example.patch)
> 
> [@Freak-luo](https://github.com/Freak-luo) 用这个 patch 试一下， 里面有 测试账号 提示音在 唤醒模式有的， 可以参考一下

这个好像不可以在和智能体聊天的时候播放提示音吧？

wuliaodejijiji, 2025-03-13:
> > > > 大佬，你那个跟智能体对话，卡不卡，能不能提供你的demo参考下
> > > 
> > > 
> > > 不卡 就是官方案例 一样的
> > > 我IDF使用的5.4D，跪求大佬demo
> 
> 你试试5.3吧 代码都是一样的

你是用5.3 还是5.3.1 还是5.3.2  我用5.3.1也存在

shootao, 2025-03-24:
你播放提示音是 打断 提示音还是两路 mixer 呢

Freak-luo, 2025-03-24:
> 你播放提示音是 打断 提示音还是两路 mixer 呢

打断的方式的

Freak-luo, 2025-03-24:
> 那直接 用 player_pipeline_stop， 然后去播放 你的 tips， 播放完 之后在 调 player_pipeline_run

这个我已经是可以了  我想请教一下如果是mixer的话应该要如何做呢？

shootao, 2025-03-24:
可以参考 [mixer](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/audio_mixer_tone) 去实现

PoohWoah, 2025-04-02:
你们有遇到过自问自答的现象吗

wuliaodejijiji, 2025-04-02:
> 你们有遇到过自问自答的现象吗
我遇到过，是AES问题，你需要优化

PoohWoah, 2025-04-02:
> > 你们有遇到过自问自答的现象吗
> > 我遇到过，是AES问题，你需要优化

AES我知道 但是我这已经优化了很久了  所有配置都试过了，你目前优化的如何呢？

wuliaodejijiji, 2025-04-02:
> > > 你们有遇到过自问自答的现象吗
> > > 我遇到过，是AES问题，你需要优化
> 
> AES我知道 但是我这已经优化了很久了 所有配置都试过了，你目前优化的如何呢？

加QQ邮箱，446547155 我私聊你

PoohWoah, 2025-04-02:
> > > > 你们有遇到过自问自答的现象吗
> > > > 我遇到过，是AES问题，你需要优化
> > 
> > 
> > AES我知道 但是我这已经优化了很久了 所有配置都试过了，你目前优化的如何呢？
> 
> 加QQ邮箱，446547155 我私聊你

这是qq吧，我已经加了

FlexLua, 2025-04-06:
同问，遇到自问自答问题，如何优化AEC？

PoohWoah, 2025-04-18:
> 同问，遇到自问自答问题，如何优化AEC？

调整结构和硬件

MJPbruce, 2025-05-27:
> 同问，遇到自问自答问题，如何优化AEC？

> > > 你们有遇到过自问自答的现象吗
> > > 我遇到过，是AES问题，你需要优化
> 
> AES我知道 但是我这已经优化了很久了 所有配置都试过了，你目前优化的如何呢？

AEC需要怎么优化呢，是需要改软件吗，乐鑫的FAE说是不用修改软件了啊