espressif/esp-adf #1409: 运行ai_agent/volc_rtc 使用的IDF版本为5.3.2，使用的开发板为ESP32-S3-korvo-2 v3 位开启智能体对话后，声音一卡一卡的。怎么解决，请看日志

wuliaodejijiji, 2025-03-13:
I (4622) volc_rtc: join room success

I (4622) volc_rtc: remote user joined  mx_yjwl:mx_esp32s3
I (4622) RAW_OPUS_ENC: Raw Opus encoder init
1970-01-01 00:00:03.814 [E]  EngineImplX.c:103 callback pEngineImplX->eventHandler.on_user_joined used too many times 6
I (4632) MODEL_LOADER: The storage free size is 22720 KB
I (4652) MODEL_LOADER: The partition size is 4152 KB
I (4652) MODEL_LOADER: Successfully load srmodels
I (4662) RECORDER_SR: The first wakenet model: wn9_xiaoaitongxue

I (4672) AFE_SR: afe interface for speech recognition

I (4672) AFE_SR: AFE version: SR_V220727

I (4682) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (4682) AFE_SR: aec_init: 1, se_init: 0, vad_init: 0

I (4692) AFE_SR: wakenet_init: 0

I (4852) AFE_SR: wake num: 2, mode: 0, (Sep  4 2024 11:49:31)

I (4852) AUDIO_RECORDER: RECORDER_CMD_TRIGGER_START
I (4852) main_task: Returned from app_main()
1970-01-01 00:00:04.088 [E]  rx_net_audio_jitterbuffer.c:1266 
1970-01-01 00:00:04.107 [E]  rx_net_audio_jitterbuffer.c:1190 
1970-01-01 00:00:04.107 [E]  rx_net_delay_manager.c:1130
1970-01-01 00:00:04.108 [E]  rx_net_delay_manager.c:1130
1970-01-01 00:00:04.112 [E]  Counter.c:90 AudioRecevied fps 0
1970-01-01 00:00:04.128 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 10
1970-01-01 00:00:04.227 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:04.378 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:04.508 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:04.568 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:00:04.596 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:04.928 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:05.021 [E]  Counter.c:90 AudioRecevied fps 39
1970-01-01 00:00:05.276 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:05.626 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:05.688 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:05.837 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:05.886 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:05.988 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:00:06.003 [E]  Counter.c:90 AudioRecevied fps 50
1970-01-01 00:00:06.116 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:06.195 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:06.286 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:06.637 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:00:06.838 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:00:06.985 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:07.007 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:00:07.125 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:07.438 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:00:07.598 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:08.000 [E]  Counter.c:90 AudioRecevied fps 55
1970-01-01 00:00:08.408 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:08.758 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:09.012 [E]  Counter.c:90 AudioRecevied fps 51
1970-01-01 00:00:09.048 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:09.153 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:00:09.288 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:00:09.428 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:00:09.545 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:00:10.003 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:00:10.128 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:00:10.307 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:00:10.558 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:10.758 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:00:10.928 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:00:10.985 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
[... 404 lines cut ...]
1970-01-01 00:01:13.002 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:13.018 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 9
1970-01-01 00:01:13.269 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:13.468 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:13.518 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:13.558 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:13.666 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:13.908 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:14.017 [E]  Counter.c:90 AudioRecevied fps 45
1970-01-01 00:01:14.083 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:14.597 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:14.676 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:14.828 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:15.020 [E]  Counter.c:90 AudioRecevied fps 53
1970-01-01 00:01:15.025 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 4
1970-01-01 00:01:15.184 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:15.297 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:15.576 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:16.000 [E]  Counter.c:90 AudioRecevied fps 49
1970-01-01 00:01:16.138 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 7
1970-01-01 00:01:16.288 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6
1970-01-01 00:01:16.448 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 5
1970-01-01 00:01:16.688 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 8
1970-01-01 00:01:16.798 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 10
1970-01-01 00:01:16.844 [E]  EngineImplX.c:242 callback pEngineImplX->eventHandler.on_audio_data used too many times 6

wuliaodejijiji, 2025-03-18:
> [@wuliaodejijiji](https://github.com/wuliaodejijiji) 我用esp32-s3-3编译运行一直报`Guru Meditation Error: Core / panic'ed (Cache disabled but cached memory region accessed)` ，大佬知道什么原因吗
      我已经调通了，你这个感觉内存越界或者啥的。

wuliaodejijiji, 2025-03-18:
> [@wuliaodejijiji](https://github.com/wuliaodejijiji) 我用esp32-s3-3编译运行一直报`Guru Meditation Error: Core / panic'ed (Cache disabled but cached memory region accessed)` ，大佬知道什么原因吗

具体问题，可以私聊，我帮你分析下

whosmyqueen, 2025-03-21:
@wuliaodejijiji  现在可以了， 但是会一卡一卡的， 我对接的是声网， 不知道咋回事

closed 2025-03-22 by wuliaodejijiji