espressif/esp-adf #1410: VoIP example FreeSwitch Microsip error

ok-home, 2025-03-16:
## Environment

- Audio development kit:     ESP32-LyraT-Mini V1,2
- IDF version ): 5.2.1
- ADF version : master
- Operating system:     wsl2
- Using an IDE: vscode
- Power supply:         USB

## Problem Description

the call goes both ways: Microsip->LyraT, LyraT->Microsip
the microphone sound goes only in the direction Microsip->LyraT
the microphone sound does not go in the direction LyraT->Microsip
example without changes [https://github.com/espressif/esp-adf/tree/master/examples/protocols/voip](url)
the microphone on the board works, verified by the example [https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_raw_http](url)

TempoTian, 2025-03-17:
@ok-home Could you please attach the serial log? So that I can analysis the possible causes for you.

ok-home, 2025-03-17:
I apologize for the delay

[logs.txt](https://github.com/user-attachments/files/19286036/logs.txt)

TempoTian, 2025-03-18:
From the log see that VOIP overall flow is OK. I think it is special related to MINI board, test use esp32s3 korvo2 is OK. I will test use mini board later, and notify to you if any findings.

ok-home, 2025-03-18:
> From the log see that VOIP overall flow is OK. I think it is special related to MINI board, test use esp32s3 korvo2 is OK. I will test use mini board later, and notify to you if any findings.

Thanks in advance ))

nosmartLin, 2025-07-02:
您好，我已经收到您的邮件，谢谢。