espressif/esp-adf #1412: ESP32S3的adf播放管道（"player_raw", "player_dec", "player_rsp"）,出现无法播放服务器发来的最后一帧音频数据

PoohWoah, 2025-03-25:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, start similar to "How do I..". If it is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [ ] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps above, please follow these instructions so we can get the needed information to help you quickly and effectively.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your issues.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other Items If Possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the `Preview` tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: Please follow the above instructions and provide as many details as possible. It will save a lot of communication time and improve the efficiency of problem solving. The more details you provide, the faster we may be able to reproduce and resolve the issue. Thanks!**

----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      [ESP32-LyraT|ESP32-LyraT-Mini|ESP32-LyraTD-MSC|ESP32-S2-Kaluga-1|ESP32-Korvo-DU1906|ESP32-S3-Korvo-2|ESP32-S3-Korvo-2-LCD|none]
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): [v1|v2|v3|v4]
- [Required] Module or chip used:  [ESP32-WROOM-32E|ESP32-WROVER-E|ESP32-S2-WROVER|ESP32-S3-WROOM-1]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v4.4-191-g83daa6dabb
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.3-171-gaac5912
- Build system:         [Make|CMake|idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     [Windows|Linux|macOS]
- (Windows only) Environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell]
- Using an IDE?: [No|Yes (please give details)]
- Power supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

// Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

PoohWoah, 2025-03-25:
在我从阿里云服务器进行语音合成接口的测试时，我发现服务器发来的数据永远播放不了最后一帧
随后我进行了测试，挂载了usb host，进行文件下载，可以完整识别到的，不会丢失最后一帧数据。那么说明问题出在播放管道解码过程。
服务器发送数据如下：
发送了1008bytes数据*n个数据包，最后一包数据位879bytes（丢失的数据包）。

具体配置如下：
`player_pipeline_handle_t player_pipeline_open(void) 
{
    ESP_LOGI(TAG, "%s", __func__);
    player_pipeline_handle_t player_pipeline = audio_calloc(1, sizeof(struct player_pipeline_t));
    AUDIO_MEM_CHECK(TAG, player_pipeline, goto _exit_open);
    player_pipeline->player_state = PIPE_STATE_IDLE;
    s_player_pipeline = player_pipeline;

    ESP_LOGI(TAG, "Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    player_pipeline->audio_pipeline = audio_pipeline_init(&pipeline_cfg);
    AUDIO_MEM_CHECK(TAG, player_pipeline, goto _exit_open);

    ESP_LOGI(TAG, "Create playback audio stream");
    player_pipeline->raw_writer = create_player_raw_stream();
    AUDIO_MEM_CHECK(TAG, player_pipeline->raw_writer, goto _exit_open);
    player_pipeline->audio_decoder = create_player_decoder_stream();
    AUDIO_MEM_CHECK(TAG, player_pipeline->audio_decoder, goto _exit_open);
    player_pipeline->i2s_stream_writer = create_player_i2s_stream(false);
    AUDIO_MEM_CHECK(TAG, player_pipeline->i2s_stream_writer, goto _exit_open);

    ESP_LOGI(TAG, "Register all elements to playback pipeline");
    audio_pipeline_register(player_pipeline->audio_pipeline, player_pipeline->raw_writer, "player_raw");
    audio_pipeline_register(player_pipeline->audio_pipeline, player_pipeline->audio_decoder, "player_dec");

#if (defined ENBALE_AUDIO_STREAM_DUAL_MIC)
    ESP_LOGI(TAG, "ENBALE_AUDIO_STREAM_DUAL_MIC");
#if CONFIG_AUDIO_SUPPORT_G711A_DECODER
    player_pipeline->player_rsp = create_8k_ch1_to_16k_ch2_rsp_stream();
#else
    player_pipeline->player_rsp = create_ch1_to_ch2_rsp_stream();   
#endif
#elif (defined ENABLE_AUDIO_STREAM_SINGLE_MIC)
    ESP_LOGI(TAG, "ENBALE_AUDIO_STREAM_DUAL_MIC");
#if CONFIG_AUDIO_SUPPORT_G711A_DECODER
    player_pipeline->player_rsp = create_8k_ch1_to_16k_ch2_rsp_stream();
#else
    player_pipeline->player_rsp = create_ch1_to_ch2_rsp_stream();   
#endif 
#else
    ESP_LOGE(TAG, "Invalid record device"); 
#endif // USE_ES8311_AS_RECORD_DEVICE
    AUDIO_MEM_CHECK(TAG, player_pipeline->player_rsp, goto _exit_open);

    ESP_LOGI(TAG, "Link playback element together raw-->audio_decoder-->rsp-->i2s_stream-->[codec_chip]");

    audio_element_set_write_cb(player_pipeline->player_rsp, _player_write_nop_cb, NULL);
    audio_pipeline_register(player_pipeline->audio_pipeline, player_pipeline->player_rsp, "player_rsp");
    const char *link_tag[3] = {"player_raw", "player_dec", "player_rsp"};
    audio_pipeline_link(player_pipeline->audio_pipeline, &link_tag[0], 3);

    audio_pipeline_run(player_pipeline->audio_pipeline);
    return player_pipeline;

_exit_open:
    audio_pipe_safe_free(player_pipeline->player_rsp, audio_element_deinit);
    audio_pipe_safe_free(player_pipeline->raw_writer, audio_element_deinit);
    audio_pipe_safe_free(player_pipeline->audio_decoder, audio_element_deinit);
    audio_pipe_safe_free(player_pipeline->i2s_stream_writer, audio_element_deinit);
    audio_pipe_safe_free(player_pipeline->audio_pipeline, audio_pipeline_deinit);
    audio_pipe_safe_free(player_pipeline, audio_free);
    return NULL;
}`
服务器下发下来的数据写入过程：
`            memcpy(frame.frame_ptr, data->data_ptr, data->data_len);
          frame.frame_len = data->data_len;
            ESP_LOGI(TAG, "Copied audio frame: len=%d", frame.frame_len);

            // 写入音频数据
             raw_stream_write(player_pipeline_get_raw_write(s_aiha_demo.player_pipeline), frame.frame_ptr, frame.frame_len);

            // 释放音频帧内存
             audio_free(frame.frame_ptr);
            frame.frame_ptr = NULL;`

我很确定的是没有播放最后一帧数据，当我在写入过程里面跳过了最后一帧数据，提前结束文件音频写入，播放出来和我MP3管道效果一致，是一定要固定长度才可以播放吗？1008bytes≠897bytes

PoohWoah, 2025-03-25:
是MP3的方式下行的，
audio_element_handle_t create_player_decoder_stream(void)
{
    audio_element_handle_t decoder_stream = NULL;
    ESP_LOGI(TAG, "Create aac decoder");
    mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_dec_cfg.task_core = 1;
    decoder_stream = mp3_decoder_init(&mp3_dec_cfg);

    return decoder_stream;
}
这是我的配置，没有任何更改

PoohWoah, 2025-03-25:
> 是MP3的方式下行的， ```
audio_element_handle_t create_player_decoder_stream(void) { audio_element_handle_t decoder_stream = NULL; ESP_LOGI(TAG, "Create aac decoder"); mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG(); mp3_dec_cfg.task_core = 1; decoder_stream = mp3_decoder_init(&mp3_dec_cfg);
> 
> ```
> return decoder_stream;
> ```
> 
```
> } 这是我的配置，没有任何更改

shootao, 2025-03-28:
MP3 的话可以直接 写 就可以了， 不需要 加 frame length 的

PoohWoah, 2025-04-02:
好的

suicideking000, 2025-04-15:
您好,我也遇到了这个问题,请问您解决了吗