espressif/esp-adf #1416: comments in esp_aac_enc_config_t ambigue

morgana2313, 2025-03-28:
The comments in `esp_aac_enc_config_t` may lead one to think the parameter `bitrate` for a 8000Hz `sample_rate` should be between `12 .. 48`. However the correct `bitrate` is `12000 .. 48000`:

This ambiguity should be avoided:

```diff
/**
 * @brief  AAC Encoder configurations
 */
typedef struct {
    int sample_rate;     /*!< Support sample rate(Hz) : 96000, 88200, 64000, 48000,
                              44100, 32000, 24000, 22050, 16000, 12000, 11025, 8000 */
    int channel;         /*!< Support channel : mono, dual */
    int bits_per_sample; /*!< Support bits per sample : 16 bit */
    int bitrate;         /*!< Support bitrate(bps) range on mono stream :
                              |samplerate(Hz)|bitrate range(Kbps)|
-                              |    8000      |    12 - 48        |
+                              |    8000      |    12000 - 48000  |
                              |    11025     |    18 - 66        |
                              |    12000     |    20 - 72        |
                              |    16000     |    22 - 96        |
                              |    22050     |    25 - 132       |
                              |    24000     |    31 - 144       |
                              |    32000     |    33 - 160       |
                              |    44100     |    57 - 160       |
                              |    48000     |    59 - 160       |
                              |    64000     |    65 - 160       |
                              |    88200     |    67 - 160       |
                              |    96000     |    70 - 160       | 
                              Note : 1) This table shows the bitrate range corresponding to each samplerate.
                                     2) The bitrate range of dual stream is twice that of mono. */
    bool adts_used;      /*!< Whether write ADTS header: true - add ADTS header, false - raw aac data only */
} esp_aac_enc_config_t;
```

hekdh, 2025-04-03:
the unit of bitrate range is Kbps

morgana2313, 2025-04-03:
In my experience the functions expect `int bitrate` values of 12000 .. 48000. With 12..48 they fail (silently!) to work. 

Therefore I think the unit of this parameter is `bps` and not `Kbps` as the comments suggest.

hekdh, 2025-04-03:
you are right, the actual unit is bps. the correct commet is  in another file `components/esp-adf-libs/esp_codec/include/codec/aac_encoder.h`
```
typedef struct {
    int sample_rate;     /*!< Support sample rate(Hz) : 96000, 88200, 64000, 48000,
                              44100, 32000, 24000, 22050, 16000, 12000, 11025, 8000 */
    int channel;         /*!< Support channel : mono, dual */
    int bit;             /*!< Support bit per sample : 16 bit */
    int bitrate;         /*!< Support bitrate(bps) range on mono stream :
                              |samplerate(Hz)|bitrate range(bps)|
                              |    8000      |    12k - 48k     |
                              |    11025     |    18k - 66k     |
                              |    12000     |    20k - 72k     |
                              |    16000     |    22k - 96k     |       
                              |    22050     |    25k - 132k    |
                              |    24000     |    31k - 144k    |
                              |    32000     |    33k - 160k    |
                              |    44100     |    57k - 160k    |
                              |    48000     |    59k - 160k    |
                              |    64000     |    65k - 160k    |
                              |    88200     |    67k - 160k    |
                              |    96000     |    70k - 160k    | 
                              Note : 1) This table shows the bitrate range corresponding to each samplerate.
                                     2) The bitrate range of dual stream is twice that of mono. */
    int out_rb_size;     /*!< Size of output ringbuffer */
    int task_stack;      /*!< Task stack size */
    int task_core;       /*!< Task running in core (0 or 1) */
    int task_prio;       /*!< Task priority (based on freeRTOS priority) */
    bool stack_in_ext;   /*!< Try to allocate stack in external memory */
} aac_encoder_cfg_t;

```

> In my experience the AAC encoder functions expect `int bitrate` values of 12000 .. 48000. With 12..48 they fail (silently!) to work.
> 
> Therefore I think the unit of this parameter is `bps` and not `Kbps` as the comments suggest.

morgana2313, 2025-04-03:
Even here I would suggest that replacing the 'k' with '000' helps to avoid time consuming mistakes.