espressif/esp-adf #1417: Add Peer Call Answer Notification Event in ESP-ADF RTC

carlosrodridel, 2025-03-28:
**Is your feature request related to a problem? Please describe.**

Currently, when initiating an `esp_rtc_call` in ESP-ADF, we receive notifications for `ESP_RTC_EVENT_CALLING` and `ESP_RTC_EVENT_AUDIO_SESSION_BEGIN`. However, there is no event that explicitly notifies when the peer picks up the call (when the remote party answers).

This limitation makes it difficult to track call state transitions accurately, particularly for use cases where developers need to trigger specific actions upon call acceptance, such as logging call status or  starting timers.

**Describe the solution you'd like**

I would like ESP-ADF to include an event that is triggered when the remote peer picks up the call. This could be something like:

```c
ESP_RTC_EVENT_CALL_ANSWERED
```

This event should be fired when the `SIP 200 OK` response is received after sending an INVITE, indicating that the call has been accepted.

**Describe alternatives you've considered**

An alternative approach could be to expose lower-level SIP messages or provide a callback that allows developers to inspect and react to SIP responses directly. However, having a built-in event for call acceptance within ESP-ADF would simplify development.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

For example, in a standard SIP exchange, when a call is accepted, the 200 OK response is received:

```
SIP/2.0 200 OK  
Via: SIP/2.0/UDP 192.168.1.100:5060;branch=z9hG4bK-524287-1---1f2e317f01a41f73  
To: <sip:user@sipserver.com>;tag=1928301774  
From: <sip:caller@sipserver.com>;tag=456248  
Call-ID: a84b4c76e66710@192.168.1.100  
CSeq: 1 INVITE  
Content-Type: application/sdp  
Content-Length: 199  
```

With this enhancement, ESP-ADF would be able to detect and notify when the 200 OK response is received.

TempoTian, 2025-03-31:
When call is accepted, peer SDP is received and media session started, so `ESP_RTC_EVENT_AUDIO_SESSION_BEGIN`  or `ESP_RTC_EVENT_VIDEO_SESSION_BEGIN` is send to user. Is it enough?

carlosrodridel, 2025-04-01:
While receiving SDP and starting the media session (triggering ESP_RTC_EVENT_AUDIO_SESSION_BEGIN or ESP_RTC_EVENT_VIDEO_SESSION_BEGIN) indicates that the call is being set up, it doesn’t necessarily mean the remote party has answered. In our application, where we use a service that bridges our VoIP server to public phone numbers, we observe that this event is triggered while the recipient’s phone is still ringing. It’s only when a definitive `200 OK` response is received that the peer has actually picked up the call. Thus, relying solely on the media session begin events is not sufficient, we need a distinct event that accurately indicates when the call is truly answered.[](url)

TempoTian, 2025-04-01:
@carlosrodridel  Could you please fetch one session wireshark dump for me? I have hardly seen `200 OK` without SDP before.

carlosrodridel, 2025-04-01:
@TempoTian ,  sorry for the misunderstanding, `200 OK` is sent with SDP information. But I think I might have a clue of the actual problem :

If you look at the logs from my application, you'll see that `ESP_RTC_EVENT_AUDIO_SESSION_BEGIN` (log 56656 ) is triggered right after receiving the `SIP/2.0 183 In band info available` message, which includes SDP data. This event is sent just before the ringing and the final answer (the `200 OK` response to the `INVITE`). It appears that in our case, the SDP is exchanged earlier in the call setup, leading to the premature triggering of the media session start event, while the call is still in the ringing state.

```
INVITE sip:07XXXXXXXX@sip3.ovh.fr:5060 SIP/2.0
Via: SIP/2.0/UDP 10.XXX.XXX.XXX:11580;branch=z9hG4bK-1234301716;rport
From: <sip:003XXXXXXXXXX@sip3.ovh.fr:5060>;tag=-962242274
To: <sip:07XXXXXXXX@sip3.ovh.fr:5060>
Contact: <sip:003XXXXXXXXXX@10.XXX.XXX.XXX:11580;transport=UDP>
Max-Forwards: 70
Call-ID: A2C586009A091938886A73D47C82CBB11D05D8F7F331
CSeq: 3 INVITE
Expires: 300
User-Agent: ADETEC_GATEWAY
Content-Length: 160
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, NOTIFY, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Content-Type: application/sdp
Authorization: Digest username="003XXXXXXXXXX", realm="sip3.ovh.fr", nonce="1d4b0af675084fac4e1113b5509fe1d8", uri="sip:sip3.ovh.fr:5060", response="a9137766a53a7c2391cc70cfe1bd9b49", algorithm=MD5, opaque="1d45f693748baf6"

v=0
o=Z 0 3 IN IP4 10.XXX.XXX.XXX
s=SIP Call
c=IN IP4 10.XXX.XXX.XXX
t=0 0
m=audio 35507 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000

I (54516) SIP: [2025-03-31/10:45:17]=======================>>
I (54676) SIP: [2025-03-31/10:45:17]<<=====READ 0308 bytes==
I (54676) SIP: 

SIP/2.0 100 Trying
Call-ID: A2C586009A091938886A73D47C82CBB11D05D8F7F331
CSeq: 3 INVITE
From: <sip:003XXXXXXXXXX@sip3.ovh.fr:5060>;tag=-962242274
To: <sip:07XXXXXXXX@sip3.ovh.fr:5060>
Via: SIP/2.0/UDP 10.XXX.XXX.XXX:11580;received=37.174.190.152;rport=60644;branch=z9hG4bK-1234301716
Content-Length: 0

I (54677) SIP: [2025-03-31/10:45:17]<<======================
I (54685) SIP: [2025-03-31/10:45:17]<<=====READ 0749 bytes==
I (54685) SIP: 

SIP/2.0 407 authentication required
Call-ID: A2C586009A091938886A73D47C82CBB11D05D8F7F331
Contact: <sip:07XXXXXXXX@91.XXX.XXX.XXX:5060;user=phone>
CSeq: 3 INVITE
From: <sip:003XXXXXXXXXX@sip3.ovh.fr:5060>;tag=-962242274
Proxy-Authenticate: Digest realm="sip3.ovh.fr",nonce="efe32eeb13ebc2934f463b9946f7ab8c",opaque="efd7614c7a628f0",stale=false,algorithm=MD5
Record-Route: <sip:91.XXX.XXX.XXX:5060;transport=udp;lr>;session=85387
To: <sip:07XXXXXXXX@sip3.ovh.fr:5060>;tag=00-07509-efe32fab-1e31a61a6
Via: SIP/2.0/UDP 10.XXX.XXX.XXX:11580;received=37.174.190.152;rport=60644;branch=z9hG4bK-1234301716
Allow: REFER,INVITE,NOTIFY,ACK,UPDATE,OPTIONS,REGISTER,SUBSCRIBE,NOTIFY,CANCEL,BYE,PRACK
Server: Cirpack/v4.76 (gw_sip)
Content-Length: 0

I (54687) SIP: [2025-03-31/10:45:17]<<======================
I (54688) SIP: Required ACK & Authentication
I (54688) SIP: [2025-03-31/10:45:17]=======WRITE 0533 bytes>>
I (54689) SIP: 

ACK sip:07XXXXXXXX@91.XXX.XXX.XXX:5060;user=phone SIP/2.0
Via: SIP/2.0/UDP 10.XXX.XXX.XXX:11580;branch=z9hG4bK-1234301716;rport
Route: <sip:91.XXX.XXX.XXX:5060;transport=udp;lr>;session=85387
From: <sip:003XXXXXXXXXX@sip3.ovh.fr:5060>;tag=-962242274
To: <sip:07XXXXXXXX@sip3.ovh.fr:5060>;tag=00-07509-efe32fab-1e31a61a6
Contact: <sip:003XXXXXXXXXX@10.XXX.XXX.XXX:11580;transport=UDP>
Max-Forwards: 70
[... 172 lines cut ...]
a=sendrecv

I (65544) SIP: [2025-03-31/10:45:23]<<======================
I (65546) SIP: Remote ARTP port=31482
I (65546) SIP: Remote RTP addr=91.121.129.146
W (65546) SIP: CHANGE STATE FROM 68, TO 96, :func: _sip_uac_process_res_ok:1333
I (65547) SIP: [2025-03-31/10:45:23]=======WRITE 0735 bytes>>
I (65547) SIP: 

ACK sip:91.XXX.XXX.XXX:5060 SIP/2.0
Via: SIP/2.0/UDP 10.XXX.XXX.XXX:11580;branch=z9hG4bK-495879169;rport
Route: <sip:91.XXX.XXX.XXX:5060;transport=udp;lr>;session=85398
From: <sip:003XXXXXXXXXX@sip3.ovh.fr:5060>;tag=1127545089
To: <sip:07XXXXXXXX@sip3.ovh.fr:5060>;tag=00-07536-efe32fbe-40f9f6b92
Contact: <sip:003XXXXXXXXXX@10.XXX.XXX.XXX:11580;transport=UDP>
Max-Forwards: 70
Call-ID: A2C586009A091938886A73D47C82CBB11D05D8F7F331
CSeq: 4 ACK
Expires: 300
User-Agent: ADETEC_GATEWAY
Content-Length: 0
Authorization: Digest username="003XXXXXXXXXX", realm="sip3.ovh.fr", nonce="efe32eeb13ebc2934f463b9946f7ab8c", uri="sip:sip3.ovh.fr:5060", response="bf063979639f4f28280215ac0c0bc05e", algorithm=MD5, opaque="efd7614c7a628f0"
```

[image]

carlosrodridel, 2025-04-02:
Yes, that would be perfect!, Thanks a lot

TempoTian, 2025-04-02:
I have add the new event `ESP_RTC_EVENT_CALL_ANSWERED`. You can test it, the lib is latest, I will prepare to merge recent changed in next week, if verify OK it will also be merged.

[add_call_answer.zip](https://github.com/user-attachments/files/19565177/add_call_answer.zip)

carlosrodridel, 2025-04-03:
@TempoTian I tested the new event on the ESP32S3, and it perfectly satisfies our needs, and we can now accurately detect when the call is actually answered. Thanks for the prompt update!