espressif/esp-adf #1424: 20250409，使用最新的master版本，在使用volc rtc 程序。编译报错，其它的历程都是编译均正常

wuliaodejijiji, 2025-04-09:
以下是日志
iltin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -Og -fno-shrink-wrap -fmacro-prefix-map=D:/Users/Administrator/Desktop/volc_rtc=. -fmacro-prefix-map=D:/espressif/idf531/Espressif/frameworks/esp-idf-v5.3.1=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/display_service/CMakeFiles/__idf_display_service.dir/display_service.c.obj -MF esp-idf\display_service\CMakeFiles\__idf_display_service.dir\display_service.c.obj.d -o esp-idf/display_service/CMakeFiles/__idf_display_service.dir/display_service.c.obj -c D:/espressif/adf_master_xx/components/display_service/display_service.c
In file included from D:/espressif/adf_master_xx/components/audio_board/esp32_s3_korvo2_v3/board.h:28,
                 from D:/espressif/adf_master_xx/components/display_service/display_service.c:34:
D:/espressif/adf_master_xx/components/audio_hal/include/audio_hal.h:150:5: error: unknown type name 'xSemaphoreHandle'
  150 |     xSemaphoreHandle audio_hal_lock;                                                                         /*!< semaphore of codec */
      |     ^~~~~~~~~~~~~~~~
In file included from D:/espressif/adf_master_xx/components/audio_board/esp32_s3_korvo2_v3/board.h:31:
D:/espressif/adf_master_xx/components/esp_peripherals/include/esp_peripherals.h:79:32: error: unknown type name 'xTimerHandle'; did you mean 'TimerHandle_t'?
   79 | typedef void (*timer_callback)(xTimerHandle tmr);
      |                                ^~~~~~~~~~~~
      |                                TimerHandle_t
D:/espressif/adf_master_xx/components/esp_peripherals/include/esp_peripherals.h:387:88: error: unknown type name 'timer_callback'
  387 | esp_err_t esp_periph_start_timer(esp_periph_handle_t periph, TickType_t interval_tick, timer_callback callback);
      |                                                                                        ^~~~~~~~~~~~~~
[1020/1396] Building C object esp-idf/adf_utils/CMakeFiles/__idf_adf_utils.dir/cloud_services/baidu_access_token.c.obj
ninja: build stopped: subcommand failed.

wuliaodejijiji, 2025-04-17:
> 使能一下 `CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY`， 默认的配置应该是使能的

谢谢你，目前已经解决了，我今天重新git了最新版本，编译已经不出错了，

closed 2025-04-18 by jason-mao

bao17634, 2025-04-26:
> > 使能一下 `CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY`， 默认的配置应该是使能的
> 
> 谢谢你，目前已经解决了，我今天重新git了最新版本，编译已经不出错了，

你的idf是哪个版本的我的idf是5.3的 adf也是最新的  但还是报这个错误