espressif/esp-adf #1425: ESP-BOX在master的aec example效果不佳

casitelwithdean, 2025-04-10:
ESP-S3-BOX 
IDF  5.2.1
adf master版本的aec example中回声消除的效果不佳，需要和korvo 一样设置 mic3的增益吗？
应该调成RM还是MR呢？
这个aec对输入的音频频率有无需求，8kHz的采样可以吗
谢谢！

shootao, 2025-04-17:
1.关于 esp-box ， 如果使用单声道的话 是 RM
2.这个不能和 korvo 一样设置 mic3的增益， 硬件的结构不一样
3. AEC 只能处理 16bit 的音频， 如果你看到 32bit， 那是应为将 32bit 拆成两个 16bit

casitelwithdean, 2025-04-20:
> 1.关于 esp-box ， 如果使用单声道的话 是 RM 2.这个不能和 korvo 一样设置 mic3的增益， 硬件的结构不一样 3. AEC 只能处理 16bit 的音频， 如果你看到 32bit， 那是应为将 32bit 拆成两个 16bit

谢谢答复。第三点还不是很明白
但我还想请问一下algorithm示例的一些问题
1.ESP_S3_BOX是否支持algorithm呢？ 我看ESP_S3_BOX的CODEC_ADC_BITS_PER_SAMPLE是32位，AEC不是只支持16位的吗，那我i2s_read_stream初始化的时候是英国初始化成32位还是16位呢？如果32位是不是要还要手动进行拆分呢？
2.我使用的algorithm代码如下，
```c
  i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA(CODEC_ADC_I2S_PORT, 16000, 16, AUDIO_STREAM_READER);
  i2s_stream_set_channel_type(&i2s_cfg, I2S_CHANNEL_TYPE_ONLY_LEFT);
  i2s_stream_reader_rtc = i2s_stream_init(&i2s_cfg);

    algorithm_stream_cfg_t algo_config = ALGORITHM_STREAM_CFG_DEFAULT();

    algo_config.sample_rate = 16000;
    algo_config.out_rb_size = 2560;
    algo_config.algo_mask = ALGORITHM_STREAM_USE_AEC;
    audio_element_handle_t element_algo = algo_stream_init(&algo_config);
    audio_element_set_music_info(element_algo, 16000, 1, ALGORITHM_STREAM_DEFAULT_SAMPLE_BIT);
    // audio_element_set_read_cb(element_algo, i2s_read_cb, NULL);
    audio_element_set_input_timeout(element_algo, portMAX_DELAY);

  raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
  raw_cfg.type = AUDIO_STREAM_READER;
  raw_cfg.out_rb_size = 20 * 1024;
  raw_read = raw_stream_init(&raw_cfg);
  audio_element_set_output_timeout(raw_read, portMAX_DELAY);

  audio_pipeline_register(recorder, i2s_stream_reader_rtc, "i2s");
  audio_pipeline_register(recorder, element_algo, "algo");
  audio_pipeline_register(recorder, raw_read, "raw_read");
 
  const char *link_tag[3] = { "i2s","algo", "raw_read"};
  audio_pipeline_link(recorder, &link_tag[0], 3);
```但会报下列warning：
```c
 AFE: Ringbuffer of AFE is empty, Please use feed() to write data
```
按理说pipeline run起来之后就应该自动feed数据吧，一直没找到错误原因，困扰许久了，还望您指点一二，感谢！！

yuezhao-ali, 2025-07-15:
ESP-S3-BOX-3B 同样遇到类似问题，如果不使用 AEC 之类算法，可以正常进行音频采集（[codec_chip]-->i2s_stream-->raw），使用的是 16kHz、16bit、单通道；

如果中间添加一个 aec_stream 或 algo_stream（[codec_chip]-->i2s_stream-->aec / algo-->raw），则音频采集会卡住，不论用 16Khz、16bit、单通道 还是 16Khz、32bit、双通道，都是如此。 提示 AFE: Ringbuffer of AFE is empty, Please use feed() to write data。换成 audio_element_set_read_cb 方式从 i2s_stream 中读数据给 aec / algo 的话，则发现首次读取操作就被阻塞住了。

@shootao 帮忙看下，ESP-S3-BOX-3B 上应该怎样才能使用 AEC？