espressif/esp-adf #1432: White Noise when switching audio - A2DP Pipeline BT Stream

NehalAR, 2025-04-23:
- [x] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [x] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- [Required] Module or chip used:  ESP32 WROOM 32E with 2MB PSRAM - 16MB flash
- [Required] IDF version : 4.4.6
- [Required] ADF version : v2.7-51-g60054d26

I used CONFIG_ESP_LYRAT_MINI_V1_1_BOARD to build code, along with LVGL v9.2 on Core 1 and Bluetooth stack on Core 0. Allocated the tasks which could be on PSRAM.

## Problem Description

When switching audio I get a constant white noise sometimes. I don't understand where is this occurring from. I used play bt source example as a base and wrote code over it. 

There is no specific pattern it happens randomly between same or different audio formats (.wav, .mp3) or even same audio but happens only when new audio loads (not in between playing).

### Additional Info

I have disabled the AVRC as it was not needed increased the task stack sizes for internal decoders like WAV, MP3 and BTC_BT tasks. Tested with different earbuds.

[sdkconfig.txt](https://github.com/user-attachments/files/19872327/sdkconfig.txt)

### Expected Behavior

I want the audio switching to be smooth, already looked into example code and similar issues but I am doing the same which is recommended in those.

### Code to Reproduce This Issue

```cpp

/* Starts Bluetooth scanning */
void A2DP::start_bt_scan() {

    ESP_LOGI(TAG, "Starting Bluetooth scan...");
    scanned_device_count = 0;  // Reset scanned devices count
    memset(scanned_devices, 0, sizeof(scanned_devices));
    esp_bt_gap_start_discovery(ESP_BT_INQ_MODE_GENERAL_INQUIRY, 5, 0);
    esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
}

/* Connects to a Bluetooth device by name */
bool A2DP::connect_to_bt_device(const char *device_name)
{
    for (int i = 0; i < scanned_device_count; i++) {
        if (strcmp(scanned_devices[i].name, device_name) == 0) {
            char bda_str[18];
            ESP_LOGI(TAG, "Connecting to %s (%s)...", device_name, bda2str(scanned_devices[i].bda, bda_str, 18));
            if (esp_a2d_source_connect(scanned_devices[i].bda) == ESP_OK) {
                ESP_LOGI(TAG, "Connection successful.");
                deviceConnected = true;
                return true;
            } else {
                ESP_LOGE(TAG, "Connection failed.");
                deviceConnected = false;
                return false;
            }
        }
    }
    memset(deviceName, 0, BUF_SIZE);
    ESP_LOGW(TAG, "Device %s not found.", device_name);
    return false;
}

void A2DP::A2dpServiceInit()
{
    ESP_LOGI(TAG, "[ 2 ] Bluetooth setup");
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));
[... 372 lines cut ...]
I (309096) AUDIO_THREAD: The file task allocate stack on internal memory
I (309103) AUDIO_ELEMENT: [file-0x3f87ff94] Element task created
I (309110) AUDIO_ELEMENT: [bt-0x3f88029c] Element task created
I (309116) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:1680287 Bytes, Inter:176531 Bytes, Dram:170347 Bytes, Dram largest free:110592Bytes

I (309131) AUDIO_ELEMENT: [dec] AEL_MSG_CMD_RESUME,state:1
I (309138) MP3_DECODER: MP3 opened
I (309142) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (309148) AUDIO_PIPELINE: Pipeline started
I (309156) BLUETOOTH_A2DP: loopAudioTask exit
I (309157) BLUETOOTH_A2DP: A2dpAudioStart started
W (309164) BLUETOOTH_A2DP: [Starting] Playback time 307688 : 307688 : 600000
I (309160) FATFS_STREAM: File size: 2888149 byte, file position: 0
I (309187) CODEC_ELEMENT_HELPER: The element is 0x3f884f60. The reserve data 2 is 0x0.
W (488951) FATFS_STREAM: No more data, ret:0
I (488951) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (488951) AUDIO_ELEMENT: IN-[dec] AEL_IO_DONE,-2
I (489071) MP3_DECODER: Closed
W (489073) BLUETOOTH_A2DP: [msg event] source: 131072 lastData: 12 msg.data: 0 lastCmd: 8 msg.cmd: 9
W (489074) BLUETOOTH_A2DP: [msg event] source: 131072 lastData: 0 msg.data: 15 lastCmd: 9 msg.cmd: 8
W (489087) BLUETOOTH_A2DP: [ LOOP ] Playback time 487611 : 307688 : 600000
I (489087) AUDIO_ELEMENT: IN-[bt] AEL_IO_DONE,-2
W (489092) AUDIO_ELEMENT: [dec] Element already stopped
W (489103) AUDIO_ELEMENT: [file] Element already stopped
```

hqMicro, 2025-04-25:
Same here. I came to open a new issue and noticed this.
I only tested MP3 files, and it’s unpredictable. Sometimes, after playing 5 or 8 tracks, it suddenly happens and continues until the end of the track. On the next track, it might be fine or it might not.

NehalAR, 2025-04-25:
> Same here. I came to open a new issue and noticed this.
> I only tested MP3 files, and it’s unpredictable. Sometimes, after playing 5 or 8 tracks, it suddenly happens and continues until the end of the track. On the next track, it might be fine or it might not.

Exactly, it sometimes happen much faster for me like on the third audio. I even tried to open and check debug logs for fatts stream for byte position or some element or audio task in background running/not properly closing but I found no difference in the audios that are working fine for me and the ones that are aren't. It's very random for me at this point.