espressif/esp-adf #1438: VOIP 例程，现场测试发现拨通对话的时候有小概率出现话机听不到设备的声音，设备端可以听到话机的声音。

ktoto2011, 2025-04-28:
[ √] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
[ √] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
[ √] I have searched the issue tracker for a similar issue and not found a similar issue.

Environment
Audio development kit: ESP32-S3-Korvo-2
Audio kit version (ESP32-S3-Korvo-2): v3
[Required] Module or chip used: ESP32-S3-WROOM-1
[Required] IDF version v4.47
[Required] ADF version v2.6-183-gce64fea5
Build system: [idf.py]
[Required] Running log: All logs from power-on to problem recurrence
Operating system: [Windows]
(Windows only) Environment type: [ESP Command Prompt]
Using an IDE?: vscode
Power supply: USB

Problem Description

现场测试发现设备主拨到话机的时候，有小概率出现设备无法听到话机的声音，但是话机是能听到设备说话的声音。这种情况出现在设备连接的热点切换的过程，想请教下，这是不是在网络切换变动过程中，重新注册sip 的时候，设备这边某些初始化不稳定造成的，这边应该注意设备那个组件的初始化/sip重连过程中那个函数的复位管理？

ktoto2011, 2025-04-28:
@TempoTian   麻烦你，有空帮忙指导下。现场最近出现问题反应比较强烈

TempoTian, 2025-04-28:
目前播放和录制都是通过I2S采集信号实现的， I2S只在 `av_stream_audio_deinit` 的时候才会销毁
https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal_audio.c#L202C5-L202C27

如果录制是OK的，并切也正常发送。可以看下 Codec 的寄存器是否正常，也可以在read函数里面打印下采集的音频信号，看是否有值
```c
int av_stream_audio_read(char *buf, int len, TickType_t wait_time, bool uac_en)
{
    int ret = ESP_FAIL;
    size_t bytes_read = 0;

    if (uac_en) {
        // TODO
        // ret = uac_mic_streaming_read(buf, len, &bytes_read, wait_time);
        // if (ret != ESP_OK) {
        //     ESP_LOGE(TAG, "uac read failed");
        // }
        return ret;
    } else {
        ret = audio_element_input(i2s_io_reader, buf, len);
        if (ret <= 0) {
            ESP_LOGE(TAG, "i2s fail read ret %d", ret);
            return ret;
        }
        int16_t* v = (int16_t*) buf;
        printf("Read %02x %02x\n", v[0], v[1]);
        bytes_read = ret;
    }

    return bytes_read;
}
```

ktoto2011, 2025-04-28:
> 目前播放和录制都是通过I2S采集信号实现的， I2S只在 `av_stream_audio_deinit` 的时候才会销毁 https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal_audio.c#L202C5-L202C27
> 
> 如果录制是OK的，并且也正常发送。可以看下 Codec 的寄存器是否正常，也可以在read函数里面打印下采集的音频信号，看是否有值
> 
> int av_stream_audio_read(char *buf, int len, TickType_t wait_time, bool uac_en)
> {
>     int ret = ESP_FAIL;
>     size_t bytes_read = 0;
> 
>     if (uac_en) {
>         // TODO
>         // ret = uac_mic_streaming_read(buf, len, &bytes_read, wait_time);
>         // if (ret != ESP_OK) {
>         //     ESP_LOGE(TAG, "uac read failed");
>         // }
>         return ret;
>     } else {
>         ret = audio_element_input(i2s_io_reader, buf, len);
>         if (ret <= 0) {
>             ESP_LOGE(TAG, "i2s fail read ret %d", ret);
>             return ret;
>         }
>         int16_t* v = (int16_t*) buf;
>         printf("Read %02x %02x\n", v[0], v[1]);
>         bytes_read = ret;
>     }
> 
>     return bytes_read;
> }

> 目前播放和录制都是通过I2S采集信号实现的， I2S只在 `av_stream_audio_deinit` 的时候才会销毁 https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal_audio.c#L202C5-L202C27
> 
> 如果录制是OK的，并且也正常发送。可以看下 Codec 的寄存器是否正常，也可以在read函数里面打印下采集的音频信号，看是否有值
> 
> int av_stream_audio_read(char *buf, int len, TickType_t wait_time, bool uac_en)
> {
>     int ret = ESP_FAIL;
>     size_t bytes_read = 0;
> 
>     if (uac_en) {
>         // TODO
>         // ret = uac_mic_streaming_read(buf, len, &bytes_read, wait_time);
>         // if (ret != ESP_OK) {
>         //     ESP_LOGE(TAG, "uac read failed");
>         // }
>         return ret;
>     } else {
>         ret = audio_element_input(i2s_io_reader, buf, len);
>         if (ret <= 0) {
>             ESP_LOGE(TAG, "i2s fail read ret %d", ret);
>             return ret;
>         }
>         int16_t* v = (int16_t*) buf;
>         printf("Read %02x %02x\n", v[0], v[1]);
>         bytes_read = ret;
>     }
> 
>     return bytes_read;
> }

感谢你的回复，那这边要重现问题，抓这个log看看。

ktoto2011, 2025-06-12:
现场抓到的log 信息，上面把sip的账号信息注释掉了。主呼：设备打给话机；被呼：话机打给设备；
被呼_.txt
第一次主呼.txt
第二次主呼.txt
前面三个log 在通话过程中都是设备听不到前台的声音，前台能听到设备的声音；

重启后正常.txt

有空麻烦帮忙看下，就目前的log 能否定位到问题，会不会sip   联网注册的过程出现对寄存器配置异常的情况，导致声音解码有问题？但是在主呼前台话机，设备的振铃播放的声音是正常的。@TempoTian

ktoto2011, 2025-06-12:
[被呼_.txt](https://github.com/user-attachments/files/20704016/_.txt)
[第二次主呼.txt](https://github.com/user-attachments/files/20704014/default.txt)
[第一次主呼.txt](https://github.com/user-attachments/files/20704013/default.txt)
[重启后正常.txt](https://github.com/user-attachments/files/20704015/default.txt)

ktoto2011, 2025-06-28:
> [@ktoto2011](https://github.com/ktoto2011) 目前从log分析看，没看到异常。只是看到SDP错误了 异常的时候是[14:41:19.154] a=recvonly 正常的时候是[14:46:22.451] a=sendrecv recvonly表示对端只接收不发了，可以看下什么原因导致的，看是否是server端有一些特殊配置

谢谢，这边我们已经在查看

ktoto2011, 2025-06-28:
目前产品在现场还有个问题，设备接听 AI （对面是文字转音频）播放出来的声音卡顿感很明显；如果设备端和设备端互打，播放出来声音也稍微有点卡顿但不是很明显。但是设备播放本地的录制文件声音是很正常的。这边也抓到在通话过程的网络时延，也不会算高。想请教下，应该怎么优化，可能是固件哪里出现的问题。@TempoTian

ktoto2011, 2025-06-28:
https://github.com/user-attachments/assets/3b995330-db48-487a-b3e1-effca09b7f6e      这是和AI 对讲的效果，卡顿很明显

ktoto2011, 2025-06-28:
https://github.com/user-attachments/assets/2df48152-e587-41aa-a9cf-2306a2c886ed  这是pc 端打给设备的效果，PC 端那边用手机播放的音频

ktoto2011, 2025-06-28:
https://github.com/user-attachments/assets/8b304562-cf75-4231-bf39-b77ce6ba9dd8   这是当时接听的过程 ping 设备的延时

TempoTian, 2025-07-01:
卡顿的问题主要是丢数据
https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/av_stream/av_stream.c#L695
这一行代码没有任何等待，如果fifo满了就会有问题，可以通过加大buffer或者适当的delay来实现。如果是实时流的话由于进入的比较均衡问题不大，如果是生成的stream由于瞬间会产生大量的数据，导致没有可用buffer就会丢数据。加大buffer会导致delay加大，增加delay会导致卡住网络发送。这里需要一个权衡，你可以加大下fifo size，加大timeout，多试几组数据看哪个比较理想
```c
    av_stream->ringbuf_dec = rb_create(3*av_stream->config.hal.audio_framesize, 1);
```

ktoto2011, 2025-07-03:
> 大timeout，多试几组数据看哪个比

谢谢，我们这边调整试试效果

ktoto2011, 2025-07-08:
> 卡顿的问题主要是丢数据 https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/av_stream/av_stream.c#L695 这一行代码没有任何等待，如果fifo满了就会有问题，可以通过加大buffer或者适当的delay来实现。如果是实时流的话由于进入的比较均衡问题不大，如果是生成的stream由于瞬间会产生大量的数据，导致没有可用buffer就会丢数据。加大buffer会导致delay加大，增加delay会导致卡住网络发送。这里需要一个权衡，你可以加大下fifo size，加大timeout，多试几组数据看哪个比较理想
> 
>     av_stream->ringbuf_dec = rb_create(3*av_stream->config.hal.audio_framesize, 1);

加大fifo size  是调整  av_stream->ringbuf_dec = rb_create(3*av_stream->config.hal.audio_framesize, 1);  n*av_stream->config.hal.audio_framesize 中n的值吗，调 成8/20 ，AI 对讲还是卡顿，跟sip 传过来的声音有关系吗

TempoTian, 2025-07-08:
抓个包看下，感觉网络进不来数据了