espressif/esp-adf #1443: Crash in voip sample when answering a call on Lyra-Mini-1.2

acf, 2025-05-06:
----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      [ESP32-LyraT-Mini]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version v5.4.1 installed as a VSCOde extention
- [Required] ADF version v2.7-89-g3deab278 installed via `install adf` in VSCode
- Operating system:     [Raspian Bullseye]
- Using an IDE?: [VSCode]
- Power supply:         [USB]

## Problem Description

Apologies for using words I only half understand. I'm very new to embedded development so am largely reading docs, following steps and just trying to get a sample to work correctly before I go any further.

I have a local Asterisk server running and am able to successfully communicate between two Macs on my local network using the `Telephone` app so I know the server is properly setup and working.

If I call a Mac from the Lyra board, the call is answered but disconnected after 30s with no audio.  I am guessing this is some kind of network config issue as I saw the same between two macs before I set up some NAT+port forwarding on my router.  This is likely my problem, not yours.

More concerning is that if I call to the Lyra **from the Mac**, the ESP32 application just crashes and I'm not smart enough to figure out why.

Relevant log section is below.  ANy help appreciated here and happy to send other files as required (though I may need some hand holding - I'm patient and willing!)
```
ACK sip:200@192.168.0.10:14789 SIP/2.0
Via: SIP/2.0/UDP 192.168.0.25:5060;rport;branch=z9hG4bKPj46603709-be00-43c7-89a5-5b0fd1ecbec4
From: "100" <sip:100@192.168.0.25>;tag=8cf1fbb9-702f-4c1a-aa86-5fb201e8ed70
To: <sip:200@192.168.0.10>;tag=1780814633
Call-ID: d9832b8f-dbac-4909-a03d-9a579c8748c2
CSeq: 27305 ACK
Max-Forwards: 70
User-Agent: FPBX-17.0.19.25(22.3.0)
Content-Length:  0

I (40828) SIP: [1970-01-01/00:00:19]<<======================
I (40836) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_BEGIN
E (40840) MODEL_LOADER: Can not find model in partition table
E (40845) ALGORITHM_STREAM: Failed to load models
I (40850) AUDIO_PIPELINE: link el->rb, el:0x3f825bb8, tag:algo, rb:0x3f825d14
I (40856) AUDIO_THREAD: The algo task allocate stack on external memory
I (40863) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4203932 Bytes, Inter:239963 Bytes, Dram:198539 Bytes, Dram largest free:110592Bytes

W (40877) AFE_CONFIG: For single microphone channel, SE is deactivated.
E (40886) AEC: Only support 16K sample rate
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401c2783  PS      : 0x00060330  A0      : 0x80107731  A1      : 0x3f827c80  
--- 0x401c2783: aec_get_chunksize at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/esp_aec.c:150

A2      : 0x00000000  A3      : 0x00000004  A4      : 0x00000001  A5      : 0x00000003  
A6      : 0x00000000  A7      : 0x00001f40  A8      : 0x80108375  A9      : 0x3f827c30  
A10     : 0x00000001  A11     : 0x3f40d934  A12     : 0x3f40d938  A13     : 0x00009fb6  
A14     : 0x3f40d934  A15     : 0x00000400  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4008cfb1  LEND    : 0x4008cfc1  LCOUNT  : 0xfffffffe  
--- 0x4008cfb1: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84
--- 0x4008cfc1: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x401c2780:0x3f827c80 0x4010772e:0x3f827ca0 0x40107ba5:0x3f827cc0 0x400e4554:0x3f827ce0 0x400e76b8:0x3f827d10 0x400e7c3c:0x3f827d40 0x400e7d42:0x3f827d60 0x400e8cc9:0x3f827d90 0x400e7f4f:0x3f827dd0 0x4008fd52:0x3f827e00
--- 0x401c2780: aec_get_chunksize at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/esp_aec.c:148
--- 0x4010772e: afe_init_aec_ns at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:171
--- 0x40107ba5: afe_create_from_config at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:470
--- 0x400e4554: _algo_open at /home/acf/.espressif/esp-adf/components/audio_stream/algorithm_stream.c:186
--- 0x400e76b8: audio_element_process_init at /home/acf/.espressif/esp-adf/components/audio_pipeline/audio_element.c:175
--- 0x400e7c3c: audio_element_on_cmd_resume at /home/acf/.espressif/esp-adf/components/audio_pipeline/audio_element.c:283
--- 0x400e7d42: audio_element_on_cmd at /home/acf/.espressif/esp-adf/components/audio_pipeline/audio_element.c:324
--- 0x400e8cc9: audio_event_iface_waiting_cmd_msg at /home/acf/.espressif/esp-adf/components/audio_pipeline/audio_event_iface.c:246
--- 0x400e7f4f: audio_element_task at /home/acf/.espressif/esp-adf/components/audio_pipeline/audio_element.c:477
--- 0x4008fd52: vPortTaskWrapper at /home/acf/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: c46dab73c
```

// Detailed problem description goes here.

### Expected Behavior

Honestly I'm not sure what I should expect the voip sample app to do ?  I dont know of I should be able to talk into the microphone or if its just supposed to play a sound.  Either way, I'm assuming it shouldnt crash :-)

### Actual Behavior

voip sample app crashes when I answer a call

### Steps to Reproduce

1. configure and run voip sample to be 200@192.168.0.1
2. call 200@192.168.0.1 from another computer ( 100@192.168.0.1 )

shootao, 2025-05-07:
Hi @acf @Hasaken 
Yes, this is due to the latest ESP-SR not supporting 8k AEC, and this issue will be fixed in the next two days

Hasaken, 2025-05-07:
@shootao Thank you for the response.
Could you please review this issue as well?
https://github.com/espressif/esp-adf/issues/1442

acf, 2025-05-07:
@Hasaken Thanks!  Sorry to be a dummy but can you explain to me how best to do that?  Is this something I need to configure In code for the audio chip... or is the problem in the tone file I downloaded in the "Download flash tone" step of the Sample readme ?

At this stage I don't need to understand everything, I'm literally just trying to get the sample code to build and run so feel free assume I know absolutely nothing give me "click here, type this" instructions if you can :-)

Hasaken, 2025-05-07:
Dear @acf , it can be done by manually updating some .h and/or .c files, but in accordance to @shootao response, I think the best way is to wait and then update your ADF.

acf, 2025-05-08:
@Hasaken Thank you!

acf, 2025-05-09:
I couldn't work out how to change the sample rate.  I tried hacking in a `aec->aec-handle->pcm_config.sample_rate = 16384;` into `_aec_open()` in `components/audio_stream/aec_stream.c` but no joy.  In the end I have disabled AEC entirely in `audio_stream/include/algorithm_stream.h` by changing `ALGORITHM_STREAM_DEFAULT_MASK` to exclude AEC.

voip sample is working now (albeit with lots of echo and feedback :-)

I'll keep checking for a proper fix, but for now I'm able to proceed. Thanks!

acf, 2025-05-15:
> Hi [@acf](https://github.com/acf) [@Hasaken](https://github.com/Hasaken) Yes, this is due to the latest ESP-SR not supporting 8k AEC, and this issue will be fixed in the next two days

@shootao I didn't see a fix go in for this, did I miss it, or are you guys just too busy ? :-). (I figured out how to disable AEC in the meantime since I couldn't figure out how to change the sample rate)

shootao, 2025-05-28:
@acf 
We have updated ESP-ADF， Pls try again

acf, 2025-05-29:
@shootao thank you!  I will try it out today.

closed 2025-06-06 by acf