espressif/esp-adf #1446: 使用ESP-RTC例程进行视频通话无法显示图像

KeysPAN0114, 2025-05-12:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, start similar to "How do I..". If it is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [Espressif Advanced Development Framework Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [ ] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps above, please follow these instructions so we can get the needed information to help you quickly and effectively.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your issues.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other Items If Possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the `Preview` tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: Please follow the above instructions and provide as many details as possible. It will save a lot of communication time and improve the efficiency of problem solving. The more details you provide, the faster we may be able to reproduce and resolve the issue. Thanks!**

----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      [ESP32-LyraT|ESP32-LyraT-Mini|ESP32-LyraTD-MSC|ESP32-S2-Kaluga-1|ESP32-Korvo-DU1906|ESP32-S3-Korvo-2|ESP32-S3-Korvo-2-LCD|none]
- Audio kit version (ESP32-S3-Korvo-2:v2)
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version V5.1.2
- [Required] ADF master
    // v2.3-171-gaac5912
- Build system:         [Make|CMake|idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     [Windows]
- (Windows only) Environment type: [PowerShell]
- Using an IDE?: [VS Code]
- Power supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
我尝试使用FreeSWITCH进行ESP-RTC例程测试，windows端使用microSIP。语音通话是正常的，但是视频通话无法显示开发板的图像。
// Detailed problem description goes here.
现象如下图：

[image]

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue
推流vedio的格式已经修改为H264
```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
    av_stream_config_t av_stream_config = {
        .algo_mask = ALGORITHM_STREAM_DEFAULT_MASK,
        .acodec_samplerate = AUDIO_CODEC_SAMPLE_RATE,
        .acodec_type = AV_ACODEC_G711A,
        .vcodec_type = AV_VCODEC_H264,
        .hal = {
            .set = set,
            .lcd_en = false,
            .uac_en = false,
            .uvc_en = false,
            .video_soft_enc = false,
            .audio_samplerate = AUDIO_HAL_SAMPLE_RATE,
            .audio_framesize = PCM_FRAME_SIZE,
            .video_framesize = VIDEO_FRAME_SIZE,
        },
    };

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

OpenWor, 2025-05-13:
我使用ESP32-s3-korvo-2开发板烧录ESP-RTC例程后也遇到了相同问题（视频无法传输，已将.vcodec_type =AV_VCODEC_MJPEG改为 .vcodec_type =AV_VCODEC_H264,），另外使用开发板与手机esp-rtc.apk应用连接，通话始终是单向的，拨打的一方听不到另一方的声音，仅另一方能听到拨打者的声音，另外想询问通常视频支持的最大像素是多少呢？

KeysPAN0114, 2025-05-13:
我用Android10的手机测试esp-rtc app会闪退，所以用microSIP测试。语音通话应该是没什么问题的。视频支持的最大像素需要自己测试，可以用ESP-RTSP测试修改合适的分辨率，例程默认的320*240我这里是获取不了图像的，我改成了1024x768，现在是15fps左右浮动。按照官网文档最高支持720P，20fps，但是我没试过。

KeysPAN0114, 2025-05-13:
> 我使用ESP32-s3-korvo-2开发板烧录ESP-RTC例程后也遇到了相同问题（视频无法传输，已将.vcodec_type =AV_VCODEC_MJPEG改为 .vcodec_type =AV_VCODEC_H264,），另外使用开发板与手机esp-rtc.apk应用连接，通话始终是单向的，拨打的一方听不到另一方的声音，仅另一方能听到拨打者的声音，另外想询问通常视频支持的最大像素是多少呢？

我用Android10的手机测试esp-rtc app会闪退，所以用microSIP测试。语音通话应该是没什么问题的。视频支持的最大像素需要自己测试，可以用ESP-RTSP测试修改合适的分辨率，例程默认的320*240我这里是获取不了图像的，我改成了1024x768，现在是15fps左右浮动。按照官网文档最高支持720P，20fps，但是我没试过。

TempoTian, 2025-05-13:
https://github.com/espressif/esp-adf/blob/master/examples/protocols/esp-rtc/main/rtc_service.c#L147
这个地方也要改下，确认下看有没有改到

KeysPAN0114, 2025-05-13:
> https://github.com/espressif/esp-adf/blob/master/examples/protocols/esp-rtc/main/rtc_service.c#L147 这个地方也要改下，确认下看有没有改到

修改这里之后会出现warring：H264_HEADER: Failed to get the next NALu in H264 payload with 0x50000005。
还是显示不出图像

[image]

TempoTian, 2025-05-14:
还漏了一个地方，使能下软件编码看

[image]

KeysPAN0114, 2025-05-15:
> 还漏了一个地方，使能下软件编码看
> 
> [image]

开启了软件编码后貌似会发生错误重启：

[image]

[image]

TempoTian, 2025-05-15:
我来跑下看，这块代码比较久没动了，看下实际跑的效果。不过在S3上面只能跑很低的分辨率和帧率  320x240 10fps左右。目前软件编码比较浪费CPU，你有什么应用场景吗需要用到H264编码的？

TempoTian, 2025-05-15:
如果分辨率有要求还是建议用MJPEG，PC端的代码可以见https://github.com/ahhfzhang/pjproject-2.13, 接上APK就可以

KeysPAN0114, 2025-05-16:
> 如果分辨率有要求还是建议用MJPEG，PC端的代码可以见https://github.com/ahhfzhang/pjproject-2.13, 接上APK就可以

好的，我试试

OpenWor, 2025-05-19:
> 我用Android10的手机测试esp-rtc app会闪退，所以用microSIP测试。 语音通话应该是没什么问题的。 视频支持的最大像素需要自己测试，可以用ESP-RTSP测试修改合适的分辨率，例程默认的320*240我这里是获取不了图像的，我改成了1024x768，现在是15fps左右浮动。按照官网文档最高支持720P，20fps，但是我没试过。

您也使用的ESP32-s3-korvo-2开发板吗？我这边烧录ESP-RTSP，如果使用.vcodec_type = AV_VCODEC_MJPEG，.video_soft_enc = false,会报错cam_hal: NO-SOI错误，使用.vcodec_type = AV_VCODEC_H264,.video_soft_enc = true，则分辨率非常小才能运行。

KeysPAN0114, 2025-05-21:
> 您也使用的ESP32-s3-korvo-2开发板吗？我这边烧录ESP-RTSP，如果使用.vcodec_type = AV_VCODEC_MJPEG，.video_soft_enc = false,会报错cam_hal: NO-SOI错误，使用.vcodec_type = AV_VCODEC_H264,.video_soft_enc = true，则分辨率非常小才能运行。

你应该检查一下硬件的供电是否正常，以及接线是否牢固。我sensor的时钟频率是19MHz的。或者你可以提供更多的log，看看大家有没有解决方法