espressif/esp-adf #1449: ai_agent/volc_rtc例程编译提示undefined reference to `__atomic_fetch_add_4'、`__atomic_fetch_add_4'、`__atomic_fetch_sub_4'

VSiR22, 2025-05-14:
[2/7] Performing build step for 'bootloader'
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D D:\ESPADF666\esp-adf\examples\ai_agent\volc_rtc\build\bootloader\esp-idf\esptool_py && d:\esp_idf\python_env\idf5.4_py3.11_env\Scripts\python.exe D:/esp/v5.4/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/ESPADF666/esp-adf/examples/ai_agent/volc_rtc/build/bootloader/bootloader.bin"
Bootloader binary size 0x6bf0 bytes. 0x410 bytes (4%) free.
[5/7] Linking CXX executable volc_rtc.elf
FAILED: volc_rtc.elf 
C:\WINDOWS\system32\cmd.exe /C "cd . && D:\esp_idf\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=D:/ESPADF666/esp-adf/examples/ai_agent/volc_rtc/build/volc_rtc.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\volc_rtc.elf.rsp -o volc_rtc.elf && cd ."
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj):(.literal.volc_atomic_exchange+0x0): undefined reference to `__atomic_exchange_4'
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj):(.literal.volc_atomic_increment+0x0): undefined reference to `__atomic_fetch_add_4'
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj):(.literal.volc_atomic_decrement+0x0): undefined reference to `__atomic_fetch_sub_4'
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj): in function `volc_atomic_exchange':
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/hal/src/platform/esp32/volc_atomic.c:15:(.text.volc_atomic_exchange+0xa): undefined reference to `__atomic_exchange_4'
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj): in function `volc_atomic_increment':
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/hal/src/platform/esp32/volc_atomic.c:19:(.text.volc_atomic_increment+0xa): undefined reference to `__atomic_fetch_add_4'
D:/esp_idf/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: D:/ESPADF666/esp-adf/components/clouds/volc_engine_rtc/libs/esp32/libVolcEngineRTCLite.a(volc_atomic.c.obj): in function `volc_atomic_decrement':
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-booka/components/VolcEngineRTCLite/ByteRTCLite/src/hal/src/platform/esp32/volc_atomic.c:23:(.text.volc_atomic_decrement+0xa): undefined reference to `__atomic_fetch_sub_4'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

shootao, 2025-05-15:
@VSiR22 确定一下你的 ADF 版本号， 可以先用 ADF master + IDF 5.4 再试一下

VSiR22, 2025-05-15:
> [@VSiR22](https://github.com/VSiR22) 确定一下你的 ADF 版本号， 可以先用 ADF master + IDF 5.4 再试一下

感谢，我尝试下

VSiR22, 2025-05-16:
> [@VSiR22](https://github.com/VSiR22) 确定一下你的 ADF 版本号， 可以先用 ADF master + IDF 5.4 再试一下

我的版本号是：v2.7-97-gf8b77be6

wujiangang, 2025-05-16:
Component config → ESP PSRAM → [*] Support for external, SPI-connected RAM

把 PSRAM 打开

VSiR22, 2025-05-19:
> Component config → ESP PSRAM → [*] Support for external, SPI-connected RAM
> 
> 把 PSRAM 打开

感谢！问题解决了

zhihanhe, 2025-05-27:
感谢感谢，我开始被AI带进沟里了。。它让我实现volc_atomic缺失的原子操作方法，还让我改MakeFile文件。折腾了几个小时