espressif/esp-adf #1450: 使用vscode编译coze_ws_app失败，且无法打开SDK配置编辑器（menuconfig）

cjl-123, 2025-05-15:
----------------------------- Delete below -----------------------------
/***************************************
 *
 *①编译失败的log
 * 
 ****************************************/

-- Found Git: D:/espidf_tool/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features       
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Dependencies lock doesn't exist, solving dependencies.
NOTICE: Using component placed at D:\espidf\v5.4.1\esp-idf\examples\common_components\protocol_examples_common for dependency "protocol_examples_common", specified in D:\espidf_tool\Espressif\test\coze_ws_app\main\idf_component.yml
....CMake Error at D:/espidf/v5.4.1/esp-idf/tools/cmake/build.cmake:610 (message):
  ERROR: Error during request

  URL:
  https://components-file.espressif.com/components/espressif/esp_gmf_audio.json

  Status code: 403 Forbidden

Call Stack (most recent call first):
  D:/espidf/v5.4.1/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)   
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!

/***************************************
 *
 *②打开编辑器失败的log
 * 
 ****************************************/

-- Found Git: D:/espidf_tool/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")

-- ccache will be used for faster recompilation

-- The C compiler identification is GNU 14.2.0

-- The CXX compiler identification is GNU 14.2.0

-- The ASM compiler identification is GNU

-- Found assembler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: D:/espidf_tool_path/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32

NOTICE: Dependencies lock doesn't exist, solving dependencies.

NOTICE: Using component placed at D:\espidf\v5.4.1\esp-idf\examples\common_components\protocol_examples_common for dependency "protocol_examples_common", specified in D:\espidf_tool\Espressif\test\coze_ws_app\main\idf_component.yml

---------------------------ERROR--------------------------

CMake Error at D:/espidf/v5.4.1/esp-idf/tools/cmake/build.cmake:610 (message):
  ERROR: Error during request

  URL:
  https://components-file.espressif.com/components/espressif/esp_gmf_audio.json

  Status code: 403 Forbidden

Call Stack (most recent call first):

-----------------------END OF ERROR-----------------------
....-- Configuring incomplete, errors occurred!

---------------------------ERROR--------------------------

  D:/espidf/v5.4.1/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:7 (project)

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the D:\espidf_tool\Espressif\test\coze_ws_app\build\log\idf_py_stderr_output_3700 and D:\espidf_tool\Espressif\test\coze_ws_app\build\log\idf_py_stdout_output_3700

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

wujiangang, 2025-05-15:
用的是哪里获取的 coze_ws_app？
https://components.espressif.com/components/espressif/esp_coze/versions/0.5.1/examples?language=en
通过这里的 idf.py create-project-from-example "espressif/esp_coze=0.5.1:coze_ws_app" 获取的？

是否有做过其他操作？

cjl-123, 2025-05-15:
通过git clone https://github.com/espressif/esp-adf.git 获取的SDK，esp-adf-master\examples\ai_agent\coze_ws_app路径下的coze_ws_app。没有做其他更改,直接在vscode上面通过esp-idf插件进行编译的

cjl-123, 2025-05-15:
> 用的是哪里获取的 coze_ws_app？ https://components.espressif.com/components/espressif/esp_coze/versions/0.5.1/examples?language=en 通过这里的 idf.py create-project-from-example "espressif/esp_coze=0.5.1:coze_ws_app" 获取的？
> 
> 是否有做过其他操作？
通过git clone https://github.com/espressif/esp-adf.git 获取的SDK，esp-adf-master\examples\ai_agent\coze_ws_app路径下的coze_ws_app。没有做其他更改,直接在vscode上面通过esp-idf插件进行编译的

shootao, 2025-05-15:
@cjl-123 可以先提供一下你的 ADF版本号看一下

cjl-123, 2025-05-15:
@shootao 在哪看版本？

wujiangang, 2025-05-15:
和 ADF 的版本没有关系，这个问题只有在 Windows 上可复现，Linux 和 MacOS 均无此问题。看上去像是 Windows 上的包管理器的依赖有点问题，esp_audio_simple_player 这个包，应该去拉最新的 0.9.3，但似乎去拉了 0.9.1，临时的改动可以是修改 https://github.com/espressif/esp-adf/blob/master/examples/ai_agent/coze_ws_app/main/idf_component.yml#L16 为
```
version: '^0.9.3'
```
你试一下是否可行。

cjl-123, 2025-05-15:
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:100:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
  100 | #define LOG_COLOR_I           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_I'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:186:86: note: in expansion of macro 'LOG_FORMAT'
  186 |         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

[image]

@wujiangang 请问编译的时候出现这个错误，该如何修改？

wujiangang, 2025-05-15:
可以编译了的话，那就是这个问题已解决，编译是另外一个问题了，请关闭此 issue，新开一个。

cjl-123, 2025-05-15:
好的

closed 2025-05-16 by wujiangang