espressif/esp-adf #1452: 使用vscode编译coze_ws_app例程时出现错误

cjl-123, 2025-05-16:
/***************************************
*
*编译时出现如下错误。尝试修改后，出现的错误更多了，帮忙看一下如何解决。
*
****************************************/

In file included from D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:15,
                 from D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:11:
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c: In function 'playback_write_callback':
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:98:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
   98 | #define LOG_COLOR_E           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_E'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:182:86: note: in expansion of macro 'LOG_FORMAT'
  182 |         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:99:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
   99 | #define LOG_COLOR_W           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_W'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:183:86: note: in expansion of macro 'LOG_FORMAT'
  183 |         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:101:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
  101 | #define LOG_COLOR_D           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_D'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:184:86: note: in expansion of macro 'LOG_FORMAT'
  184 |         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:102:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
  102 | #define LOG_COLOR_V           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_V'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:185:86: note: in expansion of macro 'LOG_FORMAT'
  185 |         else if (level==ESP_LOG_VERBOSE )   { esp_log_write(ESP_LOG_VERBOSE,    tag, LOG_FORMAT(V, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log_color.h:100:31: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
  100 | #define LOG_COLOR_I           ""
      |                               ^~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:62:37: note: in expansion of macro 'LOG_COLOR_I'
   62 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:186:86: note: in expansion of macro 'LOG_FORMAT'
  186 |         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:205:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  205 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
D:/espidf/v5.4.1/esp-idf/components/log/include/esp_log.h:112:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  112 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
D:/espidf_tool/Espressif/test/coze_ws_app/main/audio_processor.c:396:9: note: in expansion of macro 'ESP_LOGE'
  396 |         ESP_LOGE(TAG, "Write to codec dev failed (0x%x)\n");
      |         ^~~~~~~~
cc1.exe: some warnings being treated as errors
[5/11] Completed 'bootloader'
ninja: build stopped: subcommand failed.

[image]

wujiangang, 2025-05-16:
你应该是使用了较老的 ADF 版本，之前确实有这个编译问题。请拉取最新的代码，看你之前是下载的压缩包，建议使用 git 命令来拉取，能获取到实时的更新。

cjl-123, 2025-05-16:
@wujiangang 拉gitee上面的可以吗？刚才试了一下github,拉取失败。

shootao, 2025-05-16:
也可以，已经同步了

wujiangang, 2025-05-16:
参看：https://gitee.com/EspressifSystems/esp-gitee-tools/blob/master/docs/README-jihu-mirror.md
建议用 jihu 的镜像

cjl-123, 2025-05-16:
好的