espressif/esp-adf #1454: 移植 DLNA 功能到小智 1.6.2 时，DLNA 导致设备重启

jiangyanfeng, 2025-05-22:
Xiaozhi-ESP32 1.6.2
ESP-IDF 版本 5.4.1
ESP-ADF 版本 2.4
硬件 xingzhi-cube-1.54tft-wifi

直接编译例程 DLNA 没有问题。
将功能代码移植到小智（https://github.com/78/xiaozhi-esp32/）后，系统崩溃重启。由于 esp_media_protocols 没有开源，不方便进一步定位问题，请求帮助。

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4211ad40  PS      : 0x00060c30  A0      : 0x8215e24e  A1      : 0x3c5b6a00
--- 0x4211ad40: upnp_serve_root_files at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_upnp/upnp.c:102

A2      : 0x3c5b4544  A3      : 0x3c5b454c  A4      : 0x0000000d  A5      : 0x00000000
A6      : 0x3c5b6ab0  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3c5b69e0
A10     : 0x00000000  A11     : 0x3c3f976d  A12     : 0x3c3fa9ca  A13     : 0xffffffff
A14     : 0xffffffff  A15     : 0x3c5b6b00  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000044  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc
--- 0x400556d5: strlen in ROM
--- 0x400556e5: strlen in ROM

Backtrace: 0x4211ad3d:0x3c5b6a00 0x4215e24b:0x3c5b6aa0 0x4215ce2d:0x3c5b6ae0 0x4215cec1:0x3c5b6b70 0x4215d4e0:0x3c5b6b90 0x4215bc2a:0x3c5b6bb0 0x422089bf:0x3c5b6bd0 0x4215c432:0x3c5b6bf0 0x4215c49e:0x3c5b6c40 0x4038778d:0x3c5b6c60
--- 0x4211ad3d: upnp_serve_root_files at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_upnp/upnp.c:97
--- 0x4215e24b: httpd_uri at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_uri.c:331
--- 0x4215ce2d: httpd_parse_req at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_parse.c:663
--- 0x4215cec1: httpd_req_new at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_parse.c:791
--- 0x4215d4e0: httpd_sess_process at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_sess.c:427
--- 0x4215bc2a: httpd_process_session at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_main.c:267
--- 0x422089bf: httpd_sess_enum at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)
--- 0x4215c432: httpd_server at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_main.c:317
--- 0x4215c49e: httpd_thread at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_http_server/src/httpd_main.c:339
--- 0x4038778d: vPortTaskWrapper at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 0eeaea867

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037a155
--- 0x4037a155: esp_restart_noos at D:/Users/JIANG/Documents/esp/v5.4.1/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:160

shootao, 2025-05-23:
@jiangyanfeng 
可以打开 `CONFIG_ESP_GDBSTUB_ENABLED`, 先看一下 各个 frame 函数的入参有没有问题

jiangyanfeng, 2025-05-26:
这个选项是开启的，我重新确认了下，并 clean 后重新编译了，问题依旧。
以下是编译配置：
`
#
# GDB Stub
#
CONFIG_ESP_GDBSTUB_ENABLED=y
# CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME is not set
CONFIG_ESP_GDBSTUB_SUPPORT_TASKS=y
CONFIG_ESP_GDBSTUB_MAX_TASKS=32
# end of GDB Stub
`

是否和内存有关？有8MB PSRAM.
`
# SPI RAM config
#
# CONFIG_SPIRAM_MODE_QUAD is not set
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
# CONFIG_SPIRAM_XIP_FROM_PSRAM is not set
# CONFIG_SPIRAM_FETCH_INSTRUCTIONS is not set
# CONFIG_SPIRAM_RODATA is not set
CONFIG_SPIRAM_SPEED_80M=y
# CONFIG_SPIRAM_SPEED_40M is not set
CONFIG_SPIRAM_SPEED=80
# CONFIG_SPIRAM_ECC_ENABLE is not set
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
# CONFIG_SPIRAM_MEMTEST is not set
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
# CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP is not set
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=49152
# CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY is not set
# CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY is not set
# end of SPI RAM config
# end of ESP PSRAM
`

jiangyanfeng, 2025-06-03:
@shootao 能帮忙看看吗？

ThunderDai, 2025-06-04:
@jiangyanfeng  仍然建议打开 CONFIG_ESP_SYSTEM_PANIC_GDBSTUB，当系统异常 crash 时自动进入 GDB。

以下是常见指令，可输入 thread apply all backtrace 查看 log：

bt [-full] [n]：打印当前调试栈帧信息。若多线程，应执行 thread apply all backtrace
f [args]：栈帧编号；栈帧地址；函数名 都可以作为参数
info registers：打印寄存器
info frame：查看栈帧中存储的信息
info args：查看当前函数各个参数值
info locals：查看函数中局部变量值
p/x [var]：打印变量
x/[n][f][u] [addr]：打印地址对应的值，n 是打印单元数，f 是打印格式，u 是单元长度。
q：退出
list：显示源程序内容

备注：
打印格式：x 十六进制；d 十进制；c 字符显示；f 浮点数显示；
单元长度：b 字节；h 半字；w 四字节；g 八字节

如果有复现工程，最好也可以提供复现工程。此外还可以使能 HEAP_ABORT_WHEN_ALLOCATION_FAILS 确认是否是内存申请失败导致的访问空结构体问题：https://docs.espressif.com/projects/esp-techpedia/zh_CN/latest/esp-friends/advanced-development/debugging/guru-meditation.html#load-access-fault-loadprohibited