espressif/esp-adf #1458: Can't build project with custom board descrition (fatal error: board.h: No such file or directory)

P0nyHunt3r, 2025-06-05:
## Environment

- Audio development kit:      ESP32-C3-Lyra-v2.0
- IDF version: v5.4
- ADF version: v2.7-105-g4200c64d
- Build system:  CMake
- Compiler version: 14.2.0
- Operating system:     Windows
- (Windows only) Environment type: IDE? (i dont know)
- Using an IDE?: VS Code 1.100.3 with ESP-IDF 1.9.1 plugin 

## Problem Description

Сan't connect/build project with custom board description.
I understand that the problem is in the configuration of the Cmake files, but I can't figure out what is wrong.

I found similar problems with other users but couldn't find a solution.
[1](https://github.com/espressif/esp-adf/issues/575)
[2](https://github.com/espressif/esp-adf/issues/1191)

> please help me 🥺

### Expected Behavior

successful project build

### Actual Behavior

The build ends with an error:
fatal error: board.h: No such file or directory

### Steps to Reproduce

1. create project based on example play_http_mp3_example
2. select esp32-c3 and ESP32-C3-Lyra-v2.0 as target
3. try to compile... (great success✅)
4. copy all files from esp-adf\components\audio_board\esp32_c3_lyra in to project folder (project_name/components/custom_board)
5. Add Cmake/make/Kconfig files like in [example](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control/components/my_board)
6. Select custom audio board option in sdkconfig
7. Try to build... (error❌)
`fatal error: board.h: No such file or directory`

### Code to Reproduce This Issue

[project](https://github.com/P0nyHunt3r/esp32c3_pipeline_test)

### Build log 

[log](https://gist.github.com/P0nyHunt3r/625916947b6c03b457cb00a7ca24b8e6)

P0nyHunt3r, 2025-06-09:
@rjoly-getraline I tried to do it this way, but it didn't work completely.
[log](https://gist.github.com/P0nyHunt3r/c580fde595b10364ba820eb68fdcf04f)

I noticed that my "custom_board" folder appeared in the logs.
The libraries that the example uses have "audio_board" in the CMakeList. 

[image]

520lbl, 2025-07-08:
You can modify the CMakeLists.txt file as follows to add `custom_board` to the environment variable `COMPONENT_ADD_INCLUDEDIRS`:
```
# Edit following two lines to set component requirements (see docs)
set(COMPONENT_REQUIRES)
set(COMPONENT_PRIV_REQUIRES audio_sal audio_hal esp_dispatcher esp_peripherals display_service json)
 
message(STATUS "Use customs board component")
list(APPEND COMPONENT_ADD_INCLUDEDIRS ./)
set(COMPONENT_SRCS
./board.c
./board_pins_config.c
)
 
register_component()
 
IF (IDF_VERSION_MAJOR GREATER 3)
idf_component_get_property(audio_board_lib audio_board COMPONENT_LIB)
set_property(TARGET ${audio_board_lib} APPEND PROPERTY INTERFACE_LINK_LIBRARIES ${COMPONENT_LIB})
 
ELSEIF (IDF_VERSION_MAJOR EQUAL 3)
set_property(TARGET idf_component_audio_board APPEND PROPERTY INTERFACE_INCLUDE_DIRECTORIES  $<TARGET_PROPERTY:${COMPONENT_TARGET},INTERFACE_INCLUDE_DIRECTORIES>)
 
ENDIF (IDF_VERSION_MAJOR GREATER 3)