espressif/esp-adf #1461: Voip(demo) 如何支持OPUS

wjstan, 2025-06-10:
（之前在论坛提过，没有回应，发这来问一下，）我想在VOIP（esp-adf\examples\protocols\voip)里扩充编解码，
sip_service.c 
`esp_rtc_config_t sip_service_config = {
        .uri = uri,
        .ctx = av_stream,
        .local_addr = localIp,
        .acodec_type = RTC_ACODEC_OPUS, //RTC_ACODEC_G711A,
        .data_cb = &data_cb,
        .event_handler = _esp_sip_event_handler,
    };`

voip_app.c 
`av_stream_config_t av_stream_config = {
        .algo_mask = ALGORITHM_STREAM_DEFAULT_MASK,
        .acodec_samplerate = AUDIO_CODEC_SAMPLE_RATE,
        .acodec_type = AV_ACODEC_OPUS, //AV_ACODEC_G711A,
        .vcodec_type = AV_VCODEC_NULL,
        .hal = {
            .audio_samplerate = AUDIO_HAL_SAMPLE_RATE,
            .audio_framesize = PCM_FRAME_SIZE,
        },
    };
    av_stream = av_stream_init(&av_stream_config);`
并且将esp-opus(components.espressif.com/components/78/esp-opus/versions/1.0.5)相关编解码的代码扩充了av_stream.c的相应地方；然而，发现接通voip ring后， 
` esp_rtc_data_cb_t data_cb = {
        .send_audio = _send_audio,
        .receive_audio = _receive_audio,
    };`
sip的_receive_audio并没有被回调到， 请问当前的voip内部实现还没有支持这个opus payload回调吗？ 如果要扩充opus要如何做相应调整？谢谢

TempoTian, 2025-06-11:
[av_stream.zip](https://github.com/user-attachments/files/20684109/av_stream.zip)
这个版本有对OPUS进行支持，可以替换下esp-adf-libs里面的.a来做测试， av_stream还是老的代码可以和新的对比merge下
media_protocol升级版本会在后面一段时间一起升级，额可以先用这个版本来测试下看

wjstan, 2025-06-11:
感谢指导，刚才用分享的zip试了一下，编译有错误，
`D:/work/esp32/src/esp_dev/examples/protocols/components/av_stream/av_stream.c: In function 'av_audio_enc_start':
D:/work/esp32/src/esp_dev/examples/protocols/components/av_stream/av_stream.c:370:20: error: 'algorithm_stream_cfg_t' has no member named 'input_format'
  370 |         algo_config.input_format = "RM";`

将其注掉后，编译通过；正常运行，voip接通后ESP设备端受话正常（能听到手机端送来的声音），送话没有声音（且本地有回音），是否与这个algo_config.input_format有关系? 谢谢

wjstan, 2025-06-13:
> 这个你要改下调度算法，可以把一些task 放到另外一个核心上面，这样保证及时的将aec --> end --> rtc read发送出去 也可以适当的调一下task的优先级，默认example定时的会打印下task的占用，看下调度是否合理

调整了一下opus的 核心id，上面的警告消失了，非常谢谢

wjstan, 2025-06-26:
[image]
现在有个问题，接通SIP后，会报错
`E (135791) MODEL_LOADER: Can not find model in partition table
E (135797) ALGORITHM_STREAM: Failed to load models`
并且ESP32设备的受话声音不正常，说话声全变成”咚咚“声了

TempoTian, 2025-06-26:
那就是algo_mask关掉看看

wjstan, 2025-06-26:
> 那就是algo_mask关掉看看

没有效果，而且正式产品发布（需要扬声器外放）回声消除是必须要有的

wjstan, 2025-06-27:
> [image] 现在有个问题，接通SIP后，会报错 `E (135791) MODEL_LOADER: Can not find model in partition table E (135797) ALGORITHM_STREAM: Failed to load models` 并且对方受话(ESP32设备的送话)声音不正常，说话声全变成”咚咚“声了
@TempoTian 老师，这个编码问题还请帮忙看一下，另外支持opus的正式版本大概什么时候能发布？

TempoTian, 2025-06-27:
@wjstan model load失败应该不影响AEC不需要加在model的，晚一点我用最新的ADF测试下看，之前测试关掉AEC是OK的。版本我需要再确认下不知道是单独发component还是放ADF，ADF后面可能就不升级版本了，我要再确认下看。