espressif/esp-adf #1465: 使用pipeline_sonic例程时，SD卡中保存的录音不正确

Crazy-lxy, 2025-06-16:
## Environment

- Audio development kit:      ESP32-S3-Korvo-2
- Audio kit version (for ESP32-S3-Korvo-2 v3
- [Required] Module or chip used:  ESP32-S3-WROOM-1
- [Required] IDF version : v5.4.1-4c2820d377
- [Required] ADF version : v2.7-106-g40abfc5b
- Build system:         [Make|CMake|idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version:
   xtensa-esp-elf-gcc.exe (crosstool-NG esp-14.2.0_20241119) 14.2.0
- Operating system:     [Windows]
- (Windows only) Environment type: [PowerShell]
- Using an IDE?: [Yes (please give details)] VSCode + ESP-IDF
- Power supply:         [USB]

## Problem Description

使用 ESP-ADF 中 pipeline_sonic例程的时候，发现按下录音按键以后采集到的.wav录音文件播放时声音异常（将其插入windows电脑下播放），就像经过变声一样，但是我看见例程介绍中介绍的保存录音文件的时候并没有经过变声处理。

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to Reproduce This Issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
/* Voice Changer Example with Sonic

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "esp_log.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_mem.h"
#include "audio_common.h"
#include "fatfs_stream.h"
#include "i2s_stream.h"
#include "wav_encoder.h"
#include "wav_decoder.h"
#include "board.h"
#include "audio_sonic.h"
#include "esp_peripherals.h"
#include "periph_sdcard.h"
#include "periph_button.h"

static const char *TAG = "SONIC_EXAMPLE";
static esp_periph_set_handle_t set;

#define SAMPLE_RATE         16000
#define CHANNEL             1
#define BITS                16
[... 302 lines cut ...]
I (884) SONIC_EXAMPLE: [1.3] Register audio elements to recorder pipeline
I (884) SONIC_EXAMPLE: [2.2] Create audio elements for playback pipeline
I (894) SONIC_EXAMPLE: [2.3] Register audio elements to playback pipeline
I (894) SONIC_EXAMPLE: [ 3 ] Set up  event listener
W (904) SONIC_EXAMPLE: Press [Rec] to start recording
E (6434) SONIC_EXAMPLE: Now recording, release [Rec] to STOP
W (6434) AUDIO_PIPELINE: Without stop, st:1
W (6434) AUDIO_PIPELINE: Without wait stop, st:1
W (6434) AUDIO_ELEMENT: [file_reader] Element has not create when AUDIO_ELEMENT_TERMINATE
W (6444) AUDIO_ELEMENT: [wav_decoder] Element has not create when AUDIO_ELEMENT_TERMINATE
W (6444) AUDIO_ELEMENT: [i2s_writer] Element has not create when AUDIO_ELEMENT_TERMINATE
I (6454) SONIC_EXAMPLE: Setup file path to save recorded audio
W (6464) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
I (9134) SONIC_EXAMPLE: START Playback
W (9134) AUDIO_ELEMENT: IN-[wav_encoder] AEL_IO_ABORT
W (9134) AUDIO_ELEMENT: [wav_encoder] Element already stopped
I (9144) SONIC_EXAMPLE: Setup file path to read the wav audio to play
W (9144) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

520lbl, 2025-06-27:
在新版本的 i2s_stream_idf5 中，没有对 MONO 状态下制定声道，可以在 create_i2s_stream 函数中加入如下代码：

`i2s_stream_set_channel_type(&i2s_cfg, I2S_CHANNEL_TYPE_ONLY_LEFT);`

即可正常录音

Crazy-lxy, 2025-06-27:
实验了一下 问题解决了 谢谢(^🙏^)