espressif/esp-adf #1469: The Coze platform uses the wss protocol, but I couldn’t find where to configure the certificate. No server verification option set in esp_tls_cfg_t structure

skipify, 2025-06-18:
The Coze platform uses the wss protocol, but I couldn’t find where to configure the certificate.
The URL I’m using is:
wss://ws.coze.cn/v1/chat?bot_id=7480067804642050059

I also noticed that in the esp_coze component, the libcoze_websocket.a library is not open source, so currently I have no way of knowing where or how to configure the certificate.
```
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_log.h"
#include "esp_flash.h"
#include "esp_system.h"

#include "nvs_flash.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "protocol_examples_common.h"
#include "protocol_examples_utils.h"
#include "esp_opus_dec.h"

#include "driver/i2s_std.h"
#include "driver/i2s_tdm.h"
#include "soc/soc_caps.h"

#include "esp_codec_dev.h"
#include "esp_codec_dev_defaults.h"

#include "esp_coze_chat.h"
#include "driver/i2c_master.h"

// I2C
#define ESP_GMF_I2C_SDA_IO_NUM      (GPIO_NUM_0)
#define ESP_GMF_I2C_SCL_IO_NUM      (GPIO_NUM_1)

// I2S
#define ESP_GMF_I2S_DAC_MCLK_IO_NUM (GPIO_NUM_10)
#define ESP_GMF_I2S_DAC_BCLK_IO_NUM (GPIO_NUM_8)
#define ESP_GMF_I2S_DAC_WS_IO_NUM   (GPIO_NUM_12)
#define ESP_GMF_I2S_DAC_DO_IO_NUM   (GPIO_NUM_11)
#define ESP_GMF_I2S_DAC_DI_IO_NUM   (GPIO_NUM_7)

// #define ESP_GMF_I2S_ADC_MCLK_IO_NUM (GPIO_NUM_10)
// #define ESP_GMF_I2S_ADC_BCLK_IO_NUM (GPIO_NUM_8)
// #define ESP_GMF_I2S_ADC_WS_IO_NUM   (GPIO_NUM_12)
// #define ESP_GMF_I2S_ADC_DO_IO_NUM   (GPIO_NUM_7)
// #define ESP_GMF_I2S_ADC_DI_IO_NUM   (GPIO_NUM_11)
// PA
#define ESP_GMF_AMP_IO_NUM          (GPIO_NUM_13)

static const char *TAG = "COZE_TEST";

typedef struct {
    i2s_chan_handle_t tx_handle;
    i2s_chan_handle_t rx_handle;
} i2s_keep_t;
static i2s_keep_t i2s_keep;
static i2c_master_bus_handle_t i2c_bus_handle;
static void *ops_dec_handle;

static esp_codec_dev_handle_t codec_dev;
static esp_codec_dev_handle_t codec_dev_ply;
static uint8_t *opus_data = NULL;
static uint32_t opus_data_len = 0;

static int ut_i2c_init()
{
[... 186 lines cut ...]
        printf("1-> %d\n", gpio_get_level(ESP_GMF_AMP_IO_NUM));
        gpio_set_level(ESP_GMF_AMP_IO_NUM, 1);
        printf("heap: %ld, total: %ld\n", esp_get_minimum_free_heap_size(), esp_get_free_heap_size());
    }
}

```
output log：

I (10590) ES8311: Work in Slave mode
I (10597) ESP_COZE_CHAT: wss_url: wss://ws.coze.cn/v1/chat?bot_id=7480067804642050059
I (10598) ESP_COZE_CHAT: token: Bearer pat_FRF6d2kjfa5e74rDweuL5S7GFIv9QS5UUNxYUmLjA6x0DHhIQ9pMERHIdarjAPGO
I (10605) ESP_COZE_CHAT: WEBSOCKET_EVENT_BEGIN
I (10609) websocket_client: Started
E (10627) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (10628) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (10640) esp-tls: create_ssl_handle failed
E (10644) esp-tls: Failed to open new connection
E (10649) transport_base: Failed to open a new connection
E (10658) transport_ws: Error connecting to host ws.coze.cn:443
E (10662) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_ERR_MBEDTLS_SSL_SETUP_FAILED, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=0, errno=119
E (10680) ESP_COZE_CHAT: WEBSOCKET_EVENT_ERROR
I (10684) websocket_client: Reconnect after 1000 ms
I (10689) ESP_COZE_CHAT: WEBSOCKET_EVENT_DISCONNECTED
E (10694) ESP_COZE_CHAT: Last error reported from esp-tls: 0x8017

shootao, 2025-06-20:
Hi @skipify 
Try this one, it contains HTTP requests
[esp_coze_v0.6.tar.gz](https://github.com/user-attachments/files/20832023/esp_coze_v0.6.tar.gz)

jason-mao, 2025-07-17:
esp coze v0.6 has been released on 5770ffd7e262dca6b682c9692d22b473325c74e2

closed 2025-07-17 by jason-mao